lvgl/lvgl #2513: [v7.11] image to set angle will en

Littermonkey, 2021-08-31:
### Perform all steps below and tick them with [x]
- [ ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ ] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

when the image to set angle use API "lv_img_get_angle", but get the resulting image will be blurred.

### To Reproduce

	//Write codes screen
	ui->screen = lv_obj_create(NULL, NULL);

	//Write codes screen_img_1
	ui->screen_img_1 = lv_img_create(ui->screen, NULL);

	lv_obj_set_pos(ui->screen_img_1, 153, 26);
	lv_obj_set_size(ui->screen_img_1, 199, 166);
	lv_obj_set_click(ui->screen_img_1, true);
	lv_img_set_src(ui->screen_img_1,&_shark_jpg_alpha_199x166);
	lv_img_set_pivot(ui->screen_img_1, 0,0);
	lv_img_set_angle(ui->screen_img_1, 20);

### Expected behavior

### Screenshots or video

[image]

embeddedt, 2021-08-31:
You can try [enabling antialiasing on the image](https://docs.lvgl.io/7.11/widgets/img.html#_CPPv420lv_img_set_antialiasP8lv_obj_tb), however, there is a performance cost to doing so.

Littermonkey, 2021-09-06:
Sorry to be late to post the result, unfortunately, enabling antialiasing on the image as your said, but it doesn't work on my side. `lv_img_set_angle(ui->screen_img_1, 80); lv_img_set_antialias(ui->screen_img_1, true);`   am i missing something?

kisvegabor, 2021-09-07:
The edges of the rotated can be really jagged. I suggest adding a "transparent border" to your image (a few pixels of transparent area on each sides). It will allow LVGL to correctly mix the first and last colorful pixels to the transparent area.

closed 2021-09-30 by lvgl-bot