lvgl/lvgl #2531: Can´t compile the code using another built in font

danilolima, 2021-09-05:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
Can´t compile the code using another built in font.
I'm missing something?
I need change another file?
I'm using PlatformIo in Clion on my esp32 board with 8.0 lib version

### To Reproduce
1 - Enable the font in the  lv_conf.h
```
#define LV_FONT_MONTSERRAT_18    1
```
2- Add to label style
```
static lv_style_t style;
lv_style_init(&style);
lv_style_set_text_font(&style, &lv_font_montserrat_18);
lv_obj_add_style(label, &style, 0);
```

### Expected behavior
Compile normal

### Screenshots or video
[image]

embeddedt, 2021-09-05:
Try a full clean and rebuild first; some IDEs don't pick up on changes to `lv_conf.h` until you do.

Failing that, add an `#error test` directive to `lv_conf.h` and make sure you see the compilation fail due to that error. If it does not, you are editing the wrong `lv_conf.h` file, or it's in the wrong location.

danilolima, 2021-09-06:
You right!
My IDE was pointing to wrong file

Thanks!

closed 2021-09-06 by danilolima