lvgl/lvgl #2534: WIP: Support bitbake

rzr, 2021-09-06:
Hi,
I am creating this ticket to track my progress to support LVGL along Bitbake (Yocto/OE).
I plan to upstream some fixes and build recipes somewhere, and will refer to this issue for cooperation purposes

Here is a SDL2 version running on RPi3 on AllScenariOS:

[image]

Relate-to: https://git.ostc-eu.org/OSTC/planning/core-os/-/issues/219

kisvegabor, 2021-09-09:
Thanks for opening this issue. Let us know if you need help form our side.

rzr, 2021-09-09:
So far it goes fine thx,
I am also about to publish a simple app that can be used as reference,  and reference image.
It's working with sdl2 backend which is still under review ( https://github.com/lvgl/lvgl/pull/2484 )
I'll update this ticket for anyone who want to use it, feedback will be appreciated.
Then some hints about wayland backend would be welcome, but one problem at a time.

Regards

kisvegabor, 2021-09-13:
> Then some hints about wayland backend would be welcome, but one problem at a time.

Have you already seen its [README](https://github.com/lvgl/lv_drivers/tree/master/wayland)?

kisvegabor, 2021-09-28:
Not stale

rzr, 2021-09-28:
yes I plan to upstream recipes to OE, meanwhile check status at:

https://git.ostc-eu.org/rzr/dialog-lvgl/-/wikis/home

rzr, 2021-10-11:
It's merged in master branch
https://github.com/openembedded/meta-openembedded/commit/0924d41ac4cf968fc302e4f7e52d969cd3899423

Should I keep this issue open for tracking ? 

Thanks for support anyway

kisvegabor, 2021-10-12:
Thank your for your great work. 

It'd be great to tell our users that LVGL is available in bitbake. 

Could you add a new file in the [Get started](https://docs.lvgl.io/master/get-started/index.html) section of our docs?

kisvegabor, 2022-01-07:
Hi,

What is the current state of this issue?

rzr, 2022-01-07:
we can close it it's upstreamed  in oe

closed 2022-01-07 by rzr

kisvegabor, 2022-01-08:
Thanks you! What is the final place where users can find the recipe?

rzr, 2022-01-24:
Is the current place:

https://github.com/openembedded/meta-openembedded/tree/master/meta-oe/recipes-graphics/lvgl

More details to be explained at:

https://fosdem.org/2022/schedule/event/lvgl/