lvgl/lvgl #2535: Label animation throws events

qinshijing, 2021-09-06:
### Introduce the problem
Hi, when I use the label, I set the label to LV_LABEL_LONG_SCROLL_CIRCULAR, but I also want to set the scroll animation to repeat only a limited number of times, and I can call back my function after the animation ends. I looked at the label code and found that this only needs to throw events and animations before starting the animation, similar to the animation processing in lv_obj_scroll_by. Do you think this is appropriate?

### Examples and cases

### Suggested solution
lv_event_send(obj, LV_EVENT_SCROLL_BEGIN, &a);
lv_anim_start(&a);

kisvegabor, 2021-09-14:
Hi,

`lv_anim` can animate any type of variable, not only `lv_obj_t`. Therefore we can call any `lv_obj` related functions on the animated variable by default. 

It's not that elegant but you can do this:
```c

  lv_obj_t * obj = lv_label_create(lv_scr_act());
  lv_label_set_text(obj, "0123456789");
  lv_obj_set_width(obj, 20);
  lv_label_set_long_mode(obj, LV_LABEL_LONG_SCROLL_CIRCULAR);
  lv_obj_update_layout(obj);
  lv_anim_t * a = lv_anim_get(obj, NULL);
  a->repeat_cnt = 3;
```

qinshijing, 2021-09-15:
Yes. I tried your suggestion, but the value of "a" is NULL. Maybe you should use this lv_anim_t * anim_cur = lv_anim_get(obj, set_ofs_x_anim). But set_ofs_x_anim is a static function, so it cannot be done yet.

kisvegabor, 2021-09-15:
Oh, sorry, when I tested the above code I fixed it locally but forgot to push the changes. Please try with the latest master.

qinshijing, 2021-09-15:
Thanks, it works. I close this issue.

closed 2021-09-15 by qinshijing