lvgl/lvgl #2540: Enable new functions in lv_micropython

Karijn, 2021-09-08:
### Introduce the problem
I see that the new functions are in the latest lv_micropython version. 
But how can i enable them in micropython (I don't find them in the lv_mpy_example.c file)

### Examples and cases
i created a PR for lv_obj_move_to_index() and others, now i would like to use them in micropython.

### Suggested solution
probably add something to gen_mpy.py ?

embeddedt, 2021-09-08:
Ensure that you have checked out the latest `master` commit from lv_micropython and that you ran `git submodule update --recursive` afterwards.

amirgon, 2021-09-08:
I can confirm `move_to_index` is working on Micropython.
[Here is an example on the online simulator ](https://sim.lvgl.io/v8.1/micropython/ports/javascript/index.html?script_startup=https://raw.githubusercontent.com/lvgl/lvgl/eb6ae526432453e4b9dbc7a760cd65d164050548/examples/header.py&script=https://raw.githubusercontent.com/lvgl/lvgl/eb6ae526432453e4b9dbc7a760cd65d164050548/examples/widgets/list/lv_example_list_1.py&script_direct=42801b5758b3fc863827d1c4bb49a21dea43c927)

Karijn, 2021-09-09:
Hmmm, i couldn't find the functions in code after building, but they do work

closed 2021-09-09 by Karijn

amirgon, 2021-09-09:
> Hmmm, i couldn't find the functions in code after building, but they do work

`lv_mpy_example.c` is just an example (for user reference) and not really used for anything.

The actual lvgl module is automatically generated as part of the build process so you can find it under the "build" directory (it's called `lv_mpy.c`).