lvgl/lvgl #2552: Reset after adding Bluetooth or WIFI

LilyGO, 2021-09-13:
We use GitHub issues for development related discussions.
Please use the [forum](https://forum.littlevgl.com/) to ask questions.

**Describe the issue**

1. I want to add Bluetooth and WIFI functions to my project. Before adding them, all the code runs as expected, but once the statement is added, the program keeps restarting and resetting.I have tried running the WIFI scan and connection example separately on PlatfromIO, the result is normal, but as long as I add the initialization to my project, it keeps restarting.
2. In addition, I tried to add Bluetooth initialization separately. In the absence of WIFI initialization, I just added BluetoothSerial SerialBT; and it started to restart continuously.
3. If necessary, I am willing to provide the entire project to help fix this problem

**Code to reproduce the issue**

`#include "XYWatch_hal.h"
void setup()
{
   pinMode(PWR_ON, OUTPUT);
   digitalWrite(PWR_ON, HIGH);

   Serial.begin(115200); /* prepare for possible serial debug */
   Serial.println("Hello T-Watch");
   XYWatch::HAL_Init();

   XYWatchApp::AppInit();
   Serial.println("XYWatch Setup done");
}

void loop()
{
   lv_timer_handler(); /* let the GUI do its work */
   XYWatch::HAL_Update();
}`

`#ifndef __XYWATCH_HAL_H_
#define __XYWATCH_HAL_H_

#include <lvgl.h>
#include <Config/Config.h>
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include <TFT_eSPI.h>
#include "./Software/driver/Input.h"
#include "OneButton.h"
#include <CST816S.h>
#include <AiEsp32RotaryEncoder.h>
#include <QMC5883LCompass.h>
#include <TinyGPS++.h>
#include <HardwareSerial.h>
#include <Adafruit_BME280.h>
#include <Adafruit_Sensor.h>
#include <pcf8563.h>
#include <Wire.h>
#include <SPI.h>
#include <bma.h>
#include <SdFat.h>
#include "lv_demo.h"
#include "../src/Software/Pages/AppManager.h"
#include <../lv_lib_gif/lv_gif.h>
#include "WiFi.h"
//#include "BluetoothSerial.h"

namespace XYWatch
{
    void HAL_Init();
    void HAL_Update();
    void HAL_Sleep();

    uint8_t HAL_Power();

    /* WIFI */
    void WIFI_Init(void);
[... 743 lines cut ...]
Compiling .pio\build\esp32dev\FrameworkArduino\libb64\cdecode.c.o
Compiling .pio\build\esp32dev\FrameworkArduino\libb64\cencode.c.o
Compiling .pio\build\esp32dev\FrameworkArduino\main.cpp.o
C:\Users\19021\.platformio\packages\framework-arduinoespressif32\cores\esp32\esp32-hal-spi.c: In function 'spiTransferBytesNL':
C:\Users\19021\.platformio\packages\framework-arduinoespressif32\cores\esp32\esp32-hal-spi.c:922:39: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_out8 = &result[c_longs-1];
                                       ^
C:\Users\19021\.platformio\packages\framework-arduinoespressif32\cores\esp32\esp32-hal-spi.c:923:40: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_data8 = &last_data;
                                        ^
Compiling .pio\build\esp32dev\FrameworkArduino\stdlib_noniso.c.o
Compiling .pio\build\esp32dev\FrameworkArduino\wiring_pulse.c.o
Compiling .pio\build\esp32dev\FrameworkArduino\wiring_shift.c.o
Archiving .pio\build\esp32dev\libFrameworkArduino.a
Archiving .pio\build\esp32dev\lib20b\libTFT_eSPI.a
Linking .pio\build\esp32dev\firmware.elf
Retrieving maximum program size .pio\build\esp32dev\firmware.elf
Checking size .pio\build\esp32dev\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  13.9% (used 45500 bytes from 327680 bytes)
Flash: [====      ]  38.4% (used 1206915 bytes from 3145728 bytes)
Building .pio\build\esp32dev\firmware.bin
esptool.py v3.1
Merged 1 ELF section
======================================================= [SUCCESS] Took 47.97 seconds =======================================================`

kisvegabor, 2021-09-13:
Hi,

probably the stack size of the task that calls `lv_timer_hander` is too small.

LilyGO, 2021-09-13:
thank you for your reply

Sorry for the missing LVGL log in the above information

I tried to set LV_MEM_CUSTOM to 0, but its result is still the same

`/*=========================
   MEMORY SETTINGS
 *=========================*/

/*1: use custom malloc/free, 0: use the built-in `lv_mem_alloc()` and `lv_mem_free()`*/
#define LV_MEM_CUSTOM 1
#if LV_MEM_CUSTOM == 0
/*Size of the memory available for `lv_mem_alloc()` in bytes (>= 2kB)*/
#define LV_MEM_SIZE (2 * 32U * 1024U) /*[bytes]*/

/*Set an address for the memory pool instead of allocating it as a normal array. Can be in external SRAM too.*/
#define LV_MEM_ADR 0                                             /*0: unused*/
#else                                                            /*LV_MEM_CUSTOM*/
#define LV_MEM_CUSTOM_INCLUDE "esp_heap_caps.h" /* <stdlib.h> */ /*Header for the dynamic memory function*/
#define LV_MEM_CUSTOM_ALLOC malloc /* ps_malloc */               /* malloc */
#define LV_MEM_CUSTOM_FREE free /* heap_caps_free */             /* free */
#define LV_MEM_CUSTOM_REALLOC realloc /* ps_realloc */           /* realloc */
#endif                                                           /*LV_MEM_CUSTOM*/

/*Use the standard `memcpy` and `memset` instead of LVGL's own functions. (Might or might not be faster).*/
#define LV_MEMCPY_MEMSET_STD 1
`

`rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
Hello T-Watch
[I][esp32-hal-i2c.c:1138] i2cProcQueue(): Bus busy, reinit
[Warn]	(0.419, +419)	 lv_init: Memory integrity checks are enabled via LV_USE_ASSERT_MEM_INTEGRITY which makes LVGL much slower 	(in lv_obj.c line #138)
[Warn]	(0.422, +3)	 lv_init: Object sanity checks are enabled via LV_USE_ASSERT_OBJ which makes LVGL much slower 	(in lv_obj.c line #142)
[Warn]	(0.435, +13)	 lv_style_init: Style might be already inited. (Potential memory leak) 	(in lv_style.c line #44)
[Warn]	(0.445, +10)	 lv_style_init: Style might be already inited. (Potential memory leak) 	(in lv_style.c line #44)
[Warn]	(0.455, +10)	 lv_style_init: Style might be already inited. (Potential memory leak) 	(in lv_style.c line #44)
[Warn]	(0.465, +10)	 lv_style_init: Style might be already inited. (Potential memory leak) 	(in lv_style.c line #44)
[Warn]	(0.475, +10)	 lv_style_init: Style might be already inited. (Potential memory leak) 	(in lv_style.c line #44)
[Warn]	(0.486, +11)	 lv_style_init: Style might be already inited. (Potential memory leak) 	(in lv_style.c line #44)
[Warn]	(0.496, +10)	 lv_style_init: Style might be already inited. (Potential memory leak) 	(in lv_style.c line #44)
[Warn]	(0.506, +10)	 lv_style_init: Style might be already inited. (Potential memory leak) 	(in lv_style.c line #44)
[Warn]	(0.516, +10)	 lv_style_init: Style might be already inited. (Potential memory leak) 	(in lv_style.c line #44)
[Warn]	(0.527, +11)	 lv_style_init: Style might be already inited. (Potential memory leak) 	(in lv_style.c line #44)
[Warn]	(0.537, +10)	 lv_style_init: Style might be already inited. (Potential memory leak) 	(in lv_style.c line #44)
[Warn]	(0.547, +10)	 lv_style_init: Style might be already inited. (Potential memory leak) 	(in lv_style.c line #44)
[Warn]	(0.557, +10)	 lv_style_init: Style might be already inited. (Potential memory leak) 	(in lv_style.c line #44)
[Warn]	(0.568, +11)	 lv_style_init: Style might be already inited. (Potential memory leak) 	(in lv_style.c line #44)
[Warn]	(0.578, +10)	 lv_style_init: Style might be already inited. (Potential memory leak) 	(in lv_style.c line #44)
[Warn]	(0.588, +10)	 lv_style_init: Style might be already inited. (Potential memory leak) 	(in lv_style.c line #44)
[Warn]	(0.598, +10)	 lv_style_init: Style might be already inited. (Potential memory leak) 	(in lv_style.c line #44)
[Warn]	(0.609, +11)	 lv_style_init: Style might be already inited. (Potential memory leak) 	(in lv_style.c line #44)
[Warn]	(0.619, +10)	 lv_style_init: Style might be already inited. (Potential memory leak) 	(in lv_style.c line #44)
[Warn]	(0.629, +10)	 lv_style_init: Style might be already inited. (Potential memory leak) 	(in lv_style.c line #44)
[Warn]	(0.639, +10)	 lv_style_init: Style might be already inited. (Potential memory leak) 	(in lv_style.c line #44)
[Warn]	(0.650, +11)	 lv_style_init: Style might be already inited. (Potential memory leak) 	(in lv_style.c line #44)
[Warn]	(0.660, +10)	 lv_style_init: Style might be already inited. (Potential memory leak) 	(in lv_style.c line #44)
[Warn]	(0.670, +10)	 lv_style_init: Style might be already inited. (Potential memory leak) 	(in lv_style.c line #44)
[Warn]	(0.680, +10)	 lv_style_init: Style might be already inited. (Potential memory leak) 	(in lv_style.c line #44)
[Warn]	(0.691, +11)	 lv_style_init: Style might be already inited. (Potential memory leak) 	(in lv_style.c line #44)
[Warn]	(0.701, +10)	 lv_style_init: Style might be already inited. (Potential memory leak) 	(in lv_style.c line #44)
[Warn]	(0.711, +10)	 lv_style_init: Style might be already inited. (Potential memory leak) 	(in lv_style.c line #44)
[... 13 lines cut ...]
[Warn]	(0.854, +10)	 lv_style_init: Style might be already inited. (Potential memory leak) 	(in lv_style.c line #44)
BMA423 Successful connection 
QMC5883L Successful connection
[... cut ...]
EXCVADDR: 0x00000000  LBEG    : 0x400de4e1  LEND    : 0x400de4ee  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x401715aa:0x3ffb1ba0 0x400f67c3:0x3ffb1bc0 0x400dfcc9:0x3ffb1c30 0x400d61a4:0x3ffb1d10 0x40170bc3:0x3ffb1d90 0x400d3b7b:0x3ffb1db0 0x400d3c95:0x3ffb1dd0 0x400db0f0:0x3ffb1e20 0x400db1b8:0x3ffb1e70 0x400db70e:0x3ffb1e90 0x400dbb55:0x3ffb1f00 0x400eabe9:0x3ffb1f50 0x400d39db:0x3ffb1f90 0x401002d9:0x3ffb1fb0 0x4008b672:0x3ffb1fd0

Rebooting...
ets Jul 29 2019 12:21:46`

LilyGO, 2021-09-14:
I found the problem, that is, my stack space is insufficient, and it takes a lot of space to turn on WIFI and Bluetooth

closed 2021-09-14 by LilyGO