lvgl/lvgl #2556: Consider an lternative library for sphinx-markdown-tables

ckielstra, 2021-09-14:
The lvgl/docs are build using the package sphinx-markdown-tables. The problem with that library is that it's published under the GPL license, a viral license. Meaning that complete LVGL should now also be released under GPL.

I asked the maintainer of sphinx-markdown-tables if he is willing to change the license type. Until then I suggest we remove this package from the requirements.

embeddedt, 2021-09-14:
I am not a lawyer, but does the GPL extend to a program's output by default? I know GCC has an explicit exception, but I don't know if it's needed in all cases.

I have no objections to changing the library we use if a suitable alternative exists.

embeddedt, 2021-09-14:
> > I know GCC has an explicit exception, but I don't know if it's needed in all cases.
> How do you mean it?

https://www.gnu.org/licenses/gcc-exception-3.1

I am not sure if this applies just to libgcc (which is a GPL library that GCC always links with your program), or if it applies to the compiler itself as well.

ckielstra, 2021-09-14:
I'm not a lawyer person too. 
Perhaps it's no problem since since we only use the output of this library?
I also opened a ticket on the project's github to see if the license can be changed to something less restrictive.

embeddedt, 2021-09-14:
In any case, this library is only used to produce the documentation PDF and HTML, which are built by the CI system and published on a GitHub repository with public access. Therefore, I don't think we are currently violating any GPL restrictions.

I did do a search for replacements but it seems that everyone links to this library as the recommended one to use. :laughing:

puzrin, 2021-09-17:
GPL is applied to how do you [distribute] [code].

1. you don't distribute that package code, it's not part of your repo
2. things, processed by code, are not subject of code's licence

=> just close this issue as "not a problem".

ckielstra, 2021-09-17:
@puzrin Thanks for the explanation.
I agree to close this ticket.

closed 2021-09-17 by ckielstra