lvgl/lvgl #2559: the italics font Incomplete display

guoweilkd, 2021-09-16:
hi, the italics font Incomplete display.like this:

[image]

Maybe there's the multi-line problem. What do I need to do?

kisvegabor, 2021-09-16:
Please send the font you have tested with.

guoweilkd, 2021-09-17:
Can I generate italic fonts from `.ttf` use lvgl Font Converter

kisvegabor, 2021-09-17:
I mean which ttf font you have used and which size.

guoweilkd, 2021-09-17:
arial.ttf  size=32, You need this branch https://github.com/guoweilkd/lv_lib_freetype/tree/style

guoweilkd, 2021-09-17:
We now use lv_font_t->adv_w to calculate the string width,but if italic used , the End-of-line position character should use `ofs_x + box_w`to calculate the character width.

kisvegabor, 2021-09-17:
I see the issue. Changing the label's width based on the last character could cause unexpected things with word wrap. E.g. due to the extra width LVGL might think a `.` still fits into the line. This way  `.` would be the last character. So it can mess up labels in special cases. 

I think a more proper solution would be to set some `ext_draw_pad` (extra area in which the object draw outside of its coordinates).  The exact required extra size is not know now but we can estimate it e.g.  like `font_height / 4`. However, this value could be saved during font conversation considering all ofs_x, adv_w and box_w.

What do you think?

guoweilkd, 2021-09-22:
> ext_draw_pad
a good idea. How do I use this interface.

guoweilkd, 2021-09-22:
a other trouble:freetype maybe malloc memory in lv_font_t->get_glyph_dsc, but in  lv_font_t->get_glyph_bitmap Can't release it(Because the returned bitmap pointer LVGL is used) .like this:
in lv_font_t->get_glyph_dsc:
[image]

in  lv_font_t->get_glyph_bitmap :

[image]

FT_Done_Glyph() needs to be called to release resources allocated by FT_Glyph_Copy().

Do you think it's necessary to add a release interface callback?

kisvegabor, 2021-09-22:
> a good idea. How do I use this interface.

It can be managed internally by the label. But until it's implemented you can add an [LV_EVENT_REFR_EXT_DRAW_SIZE](https://docs.lvgl.io/master/overview/drawing.html#lv-event-refr-ext-draw-size) to the label and call `lv_obj_refresh_ext_draw_size(label);`.

> a other trouble:freetype maybe malloc memory in lv_font_t->get_glyph_dsc, but in lv_font_t->get_glyph_bitmap Can't release it(Because the returned bitmap pointer LVGL is used) .like this:

In the latest version of lv_freetype.c [get_glyph_bitmap_cb_cache](https://github.com/lvgl/lv_lib_freetype/blob/dc26547de741c6a147ad9193d82283337b0f6a79/lv_freetype.c#L251-L261) looks different. Are you using this version?

guoweilkd, 2021-09-22:
> In the latest version of lv_freetype.c [get_glyph_bitmap_cb_cache](https://github.com/lvgl/lv_lib_freetype/blob/dc26547de741c6a147ad9193d82283337b0f6a79/lv_freetype.c#L251-L261) looks different. Are you using this version?

The Bold code of Font is still experimental, so it is not pushed to the master

guoweilkd, 2021-09-23:
> Do you think it's necessary to add a release interface callback?

This  issue #2582  also mentions this need

guoweilkd, 2021-09-23:
> It can be managed internally by the label. But until it's implemented you can add an [LV_EVENT_REFR_EXT_DRAW_SIZE](https://docs.lvgl.io/master/overview/drawing.html#lv-event-refr-ext-draw-size) to the label and call `lv_obj_refresh_ext_draw_size(label);`.

hi, current label does not support ext_draw_size. I submitted a fix  #2587

kisvegabor, 2021-09-23:
> The Bold code of Font is still experimental, so it is not pushed to the master

Can you create a PR to see your code as a whole?

guoweilkd, 2021-09-24:
> Can you create a PR to see your code as a whole?
PR has been submitted. [add blod and italics](https://github.com/lvgl/lv_lib_freetype/pull/14). Blod is implemented in a other way, It is working normally now.

> It can be managed internally by the label. But until it's implemented you can add an [LV_EVENT_REFR_EXT_DRAW_SIZE](https://docs.lvgl.io/master/overview/drawing.html#lv-event-refr-ext-draw-size) to the label and call `lv_obj_refresh_ext_draw_size(label);`.

this way is enough for `label`. but for span,Italics may appear in the middle, like this:
[image]

Maybe we can do like this:
in font->get_glyph_dsc callback end add:
```
    if (dsc->style & FT_FONT_STYLE_ITALIC && unicode_letter_next == '\0') {
        dsc_out->adv_w = dsc_out->box_w + dsc_out->ofs_x;
    }
```

restult:
[image]

kisvegabor, 2021-09-24:
>     if (dsc->style & FT_FONT_STYLE_ITALIC && unicode_letter_next == '\0') {

Sounds good. Here we only care about word ending to be sure italic doesn't overlap with the next letter.

kisvegabor, 2021-10-16:
Not stale, I'll add a fix to lvgl.

closed 2021-10-18 by kisvegabor