lvgl/lvgl #2562: ESP32 Menuconfig LV_CONF_SKIP always YES

SinglWolf, 2021-09-17:
When starting Menuconfig, there is no selection line to disable LV_CONF_SKIP.
As a result, LV_CONF_SKIP is always YES.
[image]
Can fix it like this?
[image]

```
    config LV_CONF_MINIMAL
        bool "LVGL minimal configuration."

    config LV_CONF_SKIP
        bool "SKIP LV_CONF"
        default y
        help
            Define CONFIG_LV_CONF_SKIP so we can use LVGL
            without lv_conf.h file, the lv_conf_internal.h and
            lv_conf_kconfig.h files are used instead.

    menu "Color settings"
```

kisvegabor, 2021-09-17:
Basically it's a misformatted Kconfig option? 

cc @C47D

C47D, 2021-09-17:
When adding the `"SKIP LV_CONF"` string after the option type we expose the variable to be changed. Without it, the user doesn't see the option.

Having the description of it on the help section instead of a comment is much better tho. 

IIRC we were forcing the usage of Kconfig to use it with esp32.

kisvegabor, 2021-09-17:
> When adding the "SKIP LV_CONF" string after the option type (bool) we expose the variable to be changed. Without it, the user doesn't see the option.

Then why isn't it visible for @SinglWolf ?

C47D, 2021-09-17:
Because before changing it there wasn't any string after the symbol type.

kisvegabor, 2021-09-17:
Ah, got it now. Do you agree to fix it as @SinglWolf suggested?

C47D, 2021-09-17:
@SinglWolf do you plan to use `lv_conf.h` instead of Kconfig for LVGL configuration?. 

@kisvegabor 
If so we can test the suggestion before the change, let's remember the Kconfig file is used in other projects out of LVGL control, such as Nuttx and Zephyr

SinglWolf, 2021-09-17:
> @SinglWolf do you plan to use `lv_conf.h` instead of Kconfig for LVGL configuration?.

If not using Kconfig would shorten the recompilation time of the project when making changes to lv_conf.h, then definitely yes.

C47D, 2021-09-17:
I guess it would, but I haven't tried it tho.

kisvegabor, 2021-09-17:
> If so we can test the suggestion before the change, let's remember the Kconfig file is used in other projects out of LVGL control, such as Nuttx and Zephyr

If the users don't touch it shouldn't cause an trouble, right?

kisvegabor, 2021-10-04:
Working on it in #2555.

kisvegabor, 2021-10-19:
Will be fixed in #2555

closed 2021-10-19 by kisvegabor