lvgl/lvgl #2564: Calendar bug report

vahidajalluian, 2021-09-17:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
The bug is related to setting date in 2020 and more.
If one sets the date on 17 September 2021, it would see the set date on 17 January 2020.
Furthermore, if the selected month or the selected year gets a time distance from the previous set time, the calendar widget will not get the highlighting order.

### To Reproduce

```c
#include "../../lv_examples.h"
#if LV_USE_CALENDAR && LV_BUILD_EXAMPLES

static void event_handler(lv_event_t * e)
{
    lv_event_code_t code = lv_event_get_code(e);
    lv_obj_t * obj = lv_event_get_target(e);

    if(code == LV_EVENT_VALUE_CHANGED) {
        lv_calendar_date_t date;
        if(lv_calendar_get_pressed_date(obj, &date)) {
            static lv_calendar_date_t highlighted_days;
            highlighted_days.year = date.year;
            highlighted_days.month = date.month;
            highlighted_days.day = date.day;

            lv_calendar_set_highlighted_dates(obj, &highlighted_days, 1);
        }
    }
}

void lv_example_calendar_1(void)
{
    lv_obj_t  * calendar = lv_calendar_create(lv_scr_act());
    lv_obj_set_size(calendar, 350, 400);
    lv_obj_align(calendar, LV_ALIGN_CENTER, 0, 27);
    lv_obj_add_event_cb(calendar, event_handler, LV_EVENT_ALL, NULL);

    static lv_calendar_date_t today_day;
    today_day.year = 2021;
    today_day.month = 9;
    today_day.day = 17;
    lv_calendar_header_arrow_create(lv_scr_act(), calendar,30);
    lv_calendar_set_today_date(calendar, today_day.year, today_day.month, today_day.day);
    lv_calendar_set_showed_date(calendar, today_day.year, today_day.month);
}

#endif

```
-->

### Expected behavior

(1) Accepting the date from any date, no matter how far is the date from the previously set date.
(2) Accurately show the date

### Screenshots or video
Reproducing with the simulator on the CodeBlocks would show the effects.

### Suggestion
It seems the algorithms behind should be reworked.

kisvegabor, 2021-09-17:
It looks correct to me:
[image]
It's really how the current moth "looks like".

However the header really doesn't updated if the shown date is updated manually. Do you mean this?

vahidajalluian, 2021-09-17:
@kisvegabor On the second thought, it seems the problem is only on the header side.

C47D, 2021-09-20:
@kisvegabor With header do you mean the day names?

kisvegabor, 2021-09-20:
> On the second thought, it seems the problem is only on the header side.

Great! The problem is that now the calendar doesn't know about the header, as it's a different widget. I'm still thinking about how to manage it. 

> But being disabled to highlight a date while getting distance from present time still exists. The matter happens on microcontrollers on the first lunch. Which could be solved by setting flags.

Sorry, but I don't understand this. Can you show an example for this issue?

> With header do you mean the day names?

Nope, I mean this: 
[image]

A separate object above the calendar.

>  Is there any particular reason why there's no LV_ASSERT_OBJ check on the any of the calendar widget API?

Yes, I forgot it :slightly_smiling_face:

vahidajalluian, 2021-09-20:
@kisvegabor To clear the issue, suppose the set date to be shown is 17th of septemper 2021, now after that the screen is loaded, if one changes the year to 2020, earlier or later, and touches the date on the calendar widget, the touched date wont be highlighted with the code I'v  sent here. This doesn't happen on near monthes for up to 3 month previous or later.

kisvegabor, 2021-09-23:
I've tried it with this very minimal example: and it worked well for me:
```
    lv_obj_t  * calendar = lv_calendar_create(lv_scr_act());
    lv_calendar_set_showed_date(calendar, 2021, 02);
    lv_timer_handler();
    lv_calendar_set_showed_date(calendar, 2011, 02);
```

I pressed buttons are highlighted correctly. Does this simple code works for you?

vahidajalluian, 2021-09-24:
@kisvegabor 
No problem with 2021 February, but the problem exists on 2021 September, as of today. Here is minimal code to reproduce the problem.
```
static void calendar_event_handler(lv_event_t * e)
{
    lv_event_code_t code = lv_event_get_code(e);
    lv_obj_t * obj = lv_event_get_target(e);

    if(code == LV_EVENT_VALUE_CHANGED) {
        lv_calendar_date_t date;
        if(lv_calendar_get_pressed_date(obj, &date)) {
            lv_calendar_date_t highlighted_days[1];
            highlighted_days[0].year = date.year;
            highlighted_days[0].month = date.month;
            highlighted_days[0].day = date.day;

            lv_calendar_set_highlighted_dates(obj, highlighted_days, 1);
        }
    }
}
```
and in the main

    lv_obj_t * calendar = lv_calendar_create(lv_scr_act());
    lv_obj_add_event_cb(calendar, calendar_event_handler, LV_EVENT_ALL, NULL);
    lv_calendar_set_today_date(calendar, 2021, 9, 1);
    lv_calendar_set_showed_date(calendar, 2021,9);
    static lv_calendar_date_t highlighted_days; 
    highlighted_days.year = 2021;
    highlighted_days.month = 9;
    highlighted_days.day = 1;
    lv_calendar_set_highlighted_dates(calendar, &highlighted_days, 1);

kisvegabor, 2021-09-24:
In `get_day_of_week()` day should be in the range of `1..31`. 

But in the event `lv_calendar_date_t highlighted_days[1];` should be `static` because `lv_calendar_set_highlighted_dates` saves only a pointer.

I still can't reproduce the issue. With the original code in `lv_calendar.c` and your example it works well for me. :slightly_frowning_face: 
 
[image]

vahidajalluian, 2021-09-24:
@kisvegabor Sorry, It seems my repo was a bit old, comparing mine with the history, it is supposed to be the git of the date August 10th, the problem has been solved in the next update.

kisvegabor, 2021-09-24:
Glad to her that it's solved. 
Can we close this issue?

vahidajalluian, 2021-09-24:
Yes of course