lvgl/lvgl #2578: Added configuring external SDRAM

MaybenYi, 2021-09-22:
For STM32 microcontroller with part of small memory, it is hoped to increase the configuration of external SDRAM and store buffer in it, so that there can be space for operation

kisvegabor, 2021-09-22:
Which buffer do you mean?

MaybenYi, 2021-09-22:
> Which buffer do you mean?
such as in lv_port_disp_template.c，
The structure lv_color_t  and lv_disp_draw_buf_t  can be saved in an external SDRAM

embeddedt, 2021-09-22:
I think you can do that already, independently of LVGL's configuration, as those are variables you declare in the code yourself.

MaybenYi, 2021-09-22:
> I think you can do that already, independently of LVGL's configuration, as those are variables you declare in the code yourself.

that's true，but i think this is an optional solution for small memory devices, for beginners

closed 2021-09-22 by MaybenYi