lvgl/lvgl #2581: why is the one buffer is better than the two buffer

MaybenYi, 2021-09-22:
### Introduce the problem
On the STM32F429 development board, the one buffer is better than the two buffer

### Examples and cases
In lv_port_disp.c
better:
```
    static lv_color_t draw_buf_1[LV_HOR_RES_MAX * 10];    
    lv_disp_buf_init(&draw_buf_dsc_1, color_buf1, NULL, COLOR_BUF_SIZE);
```
poor:
```
    static lv_color_t buf_2_1[LV_HOR_RES_MAX * 10];
    static lv_color_t buf_2_2[LV_HOR_RES_MAX * 10];
    lv_disp_buf_init(&draw_buf_dsc_2, buf_2_1, buf_2_2, COLOR_BUF_SIZE); 
```

### Suggested solution
Hope this can be improved for small memory development boards

kisvegabor, 2021-09-22:
Please specify more exactly what do you mean by poor and better. It's also important how your flush_cb is implemented.

Videos about the version would help a lot!

MaybenYi, 2021-09-23:
sorry,i forget that

/*Benchmark your system*/
#define LV_USE_DEMO_BENCHMARK   1

the one buffer has an FPS of around 25
the two buffer has an FPS of around 12

kisvegabor, 2021-09-23:
Thanks. 

What about the flush_cb? And could you make a video?

The 2 buffer config really seems slower or just the indicator shows different value?

MaybenYi, 2021-09-27:
```
static lv_color_t color_buf1[COLOR_BUF_SIZE] __attribute__((section(".sram_data")));
static lv_color_t color_buf2[COLOR_BUF_SIZE] __attribute__((section(".sram_data")));
```

**the one buffer:**
`lv_disp_buf_init(&draw_buf_dsc_1, color_buf1, NULL, COLOR_BUF_SIZE);`
[image]

**the two buffer:**
`lv_disp_buf_init(&draw_buf_dsc_1, color_buf1, color_buf2, COLOR_BUF_SIZE);`
[image]

**flush_cb:**
`disp_drv.flush_cb = disp_flush;`
Disp_flush Uses DMA2D mode

closed 2021-10-20 by lvgl-bot