lvgl/lvgl #2582: Font engine using heap memory leak

pdgendt, 2021-09-22:
### Introduce the problem

I have a font engine that renders glyphs using dynamic heap memory. However there is no cleanup callback available in `lv_font_glyph_dsc_t`.

### Examples and cases

It is only relevant for use with a font rendering engine, where no font cache is available.

### Suggested solution

Add a `lv_font_clear_glyph_dsc` function and `clear_glyph_dsc` property to `lv_font_t` where a user optionally can cleanup resources.

guoweilkd, 2021-09-23:
`lv_font_get_glyph_dsc()` Used to get glyph infomation. It may not come in pairs with lv_font_get_glyph_bitmap() and lv_font_clear_glyph_dsc().  eg:[pwd_char_hider() in lv_textarea.c](https://github.com/lvgl/lvgl/blob/master/src/widgets/lv_textarea.c#L978)
the problem needs to be considered.

kisvegabor, 2021-09-23:
I don't get it. 
In `font->get_glyph_dsc()` you need to set data in a `lv_font_glyph_dsc_t *` passed as a pointer to the callback. So you can
1. Alloc memory, fetch/calculate the required glyph data  
2. Set the data in the `lv_font_glyph_dsc_t *` parameter
3. Clean up 

Or have I missed something?

pdgendt, 2021-09-23:
> Or have I missed something?

I think you are, in my case I fully render the glyphs at runtime, so I need to have a pointer ready when the `font->get_glyph_bitmap()` is called, so I can't clean up after `font->get_glyph_dsc()`.

As mentioned in the initial comment, it's without using some font cache.

Currently I use a static buffer for the last glyph which is filled when `font->get_glyph_dsc()` gets called.

kisvegabor, 2021-09-23:
So do you render the font even only `get_glyph_dsc` is called? If so, I highly discourage you to do so. `get_glyph_dsc` is called a lot of times mainly to get the glyph's width. Rendering on all calls would lead to much lower performance. 

So is there a way to decouple the handling of `dsc` and `bitmap` in your case?

pdgendt, 2021-09-23:
I've created [a gist](https://gist.github.com/pdgendt/530bdef53da4065c2b818b8fbcfea580) that shows how my font rendering is done, using a header only implementation from [nothings/stb](https://github.com/nothings/stb/blob/master/stb_truetype.h) (great library btw).

I store the info from the last requested glyph to serve as an early return, which helps a lot for performance, but suggestions on how to improve would be much appreciated.

kisvegabor, 2021-09-28:
@pdgendt 
By looking into the gist I see problems: one related to LVGL and an other in your implementation. 
1. `glyph_dsc_cb` most of the cases LVGL needs only `adv_w`. So LVGL should signal somehow that if `box_x/y/w/h` is not needed. E.g. set them to -1 or so. This way `glyph_dsc_cb` could return much earlier.
2. It just a coincidence that a `glyph_dsc_cb` is followed by `glyph_bitmap_cb` of the same glyph and font. So it needs to be checked in `glyph_bitmap_cb` you really return the correct bitmap.  

It's not the best solution, but as quick workaround you can use display driver's `monitor_cb` to clean up the resources.