lvgl/lvgl #2589: Calling lv_anim_timeline_del is deleting an active animation NOT attached to the deleted timeline

tvanfossen, 2021-09-23:
### Perform all steps below and tick them with [x]
- [ x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x ] Update lvgl to the latest version
- [x ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

I am working on a gui that has several actions on-going on the screen at any given time and I am using animation timelines to handle the stopping/starting of these updates cleanly as new data is brought onto the screen.

I have a function for View_OnDataChange that gets called every few seconds, or when there is an active update of data (ie. a button is pressed). Any necessary updates to this view are added to the data_change_timeline I am using.

If the data_change_timeline has been started on entry to this function, then I call lv_anim_timeline_stop and lv_anim_timeline_del before starting a fresh timeline with the relevant animations merged into the new timeline. This has worked great so far.

On the end of the data_change_timeline, I add an extra lv_anim_t that just points to callback to indicate when the timeline has been completed such that I can delete it (as otherwise there is a memory leak from lv_anim_timeline_create). When this callback is triggered, I delete the timeline for garbage cleanup. This again works great for the data driven animations.

Now to the actual bug. One of several views, I have a particular icon that I want to fade in/out repeatedly throughtout the life of the screen, but it is not tied to any given timeline. So, once the screen has been created, I start this particular animation with lv_anim_start, but it is never added to ANY timeline as it is meant to run repeatedly. This animation has a callback which then continuously kicks off fresh animations with the relevant fade in/out value (LV_OPA_TRANP -> LV_OPA_COVER and vice verse).

Problem is, this animation stops arbitrarily. If I remove the call to delete the data_change_timeline, the animation continues on as expected.

It would seem that lv_anim_timeline_del(at) is arbitrarily deleting animations that are not attached to the timeline at all, and have been started outside of the timeline.

Is this an intended behavior such that there can only be 1 active timeline, or no active timeline and any number of active animations? 

I have some ideas for alternative ways to perform this same behaviour, but this still seemed a very apparent bug to me that the animation timeline should stop a animation which has never been attached to it

### To Reproduce

```c
//Example of bug,

static void anim_set_arc_value_cb(void * var, int32_t v)
{
    lv_arc_set_value(var, v);
};

static void anim_set_img_opa_cb(void * var, int32_t v)
{
    lv_obj_set_style_img_opa(var, v, 0);
};

static void anim_fade_complete_cb(lv_anim_t* a)
{
    int current_opa = a->current_value;
    if (current_opa == LV_OPA_TRANSP)
    {
        lv_anim_t fade_in;
        lv_anim_init(&fade_in);
        lv_anim_set_var(&fade_in, a->var);
        lv_anim_set_exec_cb(&fade_in, anim_set_img_opa_cb);
        lv_anim_set_ready_cb(&fade_in, (lv_anim_ready_cb_t)anim_fade_complete_cb);
        lv_anim_set_path_cb(&fade_in, lv_anim_path_linear);
        lv_anim_set_time(&fade_in, 750); 
        lv_anim_set_values(&fade_in, LV_OPA_TRANSP, LV_OPA_COVER);
        lv_anim_start(&fade_in);
    }
    else if (current_opa == LV_OPA_COVER)
    {
        lv_anim_t fade_out; 
        lv_anim_init(&fade_out);
        lv_anim_set_var(&fade_out, a->var);
        lv_anim_set_exec_cb(&fade_out, anim_set_img_opa_cb);
        lv_anim_set_ready_cb(&fade_out, (lv_anim_ready_cb_t)anim_fade_complete_cb);
        lv_anim_set_path_cb(&fade_out, lv_anim_path_linear);
        lv_anim_set_time(&fade_out, 750); 
        lv_anim_set_values(&fade_out, LV_OPA_TRANSP, LV_OPA_COVER);
        lv_anim_start(&fade_out);
    }
};

static void data_change_complete_cb(lv_anim_t * a)
{
// With this line removed, fading animation continues as expected
// With it in place, fading stops when timeline is deleted
// If this timeline is not deleted after the data change has been completed, there is a memory leak
    lv_anim_timeline_del(anim_timeline);
};

void OnDataChange(lv_obj_t* arc, int new_arc_value)
{
    static lv_anim_timeline_t* anim_timeline = lv_anim_timeline_create();

    lv_anim_t a;
    lv_anim_init(&a);
    lv_anim_set_var(&a, arc);
    lv_anim_set_exec_cb(&a, (lv_anim_exec_xcb_t)anim_set_arc_value_cb);
    lv_anim_set_path_cb(&a, (lv_anim_path_cb_t)lv_anim_path_linear);
    lv_anim_set_time(&a, 500);
    lv_anim_set_values(&a, lv_arc_get_value(arc), new_arc_value);

    lv_anim_t complete;
    lv_anim_init(&complete);
    lv_anim_set_ready_cb(&complete, complete_cb);
    lv_anim_set_user_data(&complete, view);
    lv_anim_set_time(&complete, 1);
    lv_anim_set_values(&complete, 0, 1);

    lv_anim_timeline_add(anim_timeline, 0, &a);
    //Assume more animations are added to this timeline in normal use
    //this timeline is generated repeatedly where the values of arc animation are changing based on the data received
    lv_anim_timeline_add(anim_timeline, 500, &complete);

    /* Fade in/out animation init */
    lv_anim_t fade;
    lv_anim_init(&fade);
    lv_anim_set_var(&fade, obj);
    lv_anim_set_exec_cb(&fade, anim_set_img_opa_cb);
    lv_anim_set_ready_cb(&fade, (lv_anim_ready_cb_t)anim_fade_complete_cb);
    lv_anim_set_path_cb(&fade, lv_anim_path_linear);
    lv_anim_set_time(&fade, 750); 
    lv_anim_set_values(&fade, LV_OPA_COVER, LV_OPA_TRANSP);
    lv_anim_start(&fade);

    lv_anim_timeline_start(anim_timeline);
}

```

### Expected behavior

The animation and animation timeline, when not tied together, should not be effecting eachother. If I stop/delete an animation timeline it should only affect the animations tied to that timeline, not animations that are running outside of that timeline.

### Screenshots or video

kisvegabor, 2021-09-27:
This feature was added by @FASTSHIFT so I cc him. 

What I can tell is that I haven't seen such issue with the animations only (not timeline) recently. Do you use the latest version of lvgl?

I've tried to test you example but there are some unknown variables in it. Please update it to can be built in a simulator.

tvanfossen, 2021-09-27:
Missed the image object for the fade, sorry

```
extern const lv_img_dsc_t lightning_icon;
LV_IMG_DECLARE(lightning_icon);

static lv_anim_timeline_t* anim_timeline;
//Example of bug,

static void anim_set_arc_value_cb(void * var, int32_t v)
{
    lv_arc_set_value(var, v);
};

static void anim_set_img_opa_cb(void * var, int32_t v)
{
    lv_obj_set_style_img_opa(var, v, 0);
};

static void anim_fade_complete_cb(lv_anim_t* a)
{
    int current_opa = a->current_value;
    if (current_opa == LV_OPA_TRANSP)
    {
        lv_anim_t fade_in;
        lv_anim_init(&fade_in);
        lv_anim_set_var(&fade_in, a->var);
        lv_anim_set_exec_cb(&fade_in, anim_set_img_opa_cb);
        lv_anim_set_ready_cb(&fade_in, (lv_anim_ready_cb_t)anim_fade_complete_cb);
        lv_anim_set_path_cb(&fade_in, lv_anim_path_linear);
        lv_anim_set_time(&fade_in, 750); 
        lv_anim_set_values(&fade_in, LV_OPA_TRANSP, LV_OPA_COVER);
        lv_anim_start(&fade_in);
    }
    else if (current_opa == LV_OPA_COVER)
    {
        lv_anim_t fade_out; 
        lv_anim_init(&fade_out);
        lv_anim_set_var(&fade_out, a->var);
        lv_anim_set_exec_cb(&fade_out, anim_set_img_opa_cb);
        lv_anim_set_ready_cb(&fade_out, (lv_anim_ready_cb_t)anim_fade_complete_cb);
        lv_anim_set_path_cb(&fade_out, lv_anim_path_linear);
        lv_anim_set_time(&fade_out, 750); 
        lv_anim_set_values(&fade_out, LV_OPA_TRANSP, LV_OPA_COVER);
        lv_anim_start(&fade_out);
    }
};

static void data_change_complete_cb(lv_anim_t * a)
{
// With this line removed, fading animation continues as expected
// With it in place, fading stops when timeline is deleted
// If this timeline is not deleted after the data change has been completed, there is a memory leak
    lv_anim_timeline_del(anim_timeline);
};

static void OnDataChange(lv_obj_t* arc, lv_obj_t* obj, int new_arc_value)
{
    anim_timeline = lv_anim_timeline_create();

    lv_anim_t a;
    lv_anim_init(&a);
    lv_anim_set_var(&a, arc);
    lv_anim_set_exec_cb(&a, (lv_anim_exec_xcb_t)anim_set_arc_value_cb);
    lv_anim_set_path_cb(&a, (lv_anim_path_cb_t)lv_anim_path_linear);
    lv_anim_set_time(&a, 500);
    lv_anim_set_values(&a, lv_arc_get_value(arc), new_arc_value);

    lv_anim_t complete;
    lv_anim_init(&complete);
    lv_anim_set_ready_cb(&complete, data_change_complete_cb);
    lv_anim_set_time(&complete, 1);
    lv_anim_set_values(&complete, 0, 1);

    lv_anim_timeline_add(anim_timeline, 0, &a);
    //Assume more animations are added to this timeline in normal use
    //this timeline is generated repeatedly where the values of arc animation are changing based on the data received
    lv_anim_timeline_add(anim_timeline, 500, &complete);

    /* Fade in/out animation init */
    lv_anim_t fade;
    lv_anim_init(&fade);
    lv_anim_set_var(&fade, obj);
    lv_anim_set_exec_cb(&fade, anim_set_img_opa_cb);
    lv_anim_set_ready_cb(&fade, (lv_anim_ready_cb_t)anim_fade_complete_cb);
    lv_anim_set_path_cb(&fade, lv_anim_path_linear);
    lv_anim_set_time(&fade, 750); 
    lv_anim_set_values(&fade, LV_OPA_COVER, LV_OPA_TRANSP);
    lv_anim_start(&fade);

    lv_anim_timeline_start(anim_timeline);
};

static void lv_testing(){
  lv_obj_t* screen = lv_obj_create(NULL);
  lv_obj_set_width(screen, 240);
  lv_obj_set_height(screen, 320);

  lv_scr_load_anim(screen, LV_SCR_LOAD_ANIM_NONE, 0, 0, true);
  
  lv_obj_t * arc = lv_arc_create(screen);
  lv_obj_set_size(arc, 100, 100);
  lv_arc_set_value(arc, 0);
  lv_obj_align(arc, LV_ALIGN_CENTER, 0, 0);

  lv_obj_t* lightning_img = lv_img_create(screen);
  lv_img_set_src(lightning_img, &lightning_icon); 
  lv_obj_align(lightning_img, LV_ALIGN_TOP_MID, 0, 0);
  
  OnDataChange(arc, lightning_img, 100);
};
```

lv_testing called in main() in a fresh clone of lvgl simulator with lv_conf.h moved to ui/simulator/inc. LVGL version updated to latest on master in clone.

Intent: The image animation should continue indefinitely, with a fade in/out logic (logic in cb in slightly off but doesnt matter for this conversation). The animation is started outside the timeline and *should* be unaffected by the timelines deletion.

 This is the outcome when the animation timeline is deleted (as part of its own callback), where the fade animation stops
[image]

This is the outcome when the animation timeline is not deleted (lv_anim_timeline_del() is commented out), where the fade animation continues indefinitely
[image]

The icon is under main converted w/ the online LVGL converter, any image should suffice in its place.

The timeline MUST be deleted after its use as it is malloced and will be the cause of a memory leak.

From 0-500ms, fade has been started and arc value is changing from 0-100. At 501ms, the data change complete callback is hit and the timeline is deleted for cleanup, which is also deleting the fade animation, or stopping it at least, when the fade animation was never tied to the timeline

tvanfossen, 2021-09-27:
Little more background:

I have a view controller which is periodically updating data (battery level, motor settings etc). Sometimes, these changes can occur very close to one another that a previous animation could still be running, especially when accounting for user behavior.

In order to ensure clean animations/transitions, I need to be able to stop the active animation timeline, update it with the new data that has come in, and start it again relative to the position that every animation ended in when it was paused. At the moment, this involves the creation of a brand new timeline as stopping a timeline is equivalent to deleting it (in that it deletes the animations attached to it)

I.e. I have 3 arcs, arcs 1 and 2 are animating and new data is received for arc 3. I stop the animations for arcs 1/2 at their current position, delete the timeline, create a new timeline, then generate new animations relative to the current position and kick it off again. This only looks clean with lv_anim_path_linear as the hard stops/starts are not suitable for ease_in/out/overshoot calls.

In a perfect world, I could pause an animation timeline instead of hard stopping it, such that I could add a new animation to the timeline (relative to the current playtime) before resuming it. This would save a substantial number of operations imo.

FASTSHIFT, 2021-09-28:
@kisvegabor 
Do you have plans to add `pause` and `resume` functions to `lv_anim`.

FASTSHIFT, 2021-09-28:
@tvanfossen 
I don’t understand the effect you want, can you show me a complete effect with only `lv_anim`?

tvanfossen, 2021-09-28:
This should be demonstrated above? There is no issue if I run the animations as regular lv_anim, but lv_anims do not provide the desired control..

I have two animations actually performing a change on the screen. One to the arc, one to the image. A third is created to act as a callback for when the animation timeline has completed as the timeline itself does not possess this functionality.

The image animation should be performing an endless fade in/fade out loop. This is started outside of the animation timeline and is never attached to the timeline and thus should, in my mind, continue indefinitely, as the animation timeline should have no knowledge of it. 750ms to fade out, then 750ms to fade in, and looping

The arc animation should run one time when the animation timeline is started. This performs once in both gifs.

At 500ms, when the 3rd animation is called to trigger a callback in order to perform cleanup by calling lv_anim_timeline_del(). This is stopping the image animation in place, where the fade in/out animation stops at 500ms instead of continuing indefinitely. 

When lv_anim_timeline_del is commented out, we get gif #2 with an endless fading loop, when its commented in we get gif #1 where it stops in place before its even completed a full fade out.

FASTSHIFT, 2021-09-28:
@tvanfossen 
I seem to have found the problem. 
`lv_anim_timeline_del` will traverse the registered anim list and call `lv_anim_custom_del` to delete the animation. 
The following is the function implementation of `lv_anim_custom_del`:
[image]

It can be seen that when traversing to the registered `complete` animation, the `var` and `exec_cb` of the animation are both **`NULL`**, , resulting in the last execution of `lv_anim_del(NULL, NULL).` The operation of lvgl on this is to delete **`all the animations,`** causing the `fade` animation to be stopped.

FASTSHIFT, 2021-09-28:
@tvanfossen 
The current solution is to assign a `var` to the `complete` animation, so as to prevent `lv_anim_del(NULL, NULL)` from being called.
[image]

@kisvegabor 
Is there any better solution? This usage seems a bit strange.

tvanfossen, 2021-09-28:
Thanks! This resolved the issue, I can just set a null object that remains unused for the time being.

kisvegabor, 2021-10-04:
> Do you have plans to add pause and resume functions to lv_anim.

It seems doable, probably just a `paused` flag in `lv_anim_t`. However, there is an ambiguity here. `lv_anim_start` allocates a new `lv_anim_t` and adds it into a linked list. This new anim is returned from `lv_anim_start`, so `lv_anim_pause/resume` needs to be called on the returned anim not on the one that was initialized by the user.

> Is there any better solution?

I'll comment in the PR.

kisvegabor, 2021-10-19:
Should be solved by #2628

closed 2021-10-19 by kisvegabor