lvgl/lvgl #2600: button in scroll page Released.

cliffcloudlong, 2021-09-25:
### Perform all steps below and tick them with [x]
- [ ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ ] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
lv_demo_widgets.c : The button(log out) is not released when the button slides up or down。

### To Reproduce

### Expected behavior

### Screenshots or video

cliffcloudlong, 2021-09-28:
> Why is it a problem if it is kept pressed during slide?

For example, I will add button release event and add handlers (such as jump page), in the sliding page similar to the mobile phone APP icon, slide the page will inevitably press the APP button slide, slide finger release may cause APP button trigger release event, causing the page to jump, but I do not want to jump, just want to slide the page.

kisvegabor, 2021-09-30:
You can use `LV_EVENT_CLICKED`. It isn't sent if there was scrolling. 

IMO `LV_EVENT_RELEASED` is useful this way, as the object was really released.

closed 2021-10-19 by cliffcloudlong