lvgl/lvgl #2607: Discrete slider

Miguel0101, 2021-09-27:
Does LVGL have plans to add discrete slider?

[image]

embeddedt, 2021-09-27:
It works that way already if you set a small enough range compared to the width of the slider. Did you try it? Here is an example in [MicroPython](https://sim.lvgl.io/v8.1/micropython/ports/javascript/index.html?script_direct=a4d2a86703f0f2b74e25db001a4e01052e6c978b).

Miguel0101, 2021-09-27:
> It works that way already if you set a small enough range compared to the width of the slider. Did you try it? Here is an example in [MicroPython](https://sim.lvgl.io/v8.1/micropython/ports/javascript/index.html?script_direct=a4d2a86703f0f2b74e25db001a4e01052e6c978b).

But there is no way to set the step size, for example `lv_slider_set_step(slider, 10)`

embeddedt, 2021-09-27:
You can implement that independently of LVGL by multiplying the slider's value. For example, by setting the slider's range from 0-10, but multiplying the output by 20, you get a range from 0-200, but in 10 increments of 20. Here is the [modified example](https://sim.lvgl.io/v8.1/micropython/ports/javascript/index.html?script_direct=84d4a0867be9524836924672e7534da99ab17d00).

Miguel0101, 2021-09-27:
> You can implement that independently of LVGL by multiplying the slider's value. For example, by setting the slider's range from 0-10, but multiplying the output by 20, you get a range from 0-200, [...]

Is it possible to modify this without multiplying the output value?

embeddedt, 2021-09-27:
I don't think there is any other way of doing it with the current implementation of the slider. I also don't see this feature being added to the slider directly, as it is already a very complex widget for what it does. Do you see a reason why the suggested approach is less flexible?

Miguel0101, 2021-09-27:
> I don't think there is any other way of doing it with the current implementation of the slider. I also (personally) don't see this feature being added to the slider directly, as it is already a very complex widget for what it does. Do you see a reason why the suggested approach is less flexible?

We could simply add the function `lv_slider_set_step(obj, value)`.

Miguel0101, 2021-09-27:
When I manage to develop the `lv_slider_set_step` function, I will open a pull.

closed 2021-09-27 by Miguel0101

kisvegabor, 2021-09-27:
I agree with @embeddedt. We should keep LVGL's API as simple as possible (it's already too large) and this feature can be easily added on top of LVGL by the user.

So I don't think this feature is really required.  Instead an example seems more appropriate.