lvgl/lvgl #2613: v8.0 add encoder group cause esp32-s2 reboot

shuiyihang, 2021-09-28:
hey,I use lvgl release/v8.0 as a component in esp-idf, and add an encoder driver to main, but as long as the following line is added, my esp32-s2 will always restart, and the error is reported as follows, I am really Don't know what happened to him

```c
    lv_group_t *g = lv_group_create();

    lv_indev_drv_init(&indev_drv);
    indev_drv.type = LV_INDEV_TYPE_ENCODER;
    indev_drv.read_cb = encoder_read;
    lv_indev_t *enc_indev = lv_indev_drv_register(&indev_drv);

    lv_group_add_obj(g,roller1);

    lv_indev_set_group(enc_indev, g);
```
[https://github.com/shuiyihang/cmake_learn/blob/master/1.png](url)

[https://github.com/shuiyihang/cmake_learn/blob/master/2.png](url)

shuiyihang, 2021-09-28:
When I used lvgl7, everything was normal, but now, after changing some local variables to static, the controls can finally be displayed normally, but as long as the encoder driver is added, problems will occur

kisvegabor, 2021-09-29:
Note that, `lv_indev_drv_t` (and other drivers too) needs to be static in v8.

shuiyihang, 2021-09-29:
en....,i think ,this reason ,i had attempt。I still fall back to release/v7。hhh

shuiyihang, 2021-09-29:
the problems seems occur at lv_label.c 269 line,I may not look at it completely, but I feel that this type of forced conversion is more dangerous，may be text is null

shuiyihang, 2021-09-29:
[image]
[image]

kisvegabor, 2021-09-29:
If you pass valid object to `lv_label_get_text()` it should work.

Does it work with if you create only a single button?

shuiyihang, 2021-09-29:
solve it tomorrow

shuiyihang, 2021-09-30:
When I open the log of lvgl and set print the log with'printf' y. Select trace-detailed information, this problem will also appear in lvgl7. When I turn off the log and use lvgl8, it is now normal. It may not be the problem of lvgl,
In short, it's amazing~

shuiyihang, 2021-09-30:
I closed the question