lvgl/lvgl #2614: Optimization: make `lv_meter_indicator_type_t` as `uint8_t` instead of `int`

s-hadinger, 2021-09-28:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
Not a feature bug but an proposed optimization.

In `lv_meter.h`, `lv_meter_indicator_type_t` is defined as an enum without an exlicit type, which is automatically promoted to int (at least 32 bits).

https://github.com/lvgl/lvgl/blob/d57eb7614dcfd6d72fb6bae6b28bf52f843897ca/src/extra/widgets/meter/lv_meter.h#L54-L59

I suggest to explicitly define `lv_meter_indicator_type_t` as `uint8_t` and save 4 bytes in `lv_meter_indicator_t`

### To Reproduce
No actual bug

### Expected behavior
`lv_meter_indicator_t` of explicit type `uint8_t`

### Screenshots or video
n/a

kisvegabor, 2021-09-29:
Sounds reasonable. Could send a pull request?

s-hadinger, 2021-09-29:
Yes will do