lvgl/lvgl #2638: V8.XX(master). The built-in font with UTF-8 characters does not display UTF-8 characters by default. Bug or feature?

SinglWolf, 2021-10-05:
I am migrating my project from version 7 to the current one.
Stuck with UТF-8 characters not showing in embedded font. Naturally, the UТF-8 symbols have been re-added with lv_font_conv for version 8 again.
Only ASCII characters were displayed.
In one example, I found line 
```c
lv_obj_set_style_text_font(lmy_label, &lv_font_montserrat_16, 0);
```
and decided to apply it in my project.
UТF-8 symbols appeared on the screen.
Do I now have to apply this for every object? After all, in version 7 this was not.

kisvegabor, 2021-10-05:
Please send a short code snippet to reproduce the issue.

SinglWolf, 2021-10-05:
> Please send a short code snippet to reproduce the issue.

I would love to, but I don't understand English well.
In my project on version 7, the Russian font was displayed without  font declaration. Why is it necessary to declare forcibly on version 8?
Is that clearer? Sorry.

kisvegabor, 2021-10-05:
Hm, there was no Russian font by default in v7 either. I'm sure that you have added it somewhere manually in v7 too.

SinglWolf, 2021-10-05:
I say it again. I have implemented UTF-8 symbols (Russian font, symbols of European languages, my own icons) into the built-in montserrat_16 font using the lv_font_conv script and in version 7 I did not declare anything. Everything was displayed correctly. Why in version 8 the symbols of UTF-8 (Russian font, symbols of European languages, their own icons) are displayed only after a forced declaration?
I hope it's clear now?

embeddedt, 2021-10-05:
You had to have replaced the built-in font in v7, then, and you would need to do the same thing again in v8. As far as I know our built-in Montserrat font doesn't have those symbols by default.

kisvegabor, 2021-10-05:
Probably you set `LV_THEME_DEFAULT_FONT_NORMAL` to your font. In v8 `LV_FONT_DEFAULT` has similar purpose.

SinglWolf, 2021-10-05:
> You had to have replaced the built-in font in v7, then, and you would need to do the same thing again in v8. As far as I know our built-in Montserrat font doesn't have those symbols by default.

Of course I did it, I'm not that stupid!

> Probably you set `LV_THEME_DEFAULT_FONT_NORMAL` to your font. In v8 `LV_FONT_DEFAULT` has similar purpose.

I haven't changed anything in the settings. I will look again.

embeddedt, 2021-10-05:
Sorry; that wasn't what I was trying to convey, I was just trying to understand what you've tried so far and what the state of your v7 project is.

SinglWolf, 2021-10-05:
> Sorry; that wasn't what I was trying to convey, I was just trying to understand what you've tried so far and what the state of your v7 project is.

On version 7, everything worked fine for me, only at the edges of the objects there were incomprehensible artifacts in the form of intermittent vertical lines consisting of single pixels. I guess this is because of the display driver.

SinglWolf, 2021-10-19:
> As we mentioned you need to set `LV_FONT_DEFAULT` too.

I'm probably so dumb or blind from old age that I can not find where it is setting. Poke my face in this, please.

SinglWolf, 2021-10-19:
> Just as an experiment what if you remove `defconfig` and generate it again with `menuconfig`

A very useful experiment!
Can you imagine how much I then need to remember the necessary settings for the entire project?
But thanks anyway.

closed 2021-10-19 by SinglWolf

kisvegabor, 2021-10-19:
You should check from where `LV_FONT_DEFAULT_NORMAL_MONTSERRAT_16` is coming from. Certainly not from the current Kconfig file.