lvgl/lvgl #2677: JavaScript simulator problem on localhost

leidto, 2021-10-16:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
I have problem with the simulator itself. I did the complete build process but the simulator doesn't start properly on localhost.
I tried both of Linux and Windows but the error was the same.
I gave this error:

```
Traceback (most recent call last):
  File "<stdin>", line 4, in <module>
  File "imp.py", line 52, in importer
ImportError: module not found

MicroPython v1.17-771-g42009b6ee-dirty on 2021-10-16; JS with Emscripten
Type "help()" for more information.
```

### To Reproduce
I started with this note: (lvgl_javascript_v8 branch)
https://github.com/lvgl/lv_micropython/tree/lvgl_javascript_v8#javascript-port
The build was successfull.

This line is problematic:
_Run an HTTP server that serves files from the current directory, and then browse to /lvgl_editor.html on the HTTP Server._

I run a http server from the directory _(localhost/lvgl_editor.html)_, which redirects to _http://localhost/javascript/bundle_out/index.html_ but I give the error below.

Note: There were missing files. For example, I had to copy wasm_file_api.js from folder docs to bundle_out.

I'm a JavaScript developer. I'd like to run the dev server. I also did the following:
1. npm i
2. npm run dev
3. http-server ./ 
4. On browser: _localhost/lvgl_editor.html_

But it didn't work.

### Expected behavior
The simulator appears correctly. You can also edit the CSS and the dev server compiles without error.

### Screenshots or video
[image]

leidto, 2021-10-17:
@embeddedt  Thanks for your reply.

The `npm run clean` command deletes all files in folder bundle_out.
Previously I ran the `make` command which generated the wasm files in the bundle_out also.
I run the live-server, but these files are missing. (wasm and lvgl_mp.js)
So I copied back the files into the bundle_out folder and now it's working.

It was quite tricky.

Please let me know if you resolved the dev functionality.

leidto, 2021-11-03:
@embeddedt  Did you find the problem while running the developer server?

embeddedt, 2021-11-03:
@leidto Unfortunately I haven't had time to look into this yet. I've pinned the issue so it won't be closed automatically.

embeddedt, 2021-11-05:
This is (mostly) fixed now. Dev mode works, but it still won't automatically copy `lvgl.html` to `bundle_out` if you change it. I will need to look into that some other time, so I'll leave the issue open.

I've also updated the instructions about building the port to reflect its current state. [This](https://github.com/lvgl/lv_micropython/tree/lvgl_javascript_v8#hot-reloading-for-frontend-devs) is the new section on using the dev server.

Hope that helps!

Also, a heads-up: you will need to follow steps 5-7 of the setup process if you pull `lv_micropython` now, as I'm working on integrating FreeType & Rlottie, and the Makefile now expects them to be present.

leidto, 2022-01-20:
Thank you @embeddedt! The hot reloading works like a charm! 🚀

kisvegabor, 2022-01-24:
@leidto 
Feel free to close this issue if all issues are solved.

leidto, 2022-01-25:
Dev mode works fine but it doesn't copy the lvgl.html file automatically. This must be done before closing.

embeddedt, 2022-02-02:
I've switched to using Webpack as the bundler instead of Parcel. Webpack now tracks the `lvgl.html` as well as the WASM files so the manual copying is not needed. It also runs its own HTML server so `npm run serve` is not needed anymore either.

The instructions have been updated, but as a summary: `npm run dev` starts the development server, and `npm run bundle` creates the final production version that can be served independently. Those are the only two commands you should need.

Of course, make sure to run `npm install` after pulling the changes. :wink:

closed 2022-02-02 by embeddedt