lvgl/lvgl #2683: Kconfig CONFIG_LV_COLOR_CHROMA_KEY_HEX is different from CONFIG_LV_COLOR_CHROMA_KEY in internal config

steprun, 2021-10-18:
### Perform all steps below and tick them with [x]
- [ ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ ] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### A clear and concise description of what the bug is
Kconfig CONFIG_LV_COLOR_CHROMA_KEY_HEX is different from CONFIG_LV_COLOR_CHROMA_KEY in internal config

### To Reproduce
in file Kconfig and lv_internal_config.h

### Expected behavior
Change one of name in these file

### Screenshots or video

steprun, 2021-10-18:
But if use menuconfig, it produce this entry named CONFIG_LV_COLOR_CHROMA_KEY_HEX

xiaoxiang781216, 2021-10-18:
1. https://github.com/lvgl/lvgl/blob/master/Kconfig#L51-L56 define CONFIG_LV_COLOR_CHROMA_KEY_HEX to 0x????????
```
        config LV_COLOR_CHROMA_KEY_HEX
            hex "Images pixels with this color will not be drawn (if they are chroma keyed)."
            range 0x000000 0xFFFFFF
            default 0x00FF00
            help
                See misc/lv_color.h for some color values examples.
```
2. https://github.com/lvgl/lvgl/blob/master/src/lv_conf_kconfig.h#L29-L31 define CONFIG_LV_COLOR_CHROMA_KEY to lv_color_hex(CONFIG_LV_COLOR_CHROMA_KEY_HEX):
```
#ifdef CONFIG_LV_COLOR_CHROMA_KEY_HEX
#  define CONFIG_LV_COLOR_CHROMA_KEY lv_color_hex(CONFIG_LV_COLOR_CHROMA_KEY_HEX)
#endif
```
3. https://github.com/lvgl/lvgl/blob/master/src/lv_conf_internal.h#L85-L92 define LV_COLOR_CHROMA_KEY to LV_COLOR_CHROMA_KEY:
```
/*Images pixels with this color will not be drawn if they are chroma keyed)*/
#ifndef LV_COLOR_CHROMA_KEY
#  ifdef CONFIG_LV_COLOR_CHROMA_KEY
#    define LV_COLOR_CHROMA_KEY CONFIG_LV_COLOR_CHROMA_KEY
#  else
#    define LV_COLOR_CHROMA_KEY lv_color_hex(0x00ff00)         /*pure green*/
#  endif
#endif
```
I don't see any problem in these steps.

closed 2021-10-18 by steprun