lvgl/lvgl #2751: Fork lv_apps/lv_tpcal.c at version 7.XX to version 8.XX

SinglWolf, 2021-10-31:
### Introduce the problem

I was able to run [lv_apps/lv_tpcal.c](https://github.com/lvgl/lv_apps/blob/v7-test/src/lv_tpcal/lv_tpcal.c) on `LVGL` version `8.XX`.
With the first series of touches with the stylus on the first circle in the upper left corner, everything was fine.
I will say right away that the touchscreen driver was configured to output only raw data.
```c
void xpt2046_read_raw(int16_t *x, int16_t *y)
{
    (*x) = (xpt2046_cmd(CMD_X_READ) >> 4);
    (*y) = (xpt2046_cmd(CMD_Y_READ) >> 4);
}
```
My are prompted to touch the circle in the upper right corner of the display.
I couldn't go further.
Despite the fact that my screen and button size is set to `5000x5000` pixels,
```c
#define TP_MAX_VALUE     5000
//
lv_obj_set_size(scr, TP_MAX_VALUE, TP_MAX_VALUE);
//
lv_obj_set_size(big_btn, TP_MAX_VALUE, TP_MAX_VALUE);
//
```

the `btn_event_cb` callback function on the click of the button did not return any events.
```c
static void btn_event_cb(llv_event_t *e)
{
////

////
}
```
Investigation of the problem showed that the `btn_event_cb` callback function responds to a click only when it receives data from the touchscreen driver that does not exceed the real coordinates of the display.
If `x > 480` and `y > 320`, callback `btn_event_cb` does not fire.
When trying to increase the display resolution on initialization, it freezes my `ESP32-S2`.
```c
	disp_drv.hor_res = 5000;
	disp_drv.ver_res = 5000;
```
This is the first problem. Until I find a solution, maybe I can work around it by replacing the callback function with direct reading data from the display driver.

After a while, I still get all the data (by directly reading the raw coordinates from the touchscreen driver).
```
W (123572) TPCAL: Point 1 (178,226)
W (123572) TPCAL: Point 2 (1902,202)
W (123572) TPCAL: Point 3 (1935,1882)
W (123572) TPCAL: Point 4 (116,1792)
```
Several questions arose.
How to use this data in the [touchscreen driver](https://github.com/lvgl/lvgl_esp32_drivers/blob/master/lvgl_touch/xpt2046.c)?
What should be the orientation of the display when using `lv_apps/lv_tpcal.c`?
When to apply the `CIRCLE_SIZE/2 + CIRCLE_OFFSET` correction. When reading raw data from the touchscreen driver or already to the output data?
Please help find answers to the questions posed.

kisvegabor, 2021-11-05:
It's not working in v8 because LVGL clips the too large coordinates to the screen's size.

> How to use this data in the touchscreen driver?

https://github.com/lvgl/lv_apps/blob/v7-test/src/lv_tpcal/lv_tpcal.c#L17

> What should be the orientation of the display when using lv_apps/lv_tpcal.c?

You need to be sure the X and Y direction matches the displays X and Y direction.

> When to apply the CIRCLE_SIZE/2 + CIRCLE_OFFSET correction. When reading raw data from the touchscreen driver or already to the output data?

As circles' position and size set in "display coordinate system" and not "raw touch coordinate system" you need to apply this offset when you are already in  "display coordinate system" (i.e. when the data are converted)

SinglWolf, 2021-11-05:
> As circles' position and size set in "display coordinate system" and not "raw touch coordinate system" you need to apply this offset when you are already in "display coordinate system" (i.e. when the data are converted)

I needed a practical formula. In theory, I was thinking exactly like You.
But it doesn't matter anymore.
While waiting for Your answer, I did not sit idle.
I found a system for calibrating the touchscreen for three points of the display, which does not depend on the orientation of the display.
With all the formulas and examples.
Thanks for the answer.

closed 2021-11-05 by SinglWolf