lvgl/lvgl #2756: Pixelated Graphics Displayed With Linux Frame Buffer

napperley, 2021-11-02:
### Perform all steps below and tick them with [x]
- [X] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [X] Update lvgl to the latest version
- [] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html) **NOT APPLICABLE!**

Note that the PC simulator isn't designed to replicate an **Embedded Linux** environment where the **Linux Frame Buffer** backend is being used. Use a **real** Embedded Linux device like the Raspberry Pi 4 running Raspberry Pi OS (the minimal version with NO Windowing System) as the environment.

### Describe the bug

When running the program it displays some graphics but all the graphics displayed on screen (via the Linux Frame Buffer) look very pixelated (has an EGA resolution look), the displayed text isn't very legible, and the official Material theme isn't being applied.

### To Reproduce

1. Clone the project via Git: `git clone https://gitlab.com/napperley/lvgl-hello`
2. Download [LVGL 8.0.2](https://github.com/lvgl/lvgl/archive/refs/tags/v8.0.2.tar.gz) and extract it to **$HOME/lvgl**
3. Clone [LV Drivers](https://github.com/lvgl/lv_drivers) via Git to **$HOME/lv_drivers/lv_drivers-6.1.1**
4. Install JDK 11 (**required** for the Gradle build system)
5. Change working directory to the project directory
6. Build the debug binary by running the following: `./gradlew linkLvgl_helloDebugExecutableLinuxArm32Hfp`
7. Using SFTP copy the debug binary from `build/bin/linuxArm32Hfp/lvgl_helloDebugExecutable` to the home directory on the Embedded Linux device (eg Raspberry Pi 4)
8. On the Embedded Linux device run the program: `./lvgl_hello.kexe`

### Expected behaviour

All graphics are shown without extensive pixelation, the text is legible, and the official Material theme is applied.

### Screenshots or video

Since the Linux Frame Buffer is being used as the backend it isn't possible to do a screenshot.

kisvegabor, 2021-11-02:
> Note that the PC simulator isn't designed to replicate an Embedded Linux environment where the Linux Frame Buffer backend is being used. Use a real Embedded Linux device like the Raspberry Pi 4 running Raspberry Pi OS (the minimal version with NO Windowing System) as the environment.

Maybe it's only my ignorance, but what the difference? LVGL renders a pixel perfect image that needs to be shown on the display. It's SW rendered, so it's highly platform independent. Unless there is a special filtering/post processing in the frame buffer driver (which I consider unlikely) the rendered image should look exactly the some on all the devices (not considering the resolution, DPI, color depth etc differences of course).

> Clone LV Drivers via Git to $HOME/lv_drivers/lv_drivers-6.1.1

Is `-6.1.1` the version numbers? Why don't you use the latest master with LVGL 8? 

 > Since the Linux Frame Buffer is being used as the backend it isn't possible to do a screenshot.

Please take a photo about it. Maybe it's just a simple config issue. 

Have you set `LV_COLOR_DEPTH` correctly? If it's different from the frame buffer's color depth, the driver needs to convert the color format which might result in slight loss in quality.

I've already used LVGL with Linux frame buffer in a couple of projects, but the quality was the same as in the simulator.

embeddedt, 2021-11-02:
> Have you set LV_COLOR_DEPTH correctly? If it's different from the frame buffer's color depth, the driver needs to convert the color format which might result in slight loss in quality.

I have frequently had issues when relying on the driver converting the color format, so it's important to ensure that these values match. The hardware BPP should be printed to the console when the application first starts.

Once configured correctly there should be almost no difference between the frame buffer and simulator backends as @kisvegabor stated.

> Note that the PC simulator isn't designed to replicate an Embedded Linux environment where the Linux Frame Buffer backend is being used.

The main reason for the "Reproduce in a simulator" step is that we otherwise receive a lot of rendering-related issue reports that end up being a platform-specific driver issue. It encourages users to confirm that their issue actually occurs in a reproducible environment and not just on their specific hardware. This does not really apply for your issue.

napperley, 2021-11-03:
The colour depth of the target device (Raspberry Pi 4) is currently set to *16* (is the default) after running `fbset`. Unfortunately the video driver doesn't provide a **real** Frame Buffer (only an emulated one which can cause major issues), and as such the colour depth cannot be set to *32* (video driver limitation). Because of this **LV_COLOR_DEPTH** has been set to *16*, and the static library for LVGL has been regenerated. After rerunning the linking task for the Kotlin Native program, and running the program again on the target device the graphics are no longer pixelated (the results are similar to what a user would expect), although the scaling is a bit off even though the following macros have been set (using the **1280x720** resolution with a **16:9** screen ratio that matches the Computer Monitor):

- LV_VERT_RES
- LV_HOR_RES
- WINDOW_HOR_RES
- WINDOW_VER_RES

The bottom part of the button text is chopped off, and the white background only covers half of the screen with the bottom half being black. Wondering if it has anything to do with the screen resolution.

napperley, 2021-11-05:
Also noticing graphics corruption occurring when the mouse cursor moves on screen. The effect looks similar to what is reported in this [forum thread](https://forum.lvgl.io/t/corrupted-graphics-when-activating-dma2d-on-a-sm32f750/4415).

kisvegabor, 2021-11-05:
> After rerunning the linking task for the Kotlin Native program

Just out of curiosity, how can you call C functions from Kotlin? Have you generated a Kotlin-LVGL binding or somehow it just works with Kotlin?

> The bottom part of the button text is chopped off, and the white background only covers half of the screen with the bottom half being black. Wondering if it has anything to do with the screen resolution.

> Also noticing graphics corruption occurring when the mouse cursor moves on screen.

Please take photos about both problem.

napperley, 2021-11-06:
There is a process one has to go through before a C library can be accessed from Kotlin. A set of C mapping files is generated via the **cinterop** tool from header files, and stored in a separate [klib file](https://kotlinlang.org/docs/native-libraries.html) (this file acts as a wrapper for the C library). Each mapping file contains type information for the C APIs in Kotlin form that can be accessed on the Kotlin side. Below is an example from a **knm** file:
```
public final class _lv_disp_drv_t public constructor(rawPtr: kotlinx.cinterop.NativePtr /* = kotlin.native.internal.NativePtr */) : kotlinx.cinterop.CStructVar {
    @kotlinx.cinterop.internal.CStruct.VarType @kotlin.Deprecated public companion object : kotlinx.cinterop.CStructVar.Type {
    }

    public final var antialiasing: platform.posix.uint32_t /* = kotlin.UInt */ /* compiled code */

    public final var clean_dcache_cb: kotlinx.cinterop.CPointer<kotlinx.cinterop.CFunction<(kotlinx.cinterop.CPointer<lvgl._lv_disp_drv_t>?) -> kotlin.Unit>>? /* compiled code */

    public final val color_chroma_key: lvgl.lv_color_t /* = lvgl.lv_color32_t */ /* compiled code */

    public final var dpi: platform.posix.uint32_t /* = kotlin.UInt */ /* compiled code */

    public final var draw_buf: kotlinx.cinterop.CPointer<lvgl.lv_disp_draw_buf_t /* = lvgl._lv_disp_draw_buf_t */>? /* compiled code */

    public final var drv_update_cb: kotlinx.cinterop.CPointer<kotlinx.cinterop.CFunction<(kotlinx.cinterop.CPointer<lvgl._lv_disp_drv_t>?) -> kotlin.Unit>>? /* compiled code */

    public final var flush_cb: kotlinx.cinterop.CPointer<kotlinx.cinterop.CFunction<(kotlinx.cinterop.CPointer<lvgl._lv_disp_drv_t>?, kotlinx.cinterop.CPointer<lvgl.lv_area_t>?, kotlinx.cinterop.CPointer<lvgl.lv_color_t /* = lvgl.lv_color32_t */>?) -> kotlin.Unit>>? /* compiled code */

    public final var full_refresh: platform.posix.uint32_t /* = kotlin.UInt */ /* compiled code */

    public final var gpu_fill_cb: kotlinx.cinterop.CPointer<kotlinx.cinterop.CFunction<(kotlinx.cinterop.CPointer<lvgl._lv_disp_drv_t>?, kotlinx.cinterop.CPointer<lvgl.lv_color_t /* = lvgl.lv_color32_t */>?, lvgl.lv_coord_t /* = kotlin.Short */, kotlinx.cinterop.CPointer<lvgl.lv_area_t>?, kotlinx.cinterop.CValue<lvgl.lv_color_t /* = lvgl.lv_color32_t */>) -> kotlin.Unit>>? /* compiled code */

    public final var gpu_wait_cb: kotlinx.cinterop.CPointer<kotlinx.cinterop.CFunction<(kotlinx.cinterop.CPointer<lvgl._lv_disp_drv_t>?) -> kotlin.Unit>>? /* compiled code */

    public final var hor_res: lvgl.lv_coord_t /* = kotlin.Short */ /* compiled code */

    public final var monitor_cb: kotlinx.cinterop.CPointer<kotlinx.cinterop.CFunction<(kotlinx.cinterop.CPointer<lvgl._lv_disp_drv_t>?, platform.posix.uint32_t /* = kotlin.UInt */, platform.posix.uint32_t /* = kotlin.UInt */) -> kotlin.Unit>>? /* compiled code */

    public final var rotated: platform.posix.uint32_t /* = kotlin.UInt */ /* compiled code */

    public final var rounder_cb: kotlinx.cinterop.CPointer<kotlinx.cinterop.CFunction<(kotlinx.cinterop.CPointer<lvgl._lv_disp_drv_t>?, kotlinx.cinterop.CPointer<lvgl.lv_area_t>?) -> kotlin.Unit>>? /* compiled code */

    public final var screen_transp: platform.posix.uint32_t /* = kotlin.UInt */ /* compiled code */

    public final var set_px_cb: kotlinx.cinterop.CPointer<kotlinx.cinterop.CFunction<(kotlinx.cinterop.CPointer<lvgl._lv_disp_drv_t>?, kotlinx.cinterop.CPointer<platform.posix.uint8_tVar /* = kotlinx.cinterop.UByteVarOf<kotlin.UByte> */>?, lvgl.lv_coord_t /* = kotlin.Short */, lvgl.lv_coord_t /* = kotlin.Short */, lvgl.lv_coord_t /* = kotlin.Short */, kotlinx.cinterop.CValue<lvgl.lv_color_t /* = lvgl.lv_color32_t */>, lvgl.lv_opa_t /* = kotlin.UByte */) -> kotlin.Unit>>? /* compiled code */

    public final var sw_rotate: platform.posix.uint32_t /* = kotlin.UInt */ /* compiled code */

    public final var user_data: kotlinx.cinterop.COpaquePointer? /* = kotlinx.cinterop.CPointer<out kotlinx.cinterop.CPointed>? */ /* compiled code */

    public final var ver_res: lvgl.lv_coord_t /* = kotlin.Short */ /* compiled code */

    public final var wait_cb: kotlinx.cinterop.CPointer<kotlinx.cinterop.CFunction<(kotlinx.cinterop.CPointer<lvgl._lv_disp_drv_t>?) -> kotlin.Unit>>? /* compiled code */
}
```

napperley, 2021-11-06:
Attached is an image that shows the issue (using a Raspberry Pi 4):

[image]

kisvegabor, 2021-11-08:
> There is a process one has to go through before a C library can be accessed from Kotlin. 

Thanks, good to know!

> Attached is an image that shows the issue (using a Raspberry Pi 4):

Seems like an issue in the flush_cb. How does it look like now?

napperley, 2021-11-08:
Below is the code for the flush callback:
```kotlin
private fun flushFrameBuffer(
    driver: CPointer<lv_disp_drv_t>?,
    area: CPointer<lv_area_t>?,
    buf: CPointer<lv_color_t>?
) {
    fbdev_flush(drv = driver, area = area, color_p = buf)
}
```

The callback is referenced in the function that creates the display driver:
```kotlin
private fun createDisplayDriver(scope: MemScope, displayBuf: lv_disp_draw_buf_t): lv_disp_drv_t {
    val result = scope.alloc<lv_disp_drv_t>()
    lv_disp_drv_init(result.ptr)
    result.run {
        hor_res = HOR_RES.toShort()
        ver_res = VERT_RES.toShort()
        draw_buf = displayBuf.ptr
        flush_cb = staticCFunction(::flushFrameBuffer)
        antialiasing = 0u
    }
    return result
}
```

napperley, 2021-11-08:
Looks like there is either a bug in the LVGL/LV Drivers libraries, or in the Linux video driver for the Raspberry Pi. Changing the resolution results in the same issues occurring with some minor differences in scaling showing up. Adding the call to the `lv_disp_flush_ready` function in the flush callback makes no difference.

napperley, 2021-11-09:
Did some experimentation and uncovered some bugs with the LVGL/LV Drivers libraries by replacing the flush callback with the following:

```kotlin
// ...
private fun drawBackground(driver: CPointer<lv_disp_drv_t>?) = memScoped {
    val bufSize = HOR_RES * VERT_RES
    val buf = allocArray<lv_color_t>(bufSize)
    @Suppress("ReplaceRangeToWithUntil")
    (0..bufSize - 1).forEach { buf[it].full = LV_PALETTE_PINK }
    val area = alloc<lv_area_t>().apply {
        x1 = 0
        x2 = 800
        y1 = 0
        y2 = 600
    }
    fbdev_flush(drv = driver, area = area.ptr, color_p = buf)
}
// ...
private fun flushFrameBuffer(
    driver: CPointer<lv_disp_drv_t>?,
    area: CPointer<lv_area_t>?,
    buf: CPointer<lv_color_t>?
) {
    drawBackground(driver)
    lv_disp_flush_ready(driver)
}
```

When using the code above there is only a black background, however if the colour is changed to `LV_PALETTE_BLUE` then a blue rectangle is shown on screen. Changing `x2` to `HOR_RES.toShort()`, and `y2` to `VERT_RES.toShort()` will fill the entire screen as expected provided `LV_PALETTE_BLUE` is used as the colour, which seems to be the only colour that works without any issue. Looks like every other colour gets ignored.

With the way things are going it is very unlikely that there are bugs with the Linux video driver for the Raspberry Pi that are causing the rendering issues.

kisvegabor, 2021-11-09:
Tha palette were used incorrectly in your code. Try this:
```kotlin
buf[it].full = lv_palette_main(LV_PALETTE_PINK) 
```

napperley, 2021-11-11:
The **drawBackground** function now handles the colour correctly:
```kotlin
private fun drawBackground(driver: CPointer<lv_disp_drv_t>?) = memScoped {
    val bufSize = HOR_RES * VERT_RES
    val buf = allocArray<lv_color_t>(bufSize)
    var colour = 0u
    lv_palette_main(LV_PALETTE_BROWN).useContents { colour = full }
    @Suppress("ReplaceRangeToWithUntil")
    (0..bufSize - 1).forEach { buf[it].full = colour }
    val area = alloc<lv_area_t>().apply {
        x1 = 0
        x2 = 800
        y1 = 0
        y2 = 600
    }
    fbdev_flush(drv = driver, area = area.ptr, color_p = buf)
}
```

On the display it is now showing the correct colour.

kisvegabor, 2021-11-12:
Great. To solve the rest of the issues please paint the whole frame buffer to color and draw a rectangle with a different color using `fbdev_flush` but without using LVGL. E.g.
```
   ... clear the screen...
   ...fill buf with a color...
    
    val area = alloc<lv_area_t>().apply {
        x1 = 100
        x2 = 110
        y1 = 200
        y2 = 250
    }
    fbdev_flush(drv = driver, area = area.ptr, color_p = buf)

```

napperley, 2021-11-14:
@kisvegabor - I have done what you have mentioned :point_up: , and everything works as expected without any problems. Looks like there are some issues in the LVGL library that seem to be related to widget rendering. If rendering is done manually then there are no issues however if rendering is managed by LVGL then issues appear as mentioned in the first comment.

kisvegabor, 2021-11-16:
Okay, please try with these settings too. It should draw 1 px in the 2nd column and 3rd row. 
This will show if there are any off-by-on-one issues. 
```
x1 = 1,
x2 = 1,
y1 = 2,
y2 = 2,
```

I'm quite sure the LVGL renders the image correctly as it's working on other platforms and the simulator too.

napperley, 2021-11-16:
Used different values in order to ensure the graphic is shown on screen:
```
x1 = 2,
x2 = 102,
y1 = 2,
y2 = 102
```

The graphic is aligned correctly as expected. Note that **x1**, and **y1** are set to *2* to ensure the spacing is exactly the same.

napperley, 2021-11-17:
One person I have been in communication with on the Raspberry Pi forum mentioned that the terminal needs to be setup properly otherwise graphics corruption can occur. It is extremely surprising that LVGL doesn't have some functions for properly setting up and resetting the terminal, like NCurses does.

kisvegabor, 2021-11-17:
I wanted to be sure that 1 pixel can be rendered with
```
x1 = 1,
x2 = 1,
```

>  It is extremely surprising that LVGL doesn't have some functions for properly setting up and resetting the terminal, like NCurses does. 

The fbdev driver was already used in many projects, but of course, it's possible that RPi's fbdev needs some extra steps too. Can you print the FB info to see from where does the half-height comes from?

napperley, 2021-11-17:
Below is the output after running `fbset --show`:
```
mode "1280x720"
  geometry 1280 720 1280 720 16
  timings 0 0 0 0 0 0 0
  accel true
  rgba 5/11,6/5,5/0,0/0
endmode
```

napperley, 2021-11-17:
Double buffering won't work since the second buffer will be pointing to the second micro HDMI port, which isn't in use. The other day I found [something posted](https://forum.lvgl.io/t/setting-rotated-option-in-lv-disp-drv-t-causes-hard-fault/440/8?u=napperley) on the LVGL forums where the issues are the same:

https://forum.lvgl.io/uploads/default/optimized/1X/2f8a11647ace946c787f5cb523225d2ca12d36cf_2_562x750.jpeg

This does confirm that there are some bugs with the [fbdev_flush](https://github.com/lvgl/lv_drivers/blob/99d83c960e3370ec450c5427583c4823effda041/display/fbdev.c#L171) function.

kisvegabor, 2021-11-22:
I've tested it several times on my notebook and it worked well here. 

Now I'm just guessing blindly in the hope that we fill find something: what are the resolution and color depth related values in `vinfo`?

napperley, 2021-11-23:
With the simple Kotlin program I have written (using the Linux Frame Buffer library) it reports the following information from `vinfo`:

- Display Resolution: 1280x720 (DMT Mode 85)
- Colour Depth: 16

kisvegabor, 2021-11-26:
> Did you test the same stuff on a real SBC (Single Board Computer)?

I've tried/used LVGL on Raspberry with SDL, not with the pure framebuffer device. 

I'm a little bit puzzled by this issue:
- I'm pretty sure the LVGL renders the widgets correctly. If it wasn't the case, it'd be wrong on other platforms too.
- The flush_cb seems correct too.

Two uncertain points were left:
- The framebuffer of RPi: Please try [this](https://docs.lvgl.io/master/intro/index.html#why-do-i-see-only-garbage-on-the-screen). Call `my_flush_cb` with different areas and take a very close photo of the screen on which we can count the pixels. 
- The Kotlin binding: Can you try with pure C code?

napperley, 2021-11-27:
Does the **SDL** backend work without needing to use a Windowing System? No Kotlin binding is used, just the LVGL/LV Drivers APIs via mapping files that are generated by the **cinterop** tool. A Kotlin binding would require a Kotlin API to be developed (either done manually or auto generated) as a Kotlin library that **wraps** around a C library.

Are there some sample projects available that use the **SDL** backend and work on a Raspberry Pi as the target device?

napperley, 2021-11-27:
Trying with the [following](https://docs.lvgl.io/master/intro/index.html#why-do-i-see-only-garbage-on-the-screen) code only results in a black screen being displayed. The sample code doesn't show the contents of the callback.

kisvegabor, 2021-11-30:
> Does the SDL backend work without needing to use a Windowing System?

Nope.

> Are there some sample projects available that use the SDL backend and work on a Raspberry Pi as the target device?

I used [this](https://github.com/lvgl/lv_sim_eclipse_sdl). For the sake of simplicity, I built it right on RPi.

> The sample code doesn't show the contents of the callback.

The goal is to test your flush callback. Yours looks good but it seems there are some problems with it. `my_flush_cb` should be `fbdev_flush`. If it can't show anything when you copy something to the frame buffer (with `fbdev_flush`) I'm a little bit clueless.

napperley, 2021-12-01:
The right callback is now being used however a white background is displayed, and no rectangle appears. Both the Kotlin and C versions show the same result. Below is the Kotlin version:

```kotlin
private fun drawSquare(driver: CPointer<lv_disp_drv_t>?) = memScoped {
    val width = 20
    val height = 10
    val buf = allocArray<lv_color_t>(width * height)
    (0..height - 1).forEach { y ->
        val gradient = lv_color_mix(
            c1 = cValue { full = LV_PALETTE_BLUE },
            c2 = cValue { full = LV_PALETTE_RED },
            mix = ((y * 255) / height).toUByte()
        )
        (0..width - 1).forEach { x -> buf[y + x].full = gradient.useContents { full } }
    }
    val area = alloc<lv_area_t> {
        x1 = 10
        y1 = 40
        x2 = (x1 + width - 1).toShort()
        y2 = (y1 + height - 1).toShort()
    }
    flushFrameBuffer(driver = driver, area = area.ptr, buf = buf)
}
```

Below is the C version (uses a plain blue colour instead of a gradient):

```c
#define BUF_W 20
#define BUF_H 10

draw_square(lv_disp_drv_t *driver) {
  lv_color_t buf[BUF_W * BUF_H];
  lv_color_t * buf_p = buf;
  uint16_t x, y;
  for(y = 0; y < BUF_H; y++) {
    lv_color_t c = lv_color_make(0, 0, 255);
    for(x = 0; x < BUF_W; x++){
      (*buf_p) =  c;
      buf_p++;
    }
  }

  lv_area_t a;
  a.x1 = 10;
  a.y1 = 40;
  a.x2 = a.x1 + BUF_W - 1;
  a.y2 = a.y1 + BUF_H - 1;
  driver->flush_cb(driver, &a, buf);
}
```

Below is the flush callback assigned to the display driver:

```kotlin
private fun flushFrameBuffer(
    driver: CPointer<lv_disp_drv_t>?,
    area: CPointer<lv_area_t>?,
    buf: CPointer<lv_color_t>?
) {
    fbdev_flush(drv = driver, area = area, color_p = buf)
    lv_disp_flush_ready(driver)
}
```

napperley, 2021-12-04:
Been able to borrow a [Waveshare 7" HDMI touchscreen](https://www.waveshare.com/wiki/7inch_HDMI_LCD_(H)_(with_case)) (no need to install a driver, and manual calibration isn't required) for testing, and found the result to be very similar with a half height background displayed along with graphics corruption. The issue definitely resides within the [fbdev_flush](https://github.com/lvgl/lv_drivers/blob/99d83c960e3370ec450c5427583c4823effda041/display/fbdev.c#L171) function, which is provided by the LV Drivers library.

napperley, 2021-12-05:
The Linux Frame Buffer driver from LV Drivers interprets **1200** as the max height (part of the screen resolution) for the touchscreen which is incorrect, yet the driver correctly picks up on **1024** for the max width. Maximum height for the touchscreen is **600**. The supported screen resolution for the touchscreen is **1024x600**. Setting the resolution does have an effect, however the driver isn't correctly picking up the screen resolution even though the resolution has been set correctly in the program.

napperley, 2021-12-05:
Below are the contents of the **/boot/config.txt** file:
```
# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
#dtoverlay=vc4-fkms-v3d
max_framebuffers=2
# Change the colour depth.
framebuffer_depth=16

[all]
#dtoverlay=vc4-fkms-v3d

# Waveshare 7" touchscreen configuration.
max_usb_current=1
hdmi_force_hotplug=1 
config_hdmi_boost=10
hdmi_group=2 
hdmi_mode=87 
hdmi_cvt 1024 600 60 6 0 0 0
```

Here is the output after running `fbset`:
```
mode "1024x600"
    geometry 1024 600 1024 600 16
    timings 0 0 0 0 0 0 0
    rgba 5/11,6/5,5/0,0/16
endmode
```

napperley, 2021-12-06:
@kisvegabor - Graphics rendering will be different between the Linux Frame Buffer, and SDL backends. Doing the testing with the SDL backend won't work when a different backend is used. You don't seem to fully understand that graphics rendering behaviour **will not** be uniform across all backends. With software testing the testing environment **MUST** be exactly the same as the production environment!

napperley, 2021-12-06:
@kisvegabor - I have tried the [sample application](https://github.com/lvgl/lv_port_linux_frame_buffer) which works fine (no graphics corruption detected) on the Raspberry Pi, however it doesn't link the LVGL and LV Drivers libraries. Can you please provide a version of the sample application that links both libraries (using static library files). Would the distribution of the LVGL, and LV Drivers libraries affect graphics rendering? I do know that configuration is handled differently when using the libraries in static library file form.

napperley, 2021-12-07:
Has **all** of the resolution configuration been moved out of the header files into the LVGL API?

napperley, 2021-12-07:
Finally discovered the source of the problem. Incorrect configuration in the **lv_drv_conf.h** file (included in the **liblv_drivers.a** file), which took weeks to find as the **"needle in the haystack"**. It is very concerning that LVGL and LV Drivers configuration isn't backwards compatible, and that incorrect configuration of the **lv_drv_conf.h** file can cause graphics rendering issues.

Very surprising the there are still some screen resolution configuration options (eg **MONITOR_HOR_RES**) in the **lv_drv_conf.h** file that haven't been migrated to the LVGL API.

kisvegabor, 2021-12-08:
Now the driver part of LVGL is really not ideal. It's scattered to multiple repos and there is no uniform API and config workflow. 
To solve it we started to work on a [new driver architecture](https://github.com/lvgl/lvgl/issues/2720) and probably will move all drivers into the lvgl repo so it will be easier to version and test everything together.  

I'm really sorry that it gave you difficult times... :slightly_frowning_face: I hope we will have something much clear in the near future.

napperley, 2021-12-09:
Some [changes](https://gitlab.com/napperley/lvgl-hello/-/commit/34b711b10ddaa2a2efc8942ff02380bbfe485911) have been made to my program to confirm that the issue is resolved, and I can confirm that the issue is no more :) . Would be very useful to link this issue to the [new driver architecture](https://github.com/lvgl/lvgl/issues/2720) since it perfectly highlights some of the issues that can be encountered with the current configuration system.

closed 2021-12-09 by napperley

kisvegabor, 2021-12-09:
Glad to hear that! :slightly_smiling_face: