lvgl/lvgl #2758: RPi build fails

amirgon, 2021-11-02:
See [this workflow](https://github.com/lvgl/lv_micropython/runs/4085833219?check_suite_focus=true):

```
/home/runner/work/lv_micropython/lv_micropython/lib/lv_bindings/lvgl/src/draw/lv_draw_label.c:433:25: error: format '%X' expects argument of type 'unsigned int', but argument 6 has type 'uint32_t' {aka 'long unsigned int'} [-Werror=format=]
  433 |             LV_LOG_WARN("lv_draw_letter: glyph dsc. not found for U+%X", letter);
      |                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~  ~~~~~~
      |                                                                          |
      |                                                                          uint32_t {aka long unsigned int}
/home/runner/work/lv_micropython/lv_micropython/lib/lv_bindings/lvgl/src/draw/../hal/../misc/lv_log.h:106:91: note: in definition of macro 'LV_LOG_WARN'
  106 | #    define LV_LOG_WARN(...) _lv_log_add(LV_LOG_LEVEL_WARN, __FILE__, __LINE__, __func__, __VA_ARGS__)
      |                                                                                           ^~~~~~~~~~~
/home/runner/work/lv_micropython/lv_micropython/lib/lv_bindings/lvgl/src/draw/lv_draw_label.c:433:70: note: format string is defined here
  433 |             LV_LOG_WARN("lv_draw_letter: glyph dsc. not found for U+%X", letter);
      |                                                                     ~^
      |                                                                      |
      |                                                                      unsigned int
      |                                                                     %lX
```

I think it would be a good idea to add RPi build to the CI workflow (either as part of Micropython CI or indepedant)

kisvegabor, 2021-11-03:
I've pushed a fix for it https://github.com/lvgl/lvgl/commit/d9d3f271267e760c8459b65c392914143a58b89c

I agree to add RPi tests too. Do you have time/interest to add it?

embeddedt, 2021-11-03:
It's only a couple lines that need changing so I went ahead and added it. It appears to be compiling successfully now.

closed 2021-11-03 by embeddedt

kisvegabor, 2021-11-03:
Amazing, thanks!