lvgl/lvgl #2763: Excessive screen flush on creating objects

huming2207, 2021-11-05:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

- LVGL version: Git master (as of 6d955210765de972f78b8c307df2f2387e4580ed)
- Platform: ESP32 + E-Ink display (ED133UT2 etc.)

When creating a `lv_obj_t`, and then setting the position, LVGL somehow does two screen flushes. This is okay for LCD but it's definitely not ideal for e-paper display as it blinks twice, which results in a bad user experience.

### To Reproduce

```cpp
        // Draw some other objects...
        // ... ...
        // ...and here's the troublemaker
        auto *obj = lv_obj_create(lv_scr_act());
        lv_obj_align(obj, LV_ALIGN_CENTER, 0, 0);
        lv_obj_set_size(obj, 100, 100);
        vTaskDelay(portMAX_DELAY);
```

Then in my `flush_cb` I logged the flushes. 

```
W (1842) lv_drv: x1:0, y1:0; x2:1599, y2:1199; full pix cnt:1920000, actual pix cnt: 1920000, len:960000, off_start: 0, off_end: 0
W (4042) lv_drv: x1:0, y1:0; x2:99, y2:99; full pix cnt:10000, actual pix cnt: 10000, len:5000, off_start: 0, off_end: 0
W (4052) lv_drv: x1:750, y1:550; x2:849, y2:649; full pix cnt:10400, actual pix cnt: 10000, len:5200, off_start: 2, off_end: 2
```

As you may notice, there are two flushes for this `lv_obj_t`. So the screen blinks twice, one goes from (0, 0) to (99, 99), which is excessive, and the other one goes from (750, 550) to (849, 649), which is the real flush I need.

I've tried setting the config to delay the refresh period and disable the animation (as I'm working on an e-paper display) but still no luck:

- `LV_DISP_DEF_REFR_PERIOD` to 1000
- `LV_THEME_DEFAULT_TRANSITION_TIME` to 0
- `LV_THEME_DEFAULT_GROW` to false

Have I missed something or is this a bug?

### Expected behavior

As I said above, I think this object should be only drawn once, which is the one from (750, 550) to (849, 649). 

### Screenshots or video

Cannot provide any photo or video due to NDA & legal limits

embeddedt, 2021-11-05:
The first flush is most likely triggered by the fact that the object gets created at one position and then moved to another.

When I worked with an eInk display, I used the `monitor_cb` hook to trigger display updates, because my hardware had a framebuffer which could be written to at any time, and I could trigger the actual ink to update once all the individual flushes were done. Do you have that ability?

huming2207, 2021-11-05:
Hi @embeddedt 

Thanks for your reply.

> The first flush is most likely triggered by the fact that the object gets created at one position and then moved to another.

Yea, I suspect that's the original case. 

> When I worked with an eInk display, I used the `monitor_cb` hook to trigger display updates, because my hardware had a framebuffer which could be written to at any time, and I could trigger the actual ink to update once all the individual flushes were done. Do you have that ability?

I will have a try on this approach. But it may complicate our design as we have some other uses on the frame buffer and our TCON implementation.

huming2207, 2021-11-05:
Hmmm @embeddedt  But I think it is still kinda a bug, as I still need to do two flush calls (even though I refresh/update the screen once). Is there something that can effectively do this below:

```cpp
auto *obj = lv_obj_create(lv_scr_act());
lv_obj_pause_render_on_me(obj); // Don't render me till I unpause
lv_obj_align(obj, LV_ALIGN_CENTER, 0, 0);
lv_obj_set_size(obj, 100, 100);
lv_obj_unpause_render_on_me(obj); // ...okay now do the rendering!
```

huming2207, 2021-11-05:
I ended up in getting a outer bounding box + `lv_disp_flush_is_last()` API approach instead. Basically:

1. find the top left and bottom right corner of a series of refreshes;
2. if lv_disp_flush_is_last() returns true, then refresh with the top left & bottom right coordinates. 

So far it works fine for me. But still, I'd suggest LVGL should provide a guard state thing for e-paper panels as I mentioned above.

kisvegabor, 2021-11-16:
In reality, this problem is a little bit more complex. As you create an object, change its size, position, and other parameters the parent, grandparent, siblings, etc also might be affected by the changes of the object. E.g. a layout on the parent will change the position of siblings too.

So it's very hard to see what has been actually changed when LVGL gets to rendering. To do this it should save the old area and a new area of all objects and if they are different before drawing happens instruct LVGL to redraw the final old and new areas. 

It seems doable but would require much more memory for some gain in special cases. Therefore I rather suggest using the method you have found with `lv_disp_flush_is_last()`.

huming2207, 2021-11-16:
> In reality, this problem is a little bit more complex. As you create an object, change its size, position, and other parameters the parent, grandparent, siblings, etc also might be affected by the [...]

Thanks @kisvegabor . Yea we had a discussion and indeed it is not trivial to get this working, particularly LVGL is relatively more "LCD-oriented". For now we decided to just use `lv_disp_flush_is_last()` and seek for some other alternative solutions later if possible.

kisvegabor, 2021-11-17:
Okay :slightly_smiling_face: 

I close this issue then. Feel free to comment here if you have any related questions or remarks.

closed 2021-11-17 by kisvegabor

huming2207, 2021-11-17:
@kisvegabor  Yep, we will see if we can implement something helpful for LVGL with e-paper displays later on. Some improvements we can think of are:

- Grayscale support. Currently, LVGL only has 1-bit per pixel monochrome, but we need 4-bit or 8-bit per pixel grayscale. Although the `set_px_cb` can do the trick but it's slow; 
- The screen refresh issue mentioned in this issue topic;
- Waveform-aware flushing, e.g. use A2 waveform for animations, occasionally do INIT waveform for ghosting removal etc.

kisvegabor, 2021-11-18:
> Waveform-aware flushing, e.g. use A2 waveform for animations and use GC16 for the rest, occasionally do INIT waveform for ghosting removal etc.

I don't really understand it. What are A2 and GC16?

embeddedt, 2021-11-18:
@kisvegabor I am not completely familiar with it either, but I think eInk screens have different ways of refreshing that can be selected by software. I think some are faster and others are higher quality, or something along those lines.

huming2207, 2021-11-19:
> I don't really understand it. What are A2 and GC16?

A2 and GC16 are some weird namings proposed by those e-paper chip and panel makers (probably e-Ink). Basically, A2 is the fast refresh waveform usually used for animation, while GC16 is the slow-ish waveform but it can deliver a better result with less ghosting.

See here for more info: https://www.waveshare.com/w/upload/c/c4/E-paper-mode-declaration.pdf

kisvegabor, 2021-11-22:
I see, thanks. The architecture of LVGL's draw system is under [heavy redesign](https://github.com/lvgl/lvgl/pull/2803). I think we can consider how to better support eink/monochrome/grayscale displays too. 

It'll take some time to have something useable, but I'm sure v8.2 will have more options to handle this kind of technologys more optimally. 

On a side note, now with LVGL we are targeting 4k monitors driven by video cards and eink display too which is pretty amazing that we can cover the whole spectrum but it's also challenging. Anyway, we are working on making both ends smother! :slightly_smiling_face:

huming2207, 2021-11-22:
> It'll take some time to have something useable, but I'm sure v8.2 will have more options to handle this kind of technologys more optimally.

@kisvegabor yep, I've also seen that PR and it's quite promising. I guess what I can do later is probably modify the decoupled software renderer code (those at `src/draw/sw`) and add the support for 4-bit grayscale. 

For example, I guess for a quick dirty hack is probably modify something here: 

https://github.com/lvgl/lvgl/pull/2803/commits/ae09fb0a682f1cb44853d8505a6e76e651a955a5#diff-242f305692436f289db7b3cdda6298157b9ab35835506b88b28a86b1524051abR240

...and round to grayscale, e.g. like this: https://stackoverflow.com/questions/15686277/convert-rgb-to-grayscale-in-c

kisvegabor, 2021-11-22:
Glad to hear that you have followed this PR. :slightly_smiling_face: 

I'm thinking about the same. Some custom `blend` functions should help to support any color format.