lvgl/lvgl #2768: LVGL V8 slow - seems to have a Tick bug.

Satnet, 2021-11-06:
### Perform all steps below and tick them with [x]
- [ X ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ X ] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
LVGL Runs very slow. At one point, my clicking actions were delayed by almost a full minute! I have the click tied to a Serial.print so I clicked 3 times and waited, after about 1 minute the 3 clicks came in. Most of the time it is only a little bit slow with a slight delay. This works 100% perfectly on LVGL V7, been running very advanced graphics for over a year on this hardware.

### To Reproduce
`#include <Arduino.h>
#include "ILI948x_t4_mm.h"
ILI948x_t4_mm lcd = ILI948x_t4_mm(17, 16, 5); //(dc, cs, rst)

static void lv_tick_handler(void)
{
  lv_tick_inc(LVGL_TICK_PERIOD);
}

FASTRUN void my_flush_cb(lv_disp_drv_t * disp_drv, const lv_area_t * area, lv_color_t * color_p)
{
    /*The most simple case (but also the slowest) to put all pixels to the screen one-by-one
     *`put_px` is just an example, it needs to implemented by you.*/
      lcd.pushPixels16bitDMA((uint16_t*)(color_p),area->x1, area->y1, area->x2, area->y2);
    /* IMPORTANT!!!
     * Inform the graphics library that you are ready with the flushing*/
      //lv_disp_flush_ready(disp_drv);
}

FASTRUN void flushCB()
{
  lv_disp_flush_ready(&disp_drv);
}

FASTRUN bool touch_glass_input(lv_indev_drv_t * drv, lv_indev_data_t*data) {
  int XL = 0;
  int XH = 0;
  int YL = 0;
  int YH = 0;
  int tStatus = 0;

  XH = readTouchReg(TOUCH_REG_XH);
  XL = readTouchReg(TOUCH_REG_XL);
  YH = readTouchReg(TOUCH_REG_YH);
  YL = readTouchReg(TOUCH_REG_YL);

  // TOUCH INPUT, MAPPED FOR LANDSCAPE
  int x = ((XH & 0x0F) << 8) | XL;
  int y = ((YH & 0x0F) << 8) | YL;
  data->point.x = map(y, 0, 480, 480, 0);
  data->point.y = x;

  tStatus = XH>>6;
  if (tStatus == 1) {
    data->state = LV_INDEV_STATE_REL;
  } else {
    data->state = LV_INDEV_STATE_PR;
  }

  //data->state = LV_INDEV_STATE_PR; // or LV_INDEV_STATE_REL;
  return false; /*No buffering now so no more data read*/
}

void setup() {   
  Serial.begin(115200);
  //FreqMeasure.begin();
  Serial.print(CrashReport);

  pinMode(12, OUTPUT);
  pinMode(14, OUTPUT);
  digitalWrite(12, HIGH);
  digitalWrite(14, HIGH);

  init_touch();

  lcd.begin(20);
  lcd.setRotation(3);
  lcd.onCompleteCB(&flushCB);
  lcd.setFrameRate(68);

  lv_init();

  lv_disp_draw_buf_init(&disp_buf, buf_1, NULL, screenWidth * screenHeight /4);

// INIT THE DISPLAY DRIVERS
  lv_disp_drv_init(&disp_drv);            /*Basic initialization*/
  disp_drv.draw_buf = &disp_buf;          /*Set an initialized buffer*/
  disp_drv.flush_cb = my_flush_cb;        /*Set a flush callback to draw to the display*/
  disp_drv.hor_res = screenWidth;                 /*Set the horizontal resolution in pixels*/
  disp_drv.ver_res = screenHeight;                 /*Set the vertical resolution in pixels*/

  lv_disp_t * disp;
  disp = lv_disp_drv_register(&disp_drv); /*Register the driver and save the created display objects*/

// REGISTER THE TOUCH INPUT DRIVER
  static lv_indev_drv_t indev_drv;
  lv_indev_drv_init(&indev_drv);      /*Basic initialization*/
  indev_drv.type = LV_INDEV_TYPE_POINTER;                 /*See below.*/
  indev_drv.read_cb = touch_glass_input;              /*See below.*/
  /*Register the driver in LVGL and save the created input device object*/
  lv_indev_t * my_indev = lv_indev_drv_register(&indev_drv);

// MAIN SCREEN
  gaugeScreen = lv_obj_create(lv_scr_act());

  // SETTINGS ICON
    // SETTINGS ICON BOTTOM LEFT
      settingsBtnCont = lv_btn_create(gaugeScreen);
      //lv_obj_remove_style_all(settingsBtnCont);
      lv_obj_add_event_cb(settingsBtnCont, event_handler, LV_EVENT_ALL, NULL);
      lv_obj_set_size(settingsBtnCont, 80, 80);
      lv_obj_align(settingsBtnCont, LV_ALIGN_BOTTOM_LEFT, 50, -50);

    // SETTINGS SYMBOL VARIABLE
    //settingsBtn = lv_label_create(settingsBtnCont);
    //lv_label_set_text_static(settingsBtn, LV_SYMBOL_SETTINGS);

  loadStyles();
}

const int loopDelay1 = 1; // Make a request every 500ms
unsigned long timeNow1 = 0;
void loop() {
  if (millis() > timeNow1 + loopDelay1) {
    lv_tick_handler();
    lv_timer_handler();

    timeNow1 = millis();
  }
}`

### Expected behavior
Real time, butter smooth. Screen is running DMA on a Teensy Micromod at 600mhz. We get above 70fps. Works perfectly in LVGL V7.

### Screenshots or video
https://easyupload.io/dbqsrl

david-res, 2021-11-06:
I can confirm this happens to me with the same hardware/code setup

Satnet, 2021-11-06:
Cupra Gauge PSD
 Formentor Gauge.psd
<https://drive.google.com/file/d/12-ppV2AQ69G6TDMBNAWmANHM28wIidyB/view?usp=drive_web>

kevinpt, 2021-11-07:
`if (millis() > timeNow1 + loopDelay1)`

This isn't your problem but it is a red flag. When millis() rolls over after 49 days the loop stops ticking. Time delays should be computed with deltas that are safe across the overflow boundary:

`if(millis() - timeNow1 > loopDelay)`

loopDelay1 should also be unsigned.

Satnet, 2021-11-07:
That’s okay, this device will never run for even 24h without being
restarted. Thanks for the lesson tho, good to know in the future!

kisvegabor, 2021-11-08:
What is `lv_tick_handler()`? It looks and sounds like an LVGL function but there is no such function in LVGL.

Satnet, 2021-11-08:
That’s actually for V7, it’s similar for V8. I’ve done a bunch of testing
but nothing solves the lag in V8. Works fine in V7 tho.

mån 8 nov. 2021 kl. 11:03 skrev Gabor Kiss-Vamosi ***@***.***

kisvegabor, 2021-11-08:
But there is no `lv_tick_handler` in v7 either. :confused:

Satnet, 2021-11-08:
static void lv_tick_handler(void) { lv_tick_inc(LVGL_TICK_PERIOD); }

mån 8 nov. 2021 kl. 11:14 skrev Gabor Kiss-Vamosi ***@***.***

david-res, 2021-11-08:
@kisvegabor 
lv_tick_handler is just a custom function to call lv_tick_inc() that is triggered by an internal timer on a Teensy.

kisvegabor, 2021-11-08:
Ah sorry,  I missed it in your example code... 

Please try this:
```c
void loop() {
  LV_LOG_USER("outter: %d", lv_tick_get());
  if (millis() > timeNow1 + loopDelay1) {
    LV_LOG_USER("inner: %d", lv_tick_get());
    lv_tick_handler();
    lv_timer_handler();

    timeNow1 = millis();
  }
}
```

Satnet, 2021-11-08:
When I enable logging in lv_conf.h, it actually crashes. If I disable it,
no crash. Your loop code works, but #define LV_USE_LOG 1 Makes it crash.

Satnet, 2021-11-13:
Sorry for my late reply! I tried this just now and it still crashes. If I comment out the two "printf" lines it does not crash.
So something is up there.

kisvegabor, 2021-11-16:
Does printf work without LVGL?

Satnet, 2021-11-17:
I did Serial.printf and that works. I got a ton of output. I was clicking my screen while doing this and it was still slow.
This is allot of output.

```
12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 
12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594outter: 12594inner: 12594outter: 12595outter: 12595outter: 12595outter: 12595outter: 12595outter: 12595outter: 12595outter: 12595outter: 12595outter: 12595outter: 12595outter: 12595outter: 12595outter: 12595outter: 12595outter: 12595outter: 12595outter: 12595outter: 12595outter: 12595outter: 12595outter: 12595outter: 12595outter: 12595outter: 12595outter: 12595outter: 125 [... cut ...] tter: 12619outter: 12619outter: 12619outter: 12619outter: 12619outter: 12619outter: 12619outter: 12619outter: 12619outter: 12619outter: 12619outter: 12619outter: 12619outter: 12619outter: 12619outter: 12619outter: 12619outter: 12619outter: 12619outter: 12619outter: 12619outter: 12619outter: 12619outter: 12619outter: 12619outter: 
12619outter: 12619outter: 12619outter: 12619outter: 12619outter: 12619outter: 12619outter: 12619outter: 12619outter: 12619outter: 12619outter: 12619outter: 12619outter:

kisvegabor, 2021-11-17:
Looks good, but please send a log with "inner" only too. 

On Arduino, you can use the tick in another way too. Just set `LV_TICK_CUSTOM 1` in `lv_conf.h` and do not call `lv_tick_inc()`. Does it make any difference?

Satnet, 2021-11-17:
This is with the old TICK settings (TICK CUSTOM 0).

`1235inner: 1236inner: 1237inner: 1238inner: 1239inner: 1240inner: 1241inner: 1242inner: 1243inner: 1244inner: 1245inner: 1246inner: 1247inner: 1248inner: 1249inner: 1250inner: 1251inner: 1252inner: 1253inner: 1254inner: 1255inner: 1256inner: 1257inner: 1258inner: 1259inner: 1260inner: 1261inner: 1262inner: 1263inner: 1264inner: 1265inner: 1266inner: 1267inner: 1268inner: 1269inner: 1270inner: 1271inner: 1272inner: 1273inner: 1274inner: 1275inner: 1276inner: 1277inner: 1278inner: 1279inner: 1280inner: 1281inner: 1282inner: 1283inner: 1284inner: 1285inner: 1286inner: 1287inner: 1288inner: 1289inner: 1290inner: 1291inner: 1292inner: 1293inner: 1294inner: 1295inner: 1296inner: 1297inner: 1298inner: 1299inner: 1300inner: 1301inner: 1302inner: 1303inner: 1304inner: 1305inner: 1306inner: 1307inner: 1308inner: 1309inner: 1310inner: 1311inner: 1312inner: 1313inner: 1314inner: 1315inner: 1316inner: 1317inner: 1318inner: 1319inner: 1320inner: 1321inner: 1322inner: 1323inner: 1324inner: 1325inner: 1326inner: 1327inner: 1328inner: 1329inner: 1330inner: 1331inner: 1332inner: 1333inner: 1334inner: 1335inner: 1336inner: 1337inner: 1338inner: 1339inner: 1340inner: 1341inner: 1342inner: 1343inner: 1344inner: 1345inner: 1346inner: 1347inner: 1348inner: 1349inner: 1350inner: 1351inner: 1352inner: 1353inner: 1354inner: 1355inner: 1356inner: 1357inner: 1358inner: 1359inner: 1360inner: 1361inner: 1362inner: 1363inner: 1364inner: 1365inner: 1366inner: 1367inner: 1368inner: 1369inner: 1370inner: 1371inner: 1372inner: 1373inner: 1374inner: 1375inner: 1376inner: 1377inner: 1378inner: 1379inner: 1380inner: 1381inner: 1382inner: 1383inner: 1384inner: 1385inner: 1386inner: 1387inner: 1388inner: 1389inner: 1390inner: 1391inner: 1392inner: 1393inner: 1394inner: 1395inner: 1396inner: 1397inner: 1398inner: 1399inner: 1400inner: 1401inner: 1402inner: 1403inner: 1404inner: 1405inner: 1406inner: 1407inner: 1408inner: 1409inner: 1410inner: 1411inner: 1412inner: 1413inner: 1414inner: 1415inner: 1416inner: 1417inner: 1418inner: 1419inner: 1420inner: 1421inner: 1422inner: 1423inner: 1424inner: 1425inner: 1426inner: 1427inner: 1428inner: 1429inner: 1430inner: 1431inner: 1432inner: 1433inner: 1434inner: 1435inner: 1436inner: 1437inner: 1438inner: 1439inner: 1440inner: 1441inner: 1442inner: 1443inner: 1444inner: 1445inner: 1446inner: 1447inner: 1448inner: 1449inner: 1450inner: 1451inner: 1452inner: 1453inner: 1454inner: 1455inner: 1456inner: 1457inner: 1458inner: 1459inner: 1460inner: 1461inner: 1462inner: 1463inner: 1464inner: 1465inner:`

This is with TICK CUSTOM 1 and lv_tick_inc not present.

`inner: 14872inner: 14874inner: 14877inner: 14879inner: 14881inner: 14883inner: 14885inner: 14887inner: 14889inner: 14891inner: 14893inner: 14895inner: 14897inner: 14899inner: 14901inner: 14903inner: 14905inner: 14908inner: 14910inner: 14912inner: 14914inner: 14916inner: 14918inner: 14920inner: 14922inner: 14924inner: 14926inner: 14928inner: 14930inner: 14932inner: 14934inner: 14936inner: 14939inner: 14941inner: 14943inner: 14945inner: 14947inner: 14949inner: 14951inner: 14953inner: 14955inner: 14957inner: 14959inner: 14961inner: 14963inner: 14965inner: 14967inner: 14970inner: 14972inner: 14974inner: 14976inner: 14978inner: 14980inner: 14982inner: 14984inner: 14986inner: 14988inner: 14990inner: 14992inner: 14994inner: 14996inner: 14998inner: 15001inner: 15003inner: 15005inner: 15007inner: 15009inner: 15011inner: 15013inner: 15015inner: 15017inner: 15019inner: 15021inner: 15023inner: 15025inner: 15027inner: 15029inner: 15032inner: 15034inner: 15036inner: 15038inner: 15040inner: 15042inner: 15044inner: 15046inner: 15048inner: 15050inner: 15052inner: 15054inner: 15056inner: 15058inner: 15060inner: 15063inner: 15065inner: 15067inner: 15069inner: 15071inner: 15073inner: 15075inner: 15077inner: 15079inner: 15081inner: 15083inner: 15085inner: 15087inner: 15089inner: 15091inner: 15094inner: 15096inner: 15098inner: 15100inner: 15102inner: 15104inner: 15106inner: 15108inner: 15110inner: 15112inner: 15114inner: 15116inner: 15118inner: 15120inner: 15122inner: 15125inner: 15127inner: 15129inner: 15131inner: 15133inner: 15135inner: 15137inner: 15139inner: 15141inner: 15143inner: 15145inner: 15147inner: 15149inner: 15151inner: 15153inner: 15156inner: 15158inner: 15160inner: 15162inner: 15164inner: 15166inner: 15168inner: 15170inner: 15172inner: 15174inner: 15176inner: 15178inner: 15180inner: 15182inner: 15184inner: 15187inner: 15189inner: 15191inner: 15193inner: 15195inner: 15197inner: 15199inner: 15201inner: 15203inner: 15205inner: 15207inner: 15209inner: 15211inner: 15213inner: 15215inner: 15218inner: 15220inner: 15222inner: 15224inner: 15226inner: 15228inner: 15230inner: 15232inner: 15234inner: 15236inner: 15238inner: 15240inner: 15242inner: 15244inner: 15246inner: 15249inner: 15251inner: 15253inner: 15255inner: 15257inner: 15259inner: 15261inner: 15263inner: 15265inner: 15267inner: 15269inner: 15271inner: 15273inner: 15275inner: 15277inner: 15280inner: 15282inner: 15284inner: 15286inner: 15288inner: 15290inner: 15292inner: 15294inner: 15296inner: 15298inner: 15300inner: 15302inner: 15304inner: 15306inner: 15308inner: 15311inner: 15313inner: 15315inner: 15317inner: 15319inner: 15321inner: 15323inner: 15325inner: 15327inner: 15329inner: 15331inner: 15333inner: 15335inner: 15337inner: 15339inner: 15342inner: 15344inner: 15346inner: 15348inner: 15350inner: 15352inner: 15354inner: 15356inner: 15358inner: 15360i [... cut ...] nner: 16187inner: 16189inner: 16191inner: 16193inner: 16195inner: 16197inner: 16199inner: 16201inner: 16203inner: 16205inner: 16207inner: 16210inner: 16212inner: 16214inner: 16216inner: 16218inner: 16220inner: 16222inner: 16224inner: 16226inner: 16228inner: 16230inner: 16232inner: 16234inner: 16236inner: 16238inner: 16241inner: 16243inner: 16245inner: 16247inner: 16249inner: 16251inner: 16253inner: 16255inner: 16257inner: 16259`

With the new tick settings, it does work much better! Why is that?

kisvegabor, 2021-11-17:
> With the new tick settings, it does work much better! Why is that?

To be honest I don't know. Incrementing `lv_tick_inc` in the main loop should work but might result in **slight** timing issues as the execution time of `lv_timer_handler` is not added to the tick. 

So the former case should work too. Not perfectly but normally.

Satnet, 2021-11-18:
Well, seems like it might be a bug then since it works in V7. Thanks for finding a solution, I will use this way with V8 for now.

kisvegabor, 2021-11-19:
It's a pity that we couldn't find the real reason. 

Anyway, happy to hear that it works the other way.

closed 2021-11-19 by kisvegabor