lvgl/lvgl #2774: Unmatched event callback

FASTSHIFT, 2021-11-08:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
The target of the event does not match the object set when the event is added.
Is it caused by this commit(https://github.com/lvgl/lvgl/commit/5f90029811e5eb65263ed0d8843aa6be20bcdc7c)?

### To Reproduce
```c
static void root_event_cb(lv_event_t* e)
{
    lv_obj_t* target = lv_event_get_target(e);
    void* user_data = lv_obj_get_user_data(target);
    lv_event_code_t code = lv_event_get_code(e);

    if (!user_data)
    {
        LV_LOG_ERROR("target[%p] event[%d] : user_data is NULL", target, code);
    }
}

lv_obj_t* root_create(void)
{
    lv_obj_t* root = lv_obj_create(lv_scr_act());
    LV_LOG_USER("root[%p]", root);
    lv_obj_set_user_data(root, (void*)"root");
    lv_obj_add_event_cb(root, root_event_cb, LV_EVENT_ALL, NULL);

    lv_obj_t* par = lv_obj_create(root);
    LV_LOG_USER("par[%p]", par);
    lv_obj_t* child = lv_obj_create(par);
    LV_LOG_USER("child[%p]", child);
    return root;
}
```

### Expected behavior
The value of `target` should point to `root`.

### Screenshots or video
[image]

kisvegabor, 2021-11-08:
If the event is bubbled `lv_event_get_target` gets the original target and `lv_event_get_current_target` the current target. 
So using `lv_event_get_current_target` by default would produce a more intuitive result. 

In the practice I also found it confusing however this approach [comes from JavaScript](https://stackoverflow.com/a/10086501).

I'd support to refactor 
- `current_target` -> `target` (`target` the current event listener)
- `target` -> `original_target` (`original_target` the object that triggered the event)

What do you think?
cc @embeddedt

FASTSHIFT, 2021-11-08:
@kisvegabor 
I agree with this approach, it can be compatible with old habits.

embeddedt, 2021-11-08:
Personally, I'd prefer to keep the current names inspired by JavaScript, as they are more standard and there's lots written about them on the web, but I'm also biased as I work with JavaScript on a regular basis.

I'm also not sure how we would change the meaning of `target` and still keep backwards-compatibility.

kisvegabor, 2021-11-08:
> I'm also not sure how we would change the meaning of target and still keep backwards-compatibility.

As event bubbling is typically rarely used in LVGL, usually there is no difference between the different kinds of targets. But as `get_target` would do a different thing, we can not change it in a backward compatible way. 

Regarding the change itself, I think it's an acceptably minor API break, affecting only a few cases, where the fix is straightforward.

Regarding whether the JS or the custom naming is better, I'm not sure. For me `original_target` is more intuitive and more practical because usually `current_target` is used so it seems reasonable to name the common thing with the simpler name. 

So the question is how important it is to follow the JS names. Is it a real scenario that people assume that their JS knowledge can be applied in LVGL? I admit that the similarities in concepts can suggest that LVGL really follows JS naming and a lot of people knows JS to some extent.

 @xiaoxiang781216 what do you think about it?

xiaoxiang781216, 2021-11-08:
I think it depends on the target meaning before we introduce the bubbled event into LVGL.

kisvegabor, 2021-11-09:
The term `target` appeared in the same release as event bubbling. Without bubbling
 `target == original_target == current_target`.

xiaoxiang781216, 2021-11-09:
So it's better to keep the meaning as before.

kisvegabor, 2021-11-09:
Alright, if both you and @embeddedt agree on this let keep the current naming. 

I close this issue.
For solution see https://github.com/lvgl/lvgl/issues/2774#issuecomment-962906152

closed 2021-11-09 by kisvegabor