lvgl/lvgl #2778: Text with gradient

Miguel0101, 2021-11-08:
How to make text with gradient?

embeddedt, 2021-11-08:
There is an [example of how to do it with v7](https://docs.lvgl.io/7.11/widgets/objmask.html#text-mask), but I don't think it was updated to v8. You should be able to do something very similar, but I'm not 100% sure what the equivalent API to `objmask` is in v8. I think you may have to attach an event handler and adjust the masks in there.

Miguel0101, 2021-11-08:
So should I create an event using LV_EVENT_DRAW_MAIN_BEGIN to draw the masked gradient on the label and leave the LV_EVENT_DRAW_MAIN_END empty?

kisvegabor, 2021-11-09:
I've pushed an example. Meanwhile, I found a bug with horizontal gradient drawing so be sure to update lvgl too.

commit cc23afc5 (2021-11-09): feat(example) add text with gradient example

fixes #2778
commit 462fbcbf (2021-11-09): feat(example) add text with gradient example

fixes #2778
closed 2021-11-09 by kisvegabor