lvgl/lvgl #2790: v8.2 discussion

kisvegabor, 2021-11-10:
v8.1 is released so I open this issue to plan v8.2. In order really have a "Planning phase" for the next release I'm about to leave this issue open until 31 December and focus on implementation after that.

I add my current ideas here (see explanation in the next comment), please comment if you already have some features ideas in mind or share your opinion about the planned features.

**Rendering**
- [ ] Abstract render layer #2733
- [ ] Widgets to image buffer
- [ ] 18/24 bit color format support

**Features**
- [ ] Scenes
- [ ] Add/Subtract/exclusive mask types #2668
- [ ] Style components
- [ ] Overflow:visible
- [ ] Hover
- [ ] More grid features 

**Others**
- [ ] Markup language #2428
- [ ] Unit tests #2337
- [ ] Fix/Release scripts
- [ ] SVG support

kisvegabor, 2021-11-10:
### Fix/Release scripts
IMO it has the **highest priority**. I meant these:
- a CI action to merge commits with `fix*` message to `staging/v8.*` branches
- a CI or script to merge `staging/v8.*` to `release/v8.*` periodically or triggered manually

### Widgets to image buffer
Make possible to buffer widgets with all children into a single image to
- speed au rendering at the cost of memory usage 
- allow transformation on any widgets (rotate, scale)

### Scenes
It's a brand new concept about how a UI can be created. It's a very simple but very powerful thing. Each widget has an ` int scene_active` field and there is an `lv_obj_set_scene(obj, secene_id)` function which sends `LV_EVENT_SCENE_UNLOAD`, saves the `scene_id` and sends `LV_EVENT_SCENE_LOAD`.  In other words, the scenes can be considered as "states"  of the widget (i do NOT mean pressed, focused, etc state here). 
Use cases:
- There is a parent that can display 3 type of content, each is considered a scene. On scene load/load you can hide/unhide, fade, etc between the possible contents
- There is a screen with "error" and "normal" scenes: on error some styles are changed, and an error message is displayed. On "normal" these are reverted.
- A list has 2 scenes: hidden, visible. On scene load instead of hiding/unhiding any complex animation can be started.

 The gist is, this way the widgets encapsulate their scenes and there is a uniform API for changing scenes. As the current scene is saved it can be automatically unloaded. Without this feature, the user needed to keep track of the current state to know what to unload, but now it'd be backed into the widgets.

### Add/Subtract/exclusive mask types
Now the masks of LVGL can only subtract from an 0xff filled mask. It'd be great to allow "building back" parts too. 

### Style components
Right now it's a little bit cumbersome to manage differently looking versions of the same widget type. E.g. button, error button, warning, button, border only button, border only error button, rounded button, rounded error button, etc. To manage it effectively there should be  `btn`, `btn_border_only`, `btn_rounded`, `btn_error`, `btn_warning` styles. The combination of these styles need to be added manually to the created buttons. It's possible to create `btn_error_create()` and similar functions but it's also cumbersome. If the "role" of the button changes meanwhile, it's also hard to manage now (i.e. remove styles and add new ones). The idea to solve it looks like this:
```c
uint32_t ERROR_BTN_ROUNDED = lv_component_create_from(LV_COMPONENT_BTN);
lv_component_add_style(ERROR_BTN_ROUNDED, &error_btn_style, LV_PART_MAIN);
lv_component_add_style(ERROR_BTN_ROUNDED, &rounded_btn_style, LV_PART_MAIN);

lv_obj_t * btn = lv_btn_create(parent);
lv_obj_set_component_type(btn, ERROR_BTN_ROUNDED);
```

`lv_obj_set_component_type` can be called any time to change the type and `ERROR_BTN_ROUNDED` can be updated anytime too, all objects will be updated.

The implementation is quite simple: If functions like `lv_obj_style_get_bg_color` doesn't find a style on the object, they check the styles assigned to the component ID too. 
It will save memory and give more performance because the styles with the same selector can be merged in `ERROR_BTN_ROUNDED` and only the component ID needs to be saved in the object, instead of many styles.

The initial list and the features ideas are not written into stone, please tell what you think about them.
 
### Overflow:visible
Same as in HTML

### Hover
Same as in HTML

### More grid features 
I'm specifically thinking about `repeat(auto-fill, minmax( <value> px, 1fr))` but ideas are welcome

### SVG support
See https://github.com/Samsung/thorvg/issues/1014

lishaoenxm, 2021-11-15:
@kisvegabor I've created a new issue [#2800](https://github.com/lvgl/lvgl/issues/2800). Could it be added in this version?

guoweilkd, 2021-11-15:
Modify qrCode lv_qrcode_delete() interface, replace it with lv_obj_del().In other words, rewrite Qrcode using obj template.

kisvegabor, 2021-11-16:
@lishaoenxm 
I'll comment in the opened issue.

@guoweilkd It's quite straightforward, so I've just updated the qr code widget. See 318edd8a3

embeddedt, 2021-11-16:
Sorry for the delay in replying here. Here are my thoughts:

> a CI action to merge commits with fix* message to staging/v8.* branches

I think it might be safer to check for the word "backport" in the commit description, otherwise the CI will end up cherry-picking fixes for a v8.2 feature into the v8.0 and v8.1 branches, which is not what we want.

> add CI to port to automatically update submodules and conf files and minor and patch releases

Are we moving `lv_drivers` into `lvgl` for v8.2? That would allow us to have only a single submodule per port, which is ideal, otherwise we have to keep the two repositories working with each other.

> Style components

It seems to me that all that is needed here is a way of linking styles to a parent style. This is what I'm envisioning:

```c
static lv_style_t btn, btn_rounded, btn_warning, btn_error;

static lv_style_t error_btn_component;

lv_style_init(&error_btn_component);
lv_style_add_child(&error_btn_component, &btn_rounded);
lv_style_add_child(&error_btn_component, &btn_error);
```

Now `error_btn_component` can be used just like any other style, but has no properties of its own.

kisvegabor, 2021-11-16:
No problem :slightly_smiling_face: 

> I think it might be safer to check for the word "backport" in the commit description, otherwise the CI will end up cherry-picking fixes for a v8.2 feature into the v8.0 and v8.1 branches, which is not what we want.

True, although it can happen that the part being fixed already changed in v8.2. But, I agree "backport" is safer and probably will result in fewer issues.

> Are we moving lv_drivers into lvgl for v8.2? 

I think we should.

> It seems to me that all that is needed here is a way of linking styles to a parent style. This is what I'm envisioning:

As styles don't store `part` and `state` e.g. a complete slider can't be described this way where there are main, indicator, and knob parts with any combination of states.

C47D, 2021-11-17:
> 18/24 bit color format support

How hard could it be to add this feature? I'm interested in helping out with this one.

Should we also consider gray scale support?

kisvegabor, 2021-11-18:
> How hard could it be to add this feature? I'm interested in helping out with this one.

I suggest waiting a little with it. We started to work on [abstracting the drawing engines](https://github.com/lvgl/lvgl/pull/2803)and therefore the all drawing parts will be hugely refactored and rewritten. So it should be added once the architecture is stabilized to avoid complicated merging. 

> I'm also working on this, we had a couple of really busy weeks at work, I'm continuing with the new drivers architecture next weekend.

Awesome! It might be also related to the abstraction of drawing. See https://github.com/lvgl/lvgl/pull/2803#issuecomment-971426684

C47D, 2021-11-18:
> I suggest waiting a little with it. We started to work on abstracting the drawing engines and therefore the all drawing parts will be hugely refactored and rewritten. So it should be added once the architecture is stabilized to avoid complicated merging.

Sounds fair, I will focus on the drivers.

puzrin, 2021-11-20:
> Scenes
>
> It's a brand new concept about how a UI can be managed. It's a very simple but very powerful thing.

IMO this looks like mixture of CSS cascading & transforms animation. I would implement more classic features instead of invent new ones.

> Style components
>
> Right now it's a little bit cumbersome to manage differently looking versions of the same widget type. E.g. button, error button, warning button, border only button, border only error button, rounded button, rounded error button, etc. 

I don't understand the problem. https://getbootstrap.com/docs/5.1/components/buttons/ twbs is ok with mixins.

If you need syntax sugar - leave it to CPP wrapper, instead of balloon C api. It's normal to add multiple styles (modifiers) to widget.

> Hover

That's very expensive thing & not available on touch interfaces. IMO added value does not worth efforts. I'd suggest to postpone that as much as possible.

> Markup language

IMO that's standalone project, described as "GUI creator tool with code generator". I'm skeptic about inventig/promoting new languages. Sugar can be partially solved via CPP wrapper. And visual gui builder is usually faster to create with js/html (electron and other).

Anyway, i'd prefer this thing separate from lvgl core. It may take too much efforts and slow down other development.

---

Also, it would be nice to temporary pause with adding new features & focus on pending tails - true RGBA support in styles, benchmarks infrastructure, CPP wrapper and so on.

kisvegabor, 2021-11-22:
> IMO this looks like mixture of CSS cascading & transforms animation. I would implement more classic features instead of invent new ones.

It's more than that and it's new only in LVGL. It's inspired by QT. See [here](https://youtu.be/aV6kFxH3Xws?t=119). 
It helps to make otherwise public states private and encapsulate them into a component resulting in a much-much cleaner implementation. The transition can be part of it, but it's "just" a tool to organize your UI in much cleaner way.

> I don't understand the problem. https://getbootstrap.com/docs/5.1/components/buttons/ twbs is ok with mixins.

It's possible but not convenient. E.g. a React button component is much more powerful where the component's style and logic are encapsulated into one object.  My suggestion is not about logic only styles though. 

> If you need syntax sugar - leave it to CPP wrapper, instead of balloon C api. It's normal to add multiple styles (modifiers) to widget.

The problem came up while I was working on several UI projects. It's just tedious to manage styles as they are now. And if it was inconvenient for me, I suppose it's problematic for others too. 

Of course, the idea can be polished, I just wanted to raise the topic and a possible implementation direction.

> That's very expensive thing & not available on touch interfaces. IMO added value does not worth efforts. I'd suggest to postpone that as much as possible.

I'm personally interested in it because we are planning to support desktop app development too with EdgeLine so, it will come in handy. It needs to be designed in a way to not be expensive. :slightly_smiling_face: 

> IMO that's standalone project, described as "GUI creator tool with code generator". I'm skeptic about inventig/promoting new languages. Sugar can be partially solved via CPP wrapper. And visual gui builder is usually faster to create with js/html (electron and other).

I consider having a markup language very important. I can't mention exact use cases due to NDAs but the need for a platform independent textual UI description came from many sources. It's will also help to find illogical parts in LVGL's UI, and clean it up as much as possible. 

> Also, it would be nice to temporary pause with adding new features & focus on pending tails - true RGBA support in styles, benchmarks infrastructure, CPP wrapper and so on.

RGBA, would be also a new feature. :slightly_smiling_face:  I don't consider this as high priority task as it just joins to properties into one (color+opa). Cpp wrapper is [on its way](https://github.com/lvgl/lv_binding_cpp), but the development slowed down a little in the past month. 
Benchmarking would be great, but some contribution would be also great with it.

embeddedt, 2021-11-23:
Let's discuss the topic of managing old & new releases.

I think what we are thinking of right now (assuming 8.2 is the next release) is the following:

* `master` is the most bleeding-edge branch and has new features, bugfixes, etc.
* `staging/v8.2` is identical to `master`.
* `staging/v8.0` and `staging/v8.1` accumulate any backported bugfixes from `master`.
* At release time, `release/v8.0` is fast-forwarded to `staging/v8.0`, and the same for v8.1.

@kisvegabor You and I had discussed before the idea of requiring every commit to `master` to be a PR with approval.

This could work for automating backporting as well, as I could make an action which tries to backport to a specific branch when someone with appropriate privileges writes a comment like "@lvgl-bot backport v8.0" in the PR.

However, it means that every commit (including simple ones [like these](https://github.com/lvgl/lvgl/commit/81d8be13d67d6b17b663bc703c1e0e18a18890a7)) would need a PR opened. I suppose it would ensure that CI passes on `master` all the time, so maybe it's a good step?

kisvegabor, 2021-11-23:
Meanwhile, I was thinking about if it's really worse the effort to establish a complex backporting system. After v8.0 was released I manually backported some fixes but not that much. As I haven't heard complaints about bugs in v8.0. What if we say that if someone reports a problem in v8.0 (or an older release) we manually fix that problem. Seemingly there won't be too many cases. 

We can find a proper solution if there is a real need for massive backporting (maybe never :slightly_smiling_face: )

What do you think?

embeddedt, 2021-11-23:
True. The majority of the backporting only tends to be needed across major version boundaries, because a decent majority of users will upgrade to the latest 8.x release if asked.

kisvegabor, 2021-11-23:
Great! :) 

The question of a good release script is still open. I've tried to do it for v7 but wasn't confident with it at all. I always run the Python script with a debugger because it always has some problems... 

For v8 IMO the requirements are: 
- updated version number in a few files
- update CHANGELOG
- push a tag
- on minor release:  create a new branch `release/v8.x`
- on patch release:  merge to the related branch 
- also update `lv_demos` and `lv_drivers` in a similar way (the driver will be part of lvgl later)
- create a blog post from the changelog 
- trigger update on other projects (lv_port_...)

xiaoxiang781216, 2021-11-23:
Is it possible to merge lv_demos into lvgl git too? The demo is very useful to verify the porting and performance.

kisvegabor, 2021-11-23:
> Is it possible to merge lv_demos into lvgl git too? The demo is very useful to verify the porting and performance.

With this, the repo structure would be very clear. My only concern is that it'd significantly increase the repo size due to the many assets. `lvgl` is now 40MB, `lv_demos` is 65MB. With drivers and other stuff, we could easily reach 150-200 MB. It's not the end of the world, but something that's worth discussing.

xiaoxiang781216, 2021-11-24:
Yes, but compare with today's disk capacity and network bandwidth, most developer will more enjoy the convenience.:)

kisvegabor, 2021-11-26:
I agree. This way LVGL core, 3rd party lib, example, demo, and driver would be in one place. As part of the CI tests, we could reasily un the stress demo for a while too.

mysterywolf, 2021-11-26:
Is that possable to add `LV_MEMCPY_CUSTOM` and `LV_MEMSET_CUSTOM` to let users or OS to re-implement `lv_memcpy` and `lv_memset` by themselves, just like `LV_SPRINTF_CUSTOM`? RT-Thread has `rt_memset` and `rt_memcpy` in assembly language, which is faster than `lv_memcpy` and `memcpy`(no matter GCC or Keil), and to let users to accelerate the drawing and plotting speed.

kisvegabor, 2021-11-26:
Could we rather integrate the faster versions into LVGL? :smile: 

Can you give a link to your implementation?

mysterywolf, 2021-11-26:
Yes. of course, this is one of the RT-Thread software packages. It's for Cortex-M https://github.com/mysterywolf/rt_memcpy_cm

kisvegabor, 2021-11-26:
> But I didn't translate the readme to English yet.

No problem, Google Translate kindly translated it :slightly_smiling_face: 

> Copy 10,000 bytes 10,000 times in keil environment
rt_memcpy is 1261ms
This compilation version is 883ms
C ​library memcpy is 906ms

Do you still have this test environment? If so could you try LVGL's memcpy?

mysterywolf, 2021-11-26:
I retest under GCC Keil and IAR, and here is the result:

copy 10,000 bytes for 100,000 times (@STM32L745)

| Time Cost (Unit: ms)          | Keil-MDK (-O0) | GCC (-O0) | IAR(-O0) |
| ----------------------------- | -------------- | --------- | -------- |
| rt_memcpy (assembly language) | 8819ms         | 7254ms    | 7227ms   |
| lv_memcpy                     | 17406ms        | 47879ms   | 16951ms  |
| memcpy (C Standard Library)   | 9073ms         | 6940ms    | 10242ms  |

amirgon, 2021-11-26:
@mysterywolf - Why would you do any performance measurement with `-O0` ???

mysterywolf, 2021-11-26:
> @mysterywolf - Why would you do any performance measurement with `-O0` ???

To test the worst case. I want to measure the performance of different memcpy version without help compiler's help. I will test the -O2 case if you want.

amirgon, 2021-11-26:
> To test the worst case. I want to measure the performance of different memcpy version without help compiler's help. I will test the -O2 case if you want.

Comparing performance with `-O0` is meaningless. The compiler makes no effort to achieve performance.  
Showing that one compiler gives better performance than the other with `-O0` does not prove anything. The results could be opposite with `-O2` or `-O3`.

mysterywolf, 2021-11-26:
Done. Here is the -O2 result.

| Time Cost (Unit: ms)          | Keil-MDK (-O2) | GCC (-O2)    | IAR(-O2)     |
| ----------------------------- | -------------- | ------------ | ------------ |
| rt_memcpy (assembly language) | ***8806ms***   | 7212ms       | ***7224ms*** |
| lv_memcpy                     | 9510ms         | 8727ms       | 11056ms      |
| memcpy (C Standard Library)   | 9063ms         | ***6894ms*** | 10237ms      |

mysterywolf, 2021-11-26:
Give me seconds I find there is a mistake.

embeddedt, 2021-11-27:
@amirgon I'm not sure that optimization matters for something simple like a `memcpy` implementation. It makes sense for higher-level code, but `memcpy` is really just two pointers being incremented, dereferenced, and assigned. It tends to be sped up more by manual tricks like copying words instead of bytes.

@mysterywolf Thanks for these measurements.

* I wonder why Keil is so much slower than the other two compilers when `rt_memcpy` isn't even written in C.
* I suspect the only reason GCC/Newlib's implementation wins is because I've found in the past that it's not entirely compatible with all possible use cases of `memcpy`. It speeds up copying with `uint64_t` and `uint32_t` but does not check for alignment first, so it will often crash when pointers are unaligned.

In general `lv_memcpy` seems to be reasonably performant and has the advantage of being 100% portable. However, I see nothing wrong with offering an option to use a custom `memcpy` implementation.

mysterywolf, 2021-11-27:
@embeddedt 
Same like lv_memset, rt_memset also is implemented by C. The assembly rt_memset version (for Cortex-M) only can be used with the corresponding architecture BSP. E.g. if you are using the STM32 series, you can use the assembly version.
You are right. In RT-Thread side, we also find that GCC-Newlib's memcpy is not reliable. When we try to use newlib’s memcpy to operate the hardware registers, such as DMA or SDIO. It’s will crush. So lv_memcpy or rt_memcpy will be safer than newlib’s memcpy. The assembly rt_memcpy version has considered the 4 bytes alignment.

mysterywolf, 2021-11-27:
>In general lv_memcpy seems to be reasonably performant and has the advantage of being 100% portable. However, I see nothing wrong with offering an option to use a custom memcpy implementation.

For sure, We should to keep the lv_memcpy and memcpy simultaneously, and lv_memcpy should be set as default to be safty-guaranteed.  My suggestion is that it would be great that providing a marco to let users can reimplement the lv_memcpy in assembily language for specific CPU to accelerate the speed, e.g. LV_MEMCPY_CUSTOM.

kisvegabor, 2021-11-29:
Glad to see that `lv_memcpy` is not that bad with -O2.  :slightly_smiling_face: 

> I wonder why Keil is so much slower than the other two compilers when rt_memcpy isn't even written in C.

It's really interesting. 

---

I'm ok with `LV_MEMCPY_CUSTOM` and `LV_MEMSET_CUSTOM` but we should maintain the compatibility with the current `LV_MEMCPY_MEMSET_STD`. It can be done via some hardcoded wrappers in `lv_conf_internal.h`.

X-Ryl669, 2021-12-03:
Adding dithering to the list, maybe, since it's a low hanging fruit now ? See #2872

mysterywolf, 2021-12-17:
RT-Thread community has ported LVGL to two different RT-Thread simulators:
https://github.com/RT-Thread/rt-thread/tree/master/bsp/qemu-vexpress-a9
https://github.com/RT-Thread/rt-thread/tree/master/bsp/simulator

Please see this vedio: https://www.youtube.com/watch?v=f7wsl6vTdsY

We are still pushing on the third simulator.

I will translate the homepage's readme to Chinese language version as `README_zh.md` before the next version releases.

tyler-gilbert, 2021-12-27:
How about adding the ability to customize the `LV_SYMBOL_*` values? This would allow using different icon sets without having to change the keyboard icons. It should be super easy to implement, just add some guards to the current `LV_SYMBOL_*` definitions that allow the symbols (either individually or collectively) to be over-ridden by values from `lv_conf..h`

FASTSHIFT, 2021-12-27:
Because the new rendering architecture decouples the rendering logic and hardware, consider transferring GPU-related code (such as `stm32_dma2d`) to `lv_drivers` for separate maintenance. Only software rendering is reserved in the `src/draw` directory.

kisvegabor, 2022-01-05:
@tyler-gilbert 
It makes sens to me. Could you send a PR?

kisvegabor, 2022-01-05:
I've added LV_OBJ_FLAG_OVERFLOW_VISIBLE in e7ac0e419

kisvegabor, 2022-01-05:
According to (hopefully) the final plans EdgeLine will be released in end of January and it'd be great to make it work with the latest LVGL version. The do this there should be a release in few weeks. IMO only the fact that the demos were moved to `lvgl` would be worth a release as it makes the usage of LVGL with Ardunio and  PlatformIO  much easier. Besides #2957 also requires a  new release. 

Of course, we can roll the ideas in the first post to v8.3.

So is there any objections against it?

@embeddedt, @X-Ryl669, @rzr, @brgl, @mysterywolf, @tore-espressif, @MouriNaruto  is there anything you want to be included in v8.2?

rzr, 2022-01-05:
nothing special, do what you think is best to have it released for fosdem ;)

embeddedt, 2022-01-05:
Are we merging the drivers in this release or waiting for 8.3? I don't think it matters too much. Just want to confirm that that's still the plan.

kisvegabor, 2022-01-05:
@embeddedt 
I think it needs more time, thinking and work merge the drivers so I'd keep for a later release.

MouriNaruto, 2022-01-05:
I have no issue because @embeddedt has been asked for you.

Kenji Mouri

tore-espressif, 2022-01-06:
We are getting compile warning and Kconfig warnings when building LVGL in our internal pipeline with strict compiler settings.

I'll publish PR that fixes this, so it can be included in v8.2 after review

X-Ryl669, 2022-01-06:
I think I'm done with the dithering code & gradient code.
I've added a cache for the gradient map computation.
**But** it breaks previous code compatibility if one were to set the `draw_dsc.bg_color` field directly instead of using the style functions. The change is easy (replacing `.bg_color = ` by `.bg_grad.stops[0].color = `, but I don't know the policy for *small* breaking changes and minor version bump.

tyler-gilbert, 2022-01-06:
> @tyler-gilbert It makes sens to me. Could you send a PR?

Yep. Here is the PR: https://github.com/lvgl/lvgl/pull/2973

kisvegabor, 2022-01-07:
@mysterywolf 
> We are promoting more chip vendors to port LVGL to their RT-Thread BSPs. I think Nuvoton has connected with you.

Amazing, thank you very much! Yes, Nuvoton just contacted me. I didn't know you directed to them to me. :slightly_smiling_face: 

@X-Ryl669 
> I think I'm done with the dithering code & gradient code

I just reviewed it, thank you!

@tyler-gilbert 
> Yep. Here is the PR: #2973

Merged, thank you!

lanistor, 2022-01-08:
Should support epaper. [vroland/epdiy](https://github.com/vroland/epdiy) can be used.

C47D, 2022-01-08:
See https://github.com/lvgl/lvgl_esp32_drivers/pull/75 for related topic

GorgonMeducer, 2022-01-10:
Hi, when you do memcpy comparison, please note that:
MDK (Keil) supports three compilers: armcc (Arm Compiler 5), armclang (Arm Compiler 6) and arm gcc. 
When using armcc and armclang, you have to choose between MicroLib and Normal C library. The MicroLib is optimized for size but not performance. The memcpy inside microlib is much slower than the normal version.

So when you do the performance comparison, please make sure to list all whether microlib is used or not.

tore-espressif, 2022-01-14:
> We are getting compile warning and Kconfig warnings when building LVGL in our internal pipeline with strict compiler settings.

So apparently, the compiler warnings are already fixed in current `master` branch. The rest is here https://github.com/lvgl/lvgl/pull/3007. PTAL

X-Ryl669, 2022-01-14:
Is there a screen with ARGB4444 support ? (or ARGB2222 support ?)

ryanpowers, 2022-01-14:
The STM32 DMA2D supports ARGB4444 and ARGB1555. I was interested in 4444
support and tried to hack it in to 7.x, but was ultimately unsuccessful.
The person that started the thread had been interested in 2222, but it
doesn't look like the DMA2D supports it, so I don't know their usage.
Currently, I am using 4444 with a proprietary overlay controller. It is a
good format to save memory when you don't need super high color fidelity
but still need multiple levels of alpha. Currently, I use 8888 and convert
on the fly to 4444 when sending to the controller. Native 4444 would allow
me to buffer twice as much in the same RAM (and also eliminate my
conversion buffer).

mysterywolf, 2022-01-14:
@kisvegabor I suggest that we may create a folder named as `port` or `sys` ,  and move rt-thread, zephyr and cmsis into this folder; otherwise, the root of this repo will be ugly and chaos.

GorgonMeducer, 2022-01-14:
@mysterywolf cmsis-pack hasn't been accepted yet...

kisvegabor, 2022-01-15:
@X-Ryl669 @ryanpowers 
> Is there a screen with ARGB4444 support ? (or ARGB2222 support ?)

Once we rework the colors we should do it a flexible way to allow users create custom color formats too. I wasn't think much about how to do it it well yet but I think it should be the goal.

@mysterywolf 
>  suggest that we may create a folder named as port or sys , and move RT-Thread, Zephyr and CMSIS-package into this folder; otherwise, the root of the repo would be ugly and chaos.

Good idea. The `cmake` folder can go there too. Some other name ideas: `platform`, `support`, `misc`?

GorgonMeducer, 2022-01-15:
@kisvegabor One more question about cmsis-pack:

As you can see, cmsis-pack has its own version, and pack-installer (of MDK) allow people to install and use the version that they want.  Although cmsis-pack is essentially a zip file (yes, you can open it with 7zip), the font file cannot be zipped, hence it contributes at least 3MByte to a 4MByte LVGL pack. 

Do we want to keep old versions of cmsis-packs in the master branch?  or do we just keep the latest version?

Or maybe we can have a dedicated repository to store all versions of cmsis-packs?

kisvegabor, 2022-01-15:
@GorgonMeducer 
I think, in the master branch we can keep updating a single cmsis-pack (or keep only the last version) as LVGL gets new features. And on LVGL release the current cmsis-pack is released too. 

Can it work?

mysterywolf, 2022-01-15:
@kisvegabor I think `platform` would be better; misc and support are a little bit vague.

[41 more replies]