lvgl/lvgl #2792: lv_set_pos() is related to ALIGN instead parent top left corner

smarq8, 2021-11-10:
lv_obj_set_pos is described as set pos related to LU corner but it seems related to current ALIGN method. As it is LV_ALIGN_TOP_LEFT then this demo work as expected.

LVGL v8.0.1

reproduce demo (based on "Make an object draggable" from docs)
```
static void drag_event_handler(lv_event_t * e){
    lv_obj_t * obj = lv_event_get_target(e);

    lv_indev_t * indev = lv_indev_get_act();
    lv_point_t vect;
    lv_indev_get_vect(indev, &vect);
    lv_coord_t x = lv_obj_get_x(obj);
    lv_coord_t y = lv_obj_get_y(obj);

    lv_coord_t x2 = x + vect.x;
    lv_coord_t y2 = y + vect.y;
    lv_obj_set_pos(obj, x2, y2);
    printf("%i %i %i %i %i %i\n",vect.x,vect.y,x,y,x2,y2);
}

void myDemo2(){
  lv_obj_t *obj = lv_obj_create(lv_scr_act());
  lv_obj_set_size(obj,100,100);
  lv_obj_align(obj,LV_ALIGN_CENTER,0,0); // LV_ALIGN_TOP_LEFT work fine
  lv_obj_add_flag(obj,LV_OBJ_FLAG_CLICKABLE);
  lv_obj_add_event_cb(obj, drag_event_handler, LV_EVENT_PRESSING, NULL);
}
```

output with LV_ALIGN_CENTER
```
0 0 17 70 17 70
0 0 34 140 34 140
0 0 51 210 51 210
0 0 51 210 51 210
0 0 68 280 68 280
0 0 85 350 85 350
0 1 102 420 102 421
0 0 119 491 119 491
-1 0 136 561 135 561
-1 0 136 561 135 561
-1 0 152 631 151 631
0 0 168 701 168 701
-1 0 185 771 184 771
0 0 201 841 201 841
0 0 218 911 218 911
```

output with LV_ALIGN_TOP_LEFT
```
0 0 0 0 0 0
0 0 0 0 0 0
0 0 0 0 0 0
0 1 0 0 0 1
1 2 0 1 1 3
1 3 1 3 2 6
3 2 2 6 5 8
1 2 5 8 6 10
1 1 6 10 7 11
1 1 7 11 8 12
2 1 8 12 10 13
0 0 10 13 10 13
0 0 10 13 10 13
0 0 10 13 10 13
0 0 10 13 10 13
```

kisvegabor, 2021-11-12:
This is the intended behavior. See https://docs.lvgl.io/master/overview/coords.html#simple-way

You can use `lv_obj_get_x_aligned` and `lv_obj_get_y_aligned` which returns the set x/y coordinate. Normal `lv_obj_get_x/y` returns the distance from the top left corner.

smarq8, 2021-11-12:
if its intended then probably header description is outdated. Or I misunderstand something?

```
/**
 * Set the relative the position of an object (relative to the parent's top left corner)
 * @param obj       pointer to an object
 * @param x         new distance from the left side of the parent plus the parent's left padding
 * @param y         new distance from the top side of the parent  plus the parent's right padding
 */
void lv_obj_set_pos(struct _lv_obj_t * obj, lv_coord_t x, lv_coord_t y);
```

kisvegabor, 2021-11-12:
Oh, it was really outdated. I've fixed it.

commit b9a5078c (2021-11-12): fix(obj) fix comments of lv_obj_set_pos/x/y

fixes #2792
closed 2021-11-12 by kisvegabor