lvgl/lvgl #2795: lv_theme_default.c expected '}' before 'else'

HaydenHu, 2021-11-11:
**8.1.0**
vscode 
esp32s3

[image]

kisvegabor, 2021-11-11:
Does it work with this code? (else if -> if)

#if LV_USE_CALENDAR
        if(lv_obj_check_type(lv_obj_get_parent(obj), &lv_calendar_class)) {
            /*No style*/
            return;
        }
#endif

HaydenHu ***@***.***> ezt írta (időpont: 2021. nov. 11., Cs,
10:20):

HaydenHu, 2021-11-11:
`[1/157] Building C object esp-idf/lvgl/CMakeFiles/__idf_lvgl.dir/src/extra/themes/default/lv_theme_default.c.obj
FAILED: esp-idf/lvgl/CMakeFiles/__idf_lvgl.dir/src/extra/themes/default/lv_theme_default.c.obj 
D:\Users\Admin\.espressif\tools\xtensa-esp32s3-elf\esp-2021r2-8.4.0\xtensa-esp32s3-elf\bin\xtensa-esp32s3-elf-gcc.exe -DLV_CONF_INCLUDE_SIMPLE -Iconfig -I../components/lvgl -I../components/lvgl/src -I../components -ID:/Users/Admin/esp/esp-idf/components/newlib/platform_include -ID:/Users/Admin/esp/esp-idf/components/freertos/include -ID:/Users/Admin/esp/esp-idf/components/freertos/include/esp_additions/freertos -ID:/Users/Admin/esp/esp-idf/components/freertos/port/xtensa/include -ID:/Users/Admin/esp/esp-idf/components/freertos/include/esp_additions -ID:/Users/Admin/esp/esp-idf/components/esp_hw_support/include -ID:/Users/Admin/esp/esp-idf/components/esp_hw_support/include/soc -ID:/Users/Admin/esp/esp-idf/components/esp_hw_support/include/soc/esp32s3 -ID:/Users/Admin/esp/esp-idf/components/esp_hw_support/port/esp32s3/. -ID:/Users/Admin/esp/esp-idf/components/esp_hw_support/port/esp32s3/private_include -ID:/Users/Admin/esp/esp-idf/components/heap/include -ID:/Users/Admin/esp/esp-idf/components/log/include -ID:/Users/Admin/esp/esp-idf/components/lwip/include/apps -ID:/Users/Admin/esp/esp-idf/components/lwip/include/apps/sntp -ID:/Users/Admin/esp/esp-idf/components/lwip/lwip/src/include -ID:/Users/Admin/esp/esp-idf/components/lwip/port/esp32/include -ID:/Users/Admin/esp/esp-idf/components/lwip/port/esp32/include/arch -ID:/Users/Admin/esp/esp-idf/components/soc/include -ID:/Users/Admin/esp/esp-idf/components/soc/esp32s3/. -ID:/Users/Admin/esp/esp-idf/components/soc/esp32s3/include -ID:/Users/Admin/esp/esp-idf/components/hal/esp32s3/include -ID:/Users/Admin/esp/esp-idf/components/hal/include -ID:/Users/Admin/esp/esp-idf/components/hal/platform_port/include -ID:/Users/Admin/esp/esp-idf/components/esp_rom/include -ID:/Users/Admin/esp/esp-idf/components/esp_rom/include/esp32s3 -ID:/Users/Admin/esp/esp-idf/components/esp_rom/esp32s3 -ID:/Users/Admin/esp/esp-idf/components/esp_common/include -ID:/Users/Admin/esp/esp-idf/components/esp_system/include -ID:/Users/Admin/esp/esp-idf/components/esp_system/port/soc -ID:/Users/Admin/esp/esp-idf/components/esp_system/port/public_compat -ID:/Users/Admin/esp/esp-idf/components/xtensa/include -ID:/Users/Admin/esp/esp-idf/components/xtensa/esp32s3/include -ID:/Users/Admin/esp/esp-idf/components/driver/include -ID:/Users/Admin/esp/esp-idf/components/driver/esp32s3/include -ID:/Users/Admin/esp/esp-idf/components/esp_pm/include -ID:/Users/Admin/esp/esp-idf/components/esp_ringbuf/include -ID:/Users/Admin/esp/esp-idf/components/efuse/include -ID:/Users/Admin/esp/esp-idf/components/efuse/esp32s3/include -ID:/Users/Admin/esp/esp-idf/components/vfs/include -ID:/Users/Admin/esp/esp-idf/components/esp_wifi/include -ID:/Users/Admin/esp/esp-idf/components/esp_event/include -ID:/Users/Admin/esp/esp-idf/components/esp_netif/include -ID:/Users/Admin/esp/esp-idf/components/esp_eth/include -ID:/Users/Admin/esp/esp-idf/components/tcpip_adapter/include -ID:/Users/Admin/esp/esp-idf/components/esp_phy/include -ID:/Users/Admin/esp/esp-idf/components/esp_phy/esp32s3/include -ID:/Users/Admin/esp/esp-idf/components/esp_ipc/include -ID:/Users/Admin/esp/esp-idf/components/app_trace/include -ID:/Users/Admin/esp/esp-idf/components/esp_timer/include -mlongcalls -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -fmacro-prefix-map=D:/HAYDEN/MCU/ESP/ESP32S3M=. -fmacro-prefix-map=D:/Users/Admin/esp/esp-idf=IDF -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -fno-jump-tables -fno-tree-switch-conversion -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v5.0-dev-75-g1561fbd2c5-dirty\" -DESP_PLATFORM -MD -MT esp-idf/lvgl/CMakeFiles/__idf_lvgl.dir/src/extra/themes/default/lv_theme_default.c.obj -MF esp-idf\lvgl\CMakeFiles\__idf_lvgl.dir\src\extra\themes\default\lv_theme_default.c.obj.d -o esp-idf/lvgl/CMakeFiles/__idf_lvgl.dir/src/extra/themes/default/lv_theme_default.c.obj -c ../components/lvgl/src/extra/themes/default/lv_theme_default.c
../components/lvgl/src/extra/themes/default/lv_theme_default.c: In function 'theme_apply':
../components/lvgl/src/extra/themes/default/lv_theme_default.c:671:5: error: expected '}' before 'else'
     else if(lv_obj_check_type(lv_obj_get_parent(obj), &lv_calendar_class)) {
     ^~~~
../components/lvgl/src/extra/themes/default/lv_theme_default.c: At top level:
../components/lvgl/src/extra/themes/default/lv_theme_default.c:682:5: error: expected identifier or '(' before 'else'
     else if(lv_obj_check_type(obj, &lv_btn_class)) {
     ^~~~
../components/lvgl/src/extra/themes/default/lv_theme_default.c:698:5: error: expected identifier or '(' before 'else'
     else if(lv_obj_check_type(obj, &lv_line_class)) {
     ^~~~
../components/lvgl/src/extra/themes/default/lv_theme_default.c:704:5: error: expected identifier or '(' before 'else'
     else if(lv_obj_check_type(obj, &lv_btnmatrix_class)) {
     ^~~~
../components/lvgl/src/extra/themes/default/lv_theme_default.c:759:5: error: expected identifier or '(' before 'else'
     else if(lv_obj_check_type(obj, &lv_bar_class)) {
     ^~~~
[... cut ...]
../components/lvgl/src/extra/themes/default/lv_theme_default.c:1045:1: error: expected identifier or '(' before '}' token
 }
 ^
[6/157] Performing build step for 'bootloader'
[1/1] cmd.exe /C "cd /D D:\HAYDEN\MCU\ESP\ESP32S3M\build\bootloader\esp-idf\esptool_py && D:\Users\Admin\.espressif\python_env\idf5.0_py3.8_env\Scripts\python.exe D:/Users/Admin/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 D:/HAYDEN/MCU/ESP/ESP32S3M/build/bootloader/bootloader.bin"
Bootloader binary size 0x4f70 bytes. 0x3090 bytes (38%) free.
ninja: build stopped: subcommand failed.`

mariotaku, 2021-11-14:
@kisvegabor Looks like this is a syntax error in v8.1.0 which has been fixed in 075831a54c30d294879619c90ca4d16676c0775a.

kisvegabor, 2021-11-16:
@HaydenHu was it really fixed by the mentioned commit?

closed 2021-12-09 by lvgl-bot