lvgl/lvgl #2800: The border could be set border-style, border-width, border-color, border-radius respectively by the left, top, right, bottom

lishaoenxm, 2021-11-15:
### Introduce the problem

Currently the LVGL object could only be set same border-style, border-width, border-color, border-radius for all the border. But for the web developers, they often make use of setting different border style in one object. 

### Examples

[image]

### Suggested solution

The LVGL object only support one attribution _LV_PART_MAIN_ for the border. LV_PART_LEFT, LV_PART_TOP, LV_PART_RIGHT and LV_PART_BOTTOM should be defined for the style, color, width and radius. And the current style and attribution data struct also support these kind of data pair. But for the drawing function, it has a little of trouble.

Miguel0101, 2021-11-15:
I agree, it would be great to add `LV_PART_TOP/BOTTOM/LEFT/RIGHT` attributions.

kisvegabor, 2021-11-16:
With our designer, we talk a lot about things like this and it seems to us a different approach could be more flexible. It's rather like `:after` and `:before` with 2 differences:
1. You can place them only on the object itself
2. You can have any number of these extra parts

In other words, it's like an effect stack to tell LVGL what other effects to draw after/before the normal object was drawn.
An extreme example for this effect stack:

0. draw the normal object (not part of the stack)
1. draw a small intense red shadow
2. draw a larger lighter rose shadow
3. draw a green left border with additive blending
4. draw a background image with multiply blending
5. draw a purple outline close to the object
6. draw a yellow outline with 10 px padding

Even with full CSS support, these can be achieved only with some extra wrappers obejcts.

The effect stack should work on all parts and states, not only the main part. E.g. you can attach it to the pressed state of the slider's knob. 

What do you think about this approach? 

However, there is one thing that should be supported internally in LVGL: allow different radius on each corner. At first look an API like this can work (assuming `LV_USE_LARGE_COORD`):
```c
#define LV_RADIUS4(a, b, c, d)   (a << 0  +  b << 7  +  c << 14 + d << 21)

lv_style_set_radius(&style, LV_RADIUS4(10,  20,  30, 40));
```

lishaoenxm, 2021-11-18:
@kisvegabor yes, great!

But for the radius, there is a question. 
`#define LV_RADIUS4(a, b, c, d)   (a << 0  +  b << 7  +  c << 14 + d << 21)`

The a, b, c , d only have 7 bits. It means each radius is less than 127. 

There are 2 proposals,
1. Make use of a data structure.
` struct lv_radius {
  lv_coord_t top;
  lv_coord_t right;
  lv_coord_t bottom;
  lv_coord_t left;
}; 

struct lv_radius;
lv_style_set_radius(&style, &lv_radius)`

Miguel0101, 2021-11-18:
> @kisvegabor yes, great!
> 
> But for the radius, there is a question. `#define LV_RADIUS4(a, b, c, d) (a << 0 + b << 7 + c << 14 + d << 21)`
> 
> The a, b, c , d only have 7 bits. It means each radius is less than 127.
> 
> There are 2 proposals,
> 
> 1. Make use of a data structure.
> 
> ```
> struct lv_radius {
>   lv_coord_t top;
>   lv_coord_t right;
>   lv_coord_t bottom;
>   lv_coord_t left;
> }; 
> 
> struct lv_radius;
> lv_style_set_radius(&style, &lv_radius)
> ```
> 
> 1. Make use of the LV_PART
>    `lv_style_set_radius(&style, 32, LV_PART_LEFT)`

Good idea.

>    `lv_style_set_radius(&style, 32, LV_PART_LEFT)`

Will the LVGL Style API change in the next release?

kisvegabor, 2021-11-18:
> The a, b, c , d only have 7 bits. It means each radius is less than 127.

True we need something better. :slightly_smiling_face: 

> Make use of a data structure.

It can work, but we still should allow easily setting the radius for all corners.

> Make use of the LV_PART

If it were related to the main part we couldn't set the radius of e.g. a slider knob.

> Will the LVGL Style API change if we choose the second proposal?

Only in a backward-compatible way.

lishaoenxm, 2021-12-01:
@kisvegabor 
Could we add a new interface _lv_style_set_radius_ex(&style, 32, LV_PART_LEFT)_? It could solve the backward-compatible issue.

kisvegabor, 2021-12-02:
As I mentioned we can't use `LV_PART_LEFT` as it should work on `LV_PART_MAIN/KNOB/INDICATOR/etc` too.

So what about
```c
lv_style_set_radius_left(style, 10, SELECTOR)
lv_style_set_radius_right(style, 10, SELECTOR)
lv_style_set_radius_top(style, 10, SELECTOR)
lv_style_set_radius_left(style, 10, SELECTOR)
```

where `lv_style_set_radius` is just a wrapper to the all 4 above.

lishaoenxm, 2021-12-07:
@kisvegabor Yes, it's a solution.

How about the other interfaces? Like these,

_lv_style_set_border_color_left(style, RED, SELECTOR)
lv_style_set_border_color_right(style, RED, SELECTOR)
lv_style_set_border_color_top(style, RED, SELECTOR)
lv_style_set_border_color_bottom(style, RED, SELECTOR)

lv_style_set_border_width_left(style, 1, SELECTOR)
lv_style_set_border_width_right(style, 1, SELECTOR)
lv_style_set_border_width_top(style, 1, SELECTOR)
lv_style_set_border_width_bottom(style, 1, SELECTOR)

lv_style_set_border_style_left(style, solid, SELECTOR)
lv_style_set_border_style_right(style, dot, SELECTOR)
lv_style_set_border_style_top(style, dot, SELECTOR)
lv_style_set_border_style_bottom(style, dot, SELECTOR)

...._

@Miguel0101 What do you think of?

kisvegabor, 2021-12-07:
Setting different widths and styles seems a little bit comélicated for our resource-constrained SW rendered. So for first I'd skip these.

lishaoenxm, 2021-12-09:
Our web developers eagerly ask for these features...

OK. Let's do them one by one.

Miguel0101, 2021-12-18:
I agree with @kisvegabor.

kisvegabor, 2024-06-20:
I'm closing this issue as there no activity here recently. 

I still believe different border width and style would be too complicated, but different radius on corner would be possible.

closed 2024-06-20 by kisvegabor

Finomnis, 2025-08-11:
If not yet supported, I agree with @jf-uu .