lvgl/lvgl #2806: lv_img_set_zoom cause a wrong widget size in LV_IMG_SIZE_MODE_REAL

seasonyuu, 2021-11-17:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

When using `LV_IMG_SIZE_MODE_REAL`, the widget size should be change to expect size , but it didn't.

### To Reproduce

```c
        lv_obj_t *img = lv_img_create(container);
        lv_obj_set_size(img, LV_SIZE_CONTENT, LV_SIZE_CONTENT);
        lv_img_set_size_mode(img, LV_IMG_SIZE_MODE_REAL);
        lv_img_set_antialias(img, true);
        lv_obj_set_style_pad_all(img, 0, 0);
        lv_img_set_pivot(img, 0, 0);
        lv_img_set_zoom(img, 128);
        lv_img_set_src(img, &img_source);
```

### Expected behavior

When I try to scale a image to half size, it doesn't perform as expected. I'm use a image source as 66x66, when i do `lv_img_set_zoom(img, 128);` after `LV_EVENT_DRAW_MAIN_END` , the widget size is 37x37.

And I'm trying to find out why it is, then I find that the related part is the `_lv_img_buf_get_transformed_area` function in [lv_img_buf.c](https://github.com/lvgl/lvgl/blob/v8.1.0/src/draw/lv_img_buf.c#L474)

```c
...
void _lv_img_buf_get_transformed_area(lv_area_t * res, lv_coord_t w, lv_coord_t h, int16_t angle, uint16_t zoom,
                                      const lv_point_t * pivot)
{
#if LV_DRAW_COMPLEX
    if(angle == 0 && zoom == LV_IMG_ZOOM_NONE) {
        res->x1 = 0;
        res->y1 = 0;
        res->x2 = w - 1;
        res->y2 = h - 1;
        return;
    }

    res->x1 = (((int32_t)(-pivot->x) * zoom) >> 8) - 1;
    res->y1 = (((int32_t)(-pivot->y) * zoom) >> 8) - 1;
    res->x2 = (((int32_t)(w - pivot->x) * zoom) >> 8) + 2;
    res->y2 = (((int32_t)(h - pivot->y) * zoom) >> 8) + 2;
....
```
These `-1` , `+2` things make me confused. But when I modified a little code like this, everything is working correctly.

```c
...
void _lv_img_buf_get_transformed_area(lv_area_t * res, lv_coord_t w, lv_coord_t h, int16_t angle, uint16_t zoom,
                                      const lv_point_t * pivot)
{
#if LV_DRAW_COMPLEX
    if(angle == 0 && zoom == LV_IMG_ZOOM_NONE) {
        res->x1 = 0;
        res->y1 = 0;
        res->x2 = w - 1;
        res->y2 = h - 1;
        return;
    }

    res->x1 = (((int32_t)(-pivot->x) * zoom) >> 8);
    res->y1 = (((int32_t)(-pivot->y) * zoom) >> 8);
    res->x2 = (((int32_t)(w - pivot->x) * zoom) >> 8) - 1;
    res->y2 = (((int32_t)(h - pivot->y) * zoom) >> 8) - 1;
...
```

After these change, when I do `lv_img_set_zoom(img, 128);` , the widget size finnally change from 66x66 to 33x33.

### Screenshots or video

kisvegabor, 2021-11-18:
Thanks for pointing this out. These corrections were done to avoid some sneaky rounding errors. 

We are reworking the drawing backend and will review this part too when we get there. https://github.com/lvgl/lvgl/pull/2803

seasonyuu, 2021-11-19:
> Thanks for pointing this out. These corrections were done to avoid some sneaky rounding errors.
> 
> We are reworking the drawing backend and will review this part too when we get there. #2803

I'm very happy to see things going to be better, I'll waiting the 8.2 tag and try again

closed 2021-12-06 by seasonyuu