lvgl/lvgl #2809: lv_deinit() -> lv_init() causes crash lv_style_get_prop_inlined

geert-KLA-BE, 2021-11-18:
### Perform all steps below and tick them with [x]
- [ X] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ X] Update lvgl to the latest version
- [ X] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

For our project on an embedded board we need to be able to completly reinitialize lvgl into a known state.
To do this we try to use lv_deinit() and later on initialize everything again as first (using lv_init(), ...). 
When we do this we get a crash with a nullptr in lv_style_get_prop_inlined.
`    if(style->is_const) {
        const lv_style_const_prop_t * const_prop;
        for(const_prop = style->v_p.const_props; const_prop->prop != LV_STYLE_PROP_INV; const_prop++) {
`
Unhandled exception thrown: read access violation.
const_prop was nullptr.

### To Reproduce

I have used the modified lv_port_windows.
 
```c

struct lvgl_handle_t
{
    lv_disp_drv_t                   disp_drv;
    lv_disp_draw_buf_t              disp_buf;
    lv_indev_drv_t                  indev_drv;
    lv_indev_drv_t                  kb_drv;
    lv_indev_drv_t                  enc_drv;
    std::array<lv_color_t, 65536>*  raw_disp_buf_p;
    lv_group_t* volatile            g_DefaultGroup;

    lvgl_handle_t()
    {
        raw_disp_buf_p = new std::array<lv_color_t, 65536>();
    }

    ~lvgl_handle_t()
    {
        delete raw_disp_buf_p;
    }
};

static lvgl_handle_t* current_handle_p{ nullptr };
void lvgl_set_current_handle( lvgl_handle_t* handle_p )
{
    current_handle_p = handle_p;
}

lvgl_handle_t* lvgl_get_current_handle()
{
    return current_handle_p;
}

bool LvglWindowsInitialize(
    _In_ HINSTANCE hInstance,
    _In_ int nShowCmd,
    _In_opt_ LPCWSTR DefaultFontName)
{
    ::LvglWindowsGdiFontInitialize(DefaultFontName);

    HICON IconHandle = ::LoadIconW(
        ::GetModuleHandleW(nullptr),
        MAKEINTRESOURCE(IDI_LVGL));

    WNDCLASSEXW WindowClass;

    WindowClass.cbSize = sizeof(WNDCLASSEXW);

    WindowClass.style = 0;
    WindowClass.lpfnWndProc = ::WndProc;
    WindowClass.cbClsExtra = 0;
    WindowClass.cbWndExtra = 0;
    WindowClass.hInstance = hInstance;
    WindowClass.hIcon = IconHandle;
    WindowClass.hCursor = ::LoadCursorW(nullptr, IDC_ARROW);
    WindowClass.hbrBackground = reinterpret_cast<HBRUSH>(COLOR_WINDOW + 1);
    WindowClass.lpszMenuName = nullptr;
    WindowClass.lpszClassName = L"lv_port_windows";
    WindowClass.hIconSm = IconHandle;

    if (!::RegisterClassExW(&WindowClass))
    {
        return false;
    }
    g_InstanceHandle = hInstance;

    g_WindowHandle = ::CreateWindowExW(
        WS_EX_CLIENTEDGE,
        WindowClass.lpszClassName,
        L"LVGL ported to Windows Desktop",
        WS_OVERLAPPEDWINDOW,
        CW_USEDEFAULT,
        0,
        CW_USEDEFAULT,
        0,
        nullptr,
        nullptr,
        hInstance,
        nullptr);

    if (!g_WindowHandle)
    {
        return false;
    }

    ::LvglRegisterTouchWindow(g_WindowHandle, 0);

    Mile::EnableChildWindowDpiMessage(g_WindowHandle);
    g_WindowDPI = ::LvglGetDpiForWindow(g_WindowHandle);

    ::ShowWindow(g_WindowHandle, nShowCmd);
    ::UpdateWindow(g_WindowHandle);
   
    return true;
}

void lvgl_initialize(lvgl_handle_t& handle)
{
    lv_disp_drv_init(&handle.disp_drv);
    lv_disp_draw_buf_init(
        &handle.disp_buf,
        handle.raw_disp_buf_p->data(),
        nullptr,
        65536
    );
    handle.disp_drv.hor_res = static_cast<lv_coord_t>(g_WindowWidth);
    handle.disp_drv.ver_res = static_cast<lv_coord_t>(g_WindowHeight);
    handle.disp_drv.flush_cb = LvglDisplayDriverFlushCallback;
    handle.disp_drv.draw_buf = &handle.disp_buf;
    handle.disp_drv.dpi = g_WindowDPI;
    lv_disp_drv_register(&handle.disp_drv);

    handle.g_DefaultGroup = lv_group_create();
    lv_group_set_default(handle.g_DefaultGroup);

    lv_indev_drv_init(&handle.indev_drv);
    handle.indev_drv.type = LV_INDEV_TYPE_POINTER;
    handle.indev_drv.read_cb = LvglMouseDriverReadCallback;
    lv_indev_set_group(
        lv_indev_drv_register(&handle.indev_drv),
        handle.g_DefaultGroup);

    lv_indev_drv_init(&handle.kb_drv);
    handle.kb_drv.type = LV_INDEV_TYPE_KEYPAD;
    handle.kb_drv.read_cb = LvglKeyboardDriverReadCallback;
    lv_indev_set_group(
        lv_indev_drv_register(&handle.kb_drv),
        handle.g_DefaultGroup);

    lv_indev_drv_init(&handle.enc_drv);
    handle.enc_drv.type = LV_INDEV_TYPE_ENCODER;
    handle.enc_drv.read_cb = ::LvglMousewheelDriverReadCallback;
    lv_indev_set_group(
        lv_indev_drv_register(&handle.enc_drv),
        handle.g_DefaultGroup);
}

int WINAPI wWinMain(
    _In_ HINSTANCE hInstance,
    _In_opt_ HINSTANCE hPrevInstance,
    _In_ LPWSTR lpCmdLine,
    _In_ int nShowCmd)
{
    UNREFERENCED_PARAMETER(hPrevInstance);
    UNREFERENCED_PARAMETER(lpCmdLine);

    if (!LvglWindowsInitialize(
        hInstance,
        nShowCmd,
        nullptr))
    {
        return -1;
    }

    while(1)
    {
        lv_init();
        {
            lvgl_handle_t lv_handle{};
            lvgl_initialize( lv_handle );
            lvgl_set_current_handle( &lv_handle );
            //lv_demo_widgets();
            //lv_demo_keypad_encoder();
            //lv_demo_benchmark();
            auto thrd = std::thread( LvglTaskSchedulerLoop );
            Sleep( 10000 );
            g_WindowQuitSignal = true;
            thrd.join();
            lvgl_set_current_handle( nullptr );
        } // Destroys the handle & frees buffer memory
        lv_deinit();
    }
    return ::LvglWindowsLoop();
}
```

### Expected behavior

The display restarts again correctly. No crashes.

### Screenshots or video

kisvegabor, 2021-11-26:
Hi,

Please send a call stack trace to see exactly where the crash happened.

geert-KLA-BE, 2021-11-29:
```
>	LVGL.Windows.Desktop.exe!lv_style_get_prop_inlined(lv_style_t * style, lv_style_prop_t prop, lv_style_value_t * value) Line 357	C
 	LVGL.Windows.Desktop.exe!lv_style_get_prop(lv_style_t * style, lv_style_prop_t prop, lv_style_value_t * value) Line 214	C
 	LVGL.Windows.Desktop.exe!get_prop_core(const _lv_obj_t * obj, unsigned int part, lv_style_prop_t prop, lv_style_value_t * v) Line 580	C
 	LVGL.Windows.Desktop.exe!lv_obj_get_style_prop(const _lv_obj_t * obj, unsigned int part, lv_style_prop_t prop) Line 220	C
 	LVGL.Windows.Desktop.exe!lv_obj_get_style_shadow_width(const _lv_obj_t * obj, unsigned int part) Line 418	C
 	LVGL.Windows.Desktop.exe!lv_obj_calculate_ext_draw_size(_lv_obj_t * obj, unsigned int part) Line 295	C
 	LVGL.Windows.Desktop.exe!lv_obj_event(const _lv_obj_class_t * class_p, _lv_event_t * e) Line 838	C
 	LVGL.Windows.Desktop.exe!lv_obj_event_base(const _lv_obj_class_t * class_p, _lv_event_t * e) Line 97	C
 	LVGL.Windows.Desktop.exe!event_send_core(_lv_event_t * e) Line 415	C
 	LVGL.Windows.Desktop.exe!lv_event_send(_lv_obj_t * obj, lv_event_code_t event_code, void * param) Line 74	C
 	LVGL.Windows.Desktop.exe!lv_obj_refresh_ext_draw_size(_lv_obj_t * obj) Line 343	C
 	LVGL.Windows.Desktop.exe!lv_obj_refresh_style(_lv_obj_t * obj, unsigned int selector, lv_style_prop_t prop) Line 193	C
 	LVGL.Windows.Desktop.exe!lv_obj_class_init_obj(_lv_obj_t * obj) Line 112	C
 	LVGL.Windows.Desktop.exe!lv_obj_create(_lv_obj_t * parent) Line 212	C
 	LVGL.Windows.Desktop.exe!lv_disp_drv_register(_lv_disp_drv_t * driver) Line 165	C
 	LVGL.Windows.Desktop.exe!lvgl_initialize(lvgl_handle_t & handle) Line 743	C++
 	LVGL.Windows.Desktop.exe!wWinMain(HINSTANCE__ * hInstance, HINSTANCE__ * hPrevInstance, wchar_t * lpCmdLine, int nShowCmd) Line 844	C++
 	[External Code]	
 	kernel32.dll![Frames below may be incorrect and/or missing, no symbols loaded for kernel32.dll]	Unknown
```

<!DOCTYPE html>
<HTML>
<body>

  | Name | Value | Type
-- | -- | -- | --
◢ | const_prop | 0x00000000 <NULL> | const lv_style_const_prop_t *
  | prop | <Unable to read memory> |  
  | ▶ value | {num=??? ptr=??? color={ch={blue=??? green=??? red=??? ...} full=??? } } | lv_style_value_t
  | prop | LV_STYLE_SHADOW_WIDTH (2112) | lv_style_prop_t
◢ | style | LVGL.Windows.Desktop.exe!0x00d3d158 {sentinel=2864434396 v_p={value1={num=0 ptr=0x00000000 color={ch=...} } ...} ...} | lv_style_t *
  | sentinel | 2864434396 | unsigned int
  | ▶ v_p | {value1={num=0 ptr=0x00000000 color={ch={blue=0 '\0' green=0 '\0' red=0 '\0' ...} full=0 } } values_and_props=...} | lv_style_t::<unnamed-type-v_p>
  | prop1 | 20336 | unsigned short
  | is_const | 1 | unsigned short
  | has_group | 211 'Ó' | unsigned char
  | prop_cnt | 0 '\0' | unsigned char
◢ | value | 0x0073e9ec {num=-858993460 ptr=0xcccccccc color={ch={blue=204 'Ì' green=204 'Ì' red=204 'Ì' ...} full=...} } | lv_style_value_t *
  | num | -858993460 | int
  | ptr | 0xcccccccc | const void *
  | ▶ color | {ch={blue=204 'Ì' green=204 'Ì' red=204 'Ì' ...} full=3435973836 } | lv_color32_t

</body>
</HTML>

geert-KLA-BE, 2021-11-29:
I've pushed the sample to reproduce here:
https://github.com/geert-KLA-BE/lv_indev_drv_update_crash

kisvegabor, 2021-11-30:
I have an idea what can be the problem. Please chnage `lv_theme_default_init` to this:
```c
lv_theme_t * lv_theme_default_init(lv_disp_t * disp, lv_color_t color_primary, lv_color_t color_secondary, bool dark,
                                   const lv_font_t * font)
{

    /*This trick is required only to avoid the garbage collection of
     *styles' data if LVGL is used in a binding (e.g. Micropython)
     *In a general case styles could be in simple `static lv_style_t my_style...` variables*/
    if(LV_GC_ROOT(_lv_theme_default_styles) == NULL) {
        inited = false;
        LV_GC_ROOT(_lv_theme_default_styles) = lv_mem_alloc(sizeof(my_theme_styles_t));
        styles = (my_theme_styles_t *)LV_GC_ROOT(_lv_theme_default_styles);
    }

    if(LV_HOR_RES <= 320) disp_size = DISP_SMALL;
    else if(LV_HOR_RES < 720) disp_size = DISP_MEDIUM;
    else disp_size = DISP_LARGE;

    theme.disp = disp;
    theme.color_primary = color_primary;
    theme.color_secondary = color_secondary;
    theme.font_small = font;
    theme.font_normal = font;
    theme.font_large = font;
    theme.apply_cb = theme_apply;
    theme.flags = dark ? MODE_DARK : 0;

    style_init();

    inited = true;

    if(disp == NULL || lv_disp_get_theme(disp) == &theme) lv_obj_report_style_change(NULL);

    return (lv_theme_t *)&theme;
}
```

geert-KLA-BE, 2021-12-01:
I don't see a difference with the change. Still nullptr at the same location.

geert-KLA-BE, 2021-12-02:
When further debugging the issue I notice that not the init path is chosen but the get path of the default theme:
```C
    if(lv_theme_default_is_inited() == false) {
        disp->theme = lv_theme_default_init(disp, lv_palette_main(LV_PALETTE_BLUE), lv_palette_main(LV_PALETTE_RED),
                                            LV_THEME_DEFAULT_DARK, LV_FONT_DEFAULT);
    } else {
        disp->theme = lv_theme_default_get(); <--- default theme is still thought to exist
    }
```

kisvegabor, 2021-12-08:
Ah, I see. 

I belive this change should fix it:
```c

bool lv_theme_default_is_inited(void)
{
    return  LV_GC_ROOT(_lv_theme_default_styles) == NULL ? false : true;
}
```

It doesn't matter if these static variables remain initialized if their value is updated on the next init. 
The roots (garbage collection roots from where all other mallocs start) are collected in `lv_gc.c/h` and they are [cleaned in lv_deint](https://github.com/lvgl/lvgl/blob/master/src/core/lv_obj.c#L196).

geert-KLA-BE, 2021-12-17:
Seems the crash is gone now. But it doesn't update the screen anymore after the second startup.

geert-KLA-BE, 2021-12-17:
Seems like the culprit is that the label is a static object pointer that doesn't get reset.
```C
    static lv_obj_t * perf_label = NULL;
    if(perf_label == NULL) {
        perf_label = lv_label_create(lv_layer_sys());
```

geert-KLA-BE, 2021-12-17:
I've made a pull request with some fixes already #2910 
I will need to further investigate if there are other comparable issues.

kisvegabor, 2021-12-20:
#2910 is merged so I close this issue. 

Please, let us know if you find any other suspicious places.

PR #2910 (2021-12-20): Fix lv_deinit lv_init crash or hang
### Description of the feature or fix
#2809

### Checkpoints
- [ ] Follow the [styling guide](https://github.com/lvgl/lvgl/blob/master/docs/CODING_STYLE.md)
- [ ] Run `code-format.py` from the `scripts` folder. [astyle](http://astyle.sourceforge.net/install.html) needs to be installed.
- [ ] Update the documentation
closed 2021-12-20 by kisvegabor