lvgl/lvgl #2814: How to blur background canvas?

Miguel0101, 2021-11-19:
How to blur background canvas with transparency?

kisvegabor, 2021-11-19:
It should work with the alpha channel too. If you see any issues please send a code snippet to reproduce.

Miguel0101, 2021-11-19:
> It should work with the alpha channel too. If you see any issues please send a code snippet to reproduce.

I want to blur the objects behind the canvas object.

kisvegabor, 2021-11-22:
There is no such blur feature at this moment. You can take a snapshot of the bg widget with `lv_snapshot`, and blur it with the canvas and use it as the background of your widget. 

I know it's quite cumbersome, but that's what we have now :slightly_frowning_face:

closed 2021-12-15 by lvgl-bot

joelsand, 2023-05-08:
This would be a great feature