lvgl/lvgl #2826: Saving LVGL render state to backup RAM before sleep & restore after wake-up?

huming2207, 2021-11-22:
### Introduce the problem

I'm not sure if LVGL has a similar API for dumping out all the object render states from the main RAM region to backup storage (e.g. external low-power RAM like FRAM, RTC RAM or even SPI flash etc). It would be great if we have that already, or if not, are there any hints or suggestions for implementing something like that?

### Examples and cases

This can be useful for any low-power use cases, particularly the e-paper & memory LCD display and other power budget sensitive products. We then don't need the "lighter" sleep modes for spending extra power on full RAM retention but put the microcontroller into the "deeper" sleep modes, or even shut down the microcontroller.

### Suggested solution

Assume we have a `backup_buffer` and a `has_backup` flag being put to RTC RAM region, and it will be retained in deep sleep mode.

#### Before sleep

```c
static RTC_RAM_ATTR uint8_t backup_buffer[32768];
static RTC_RAM_ATTR bool has_backup;
// ...

lv_obj_backup(lv_scr_act(), backup_buffer, sizeof(backup_buffer));
has_backup = true;
enter_deep_sleep(); 
```

#### After sleep

```c
lv_init();
// ...some setup code for driver here

if (has_backup) {
  lv_obj_restore(lv_scr_act(), backup_buffer, sizeof(backup_buffer));
  has_backup = false;
} else {
  render_ui(); // If there's no backup, then we render the UI instead.
}
```

embeddedt, 2021-11-23:
LVGL itself doesn't have a good method of doing this. Even if you managed to fit the entire heap into RTC RAM, there are still some globals that wouldn't be saved.

The cleanest solution I can think of is to design your UI in such a way that the state is kept outside of LVGL, then you can save the state in the backup buffer and recreate all the widgets from scratch when the system resumes.

xiaoxiang781216, 2021-11-23:
Or, you can save/restore work_mem_int if you use the internal heap manager:
https://github.com/lvgl/lvgl/blob/master/src/misc/lv_mem.c#L93-L94
Of course, you need save/restore the global and stack memory by self if they also lose after suspend.

huming2207, 2021-11-23:
@embeddedt @xiaoxiang781216 Thanks for your reply.

For our use cases with e-Paper, since the display content is kept even the system is powered off, we actually don't care about the LVGL draw buffer etc. What we do really need is probably the object tree and render state so that we can prevent our stuff from rendering something unnecessary again. For example, on the next wake-up from deep sleep, we only need to update a row of text, instead of transferring the whole  framebuffer and let the e-paper update the whole screen, we then restore the state and only need to circle out a small fraction of area and update it. 

I guess @xiaoxiang781216 's way may works. I'll probably do some tests on it later. Alternatively I'm also thinking to investigate a way to traverse the object tree (from `lv_scr_act()`) and take a snapshot. But this may be even tougher.

closed 2021-12-16 by lvgl-bot