lvgl/lvgl #2831: lv_indev_drv_update not working (leaking memory)

geert-KLA-BE, 2021-11-24:
### Perform all steps below and tick them with [x]
- [X] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [X] Update lvgl to the latest version
- [X] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

When using `lv_indev_drv_update` the touch input does not work anymore.

`lv_indev_drv_update `does not correctly update the driver. It only does: 
`indev->driver = new_drv;`

It is this way leaking memory because `indev->driver->read_timer` is not copied or released & recreated.
Also no display is set when not present.

### To Reproduce

Just call `lv_indev_drv_update ` after initial setup causes the input to fail.

### Expected behavior

When using `lv_indev_drv_update `the touch input keeps working without memory leaking.

geert-KLA-BE, 2021-11-24:
This is a partial fix. Since it still doesn't work correctly on my device.
`void lv_indev_drv_update(lv_indev_t * indev, lv_indev_drv_t * new_drv)
{
    LV_ASSERT_NULL( indev );
    lv_timer_del( indev->driver->read_timer );

    if ( new_drv->disp == NULL )
    {
        new_drv->disp = lv_disp_get_default();
    }

    if(new_drv->disp == NULL) {
        LV_LOG_WARN("lv_indev_drv_register: no display registered hence can't attach the indev to "
                     "a display");
        indev->proc.disabled = true;
        return;
    }

    indev->driver = new_drv;
    indev->proc.reset_query  = 1;
    indev->driver->read_timer = lv_timer_create(lv_indev_read_timer_cb, LV_INDEV_DEF_READ_PERIOD, indev);
}
`

kisvegabor, 2021-11-26:
Your suggested code looks good. 

Why is it not working? What do you experience?

geert-KLA-BE, 2021-11-29:
Ok the remaining issue was on my part. I was updating the input driver being used. So that gave issues off course.
In my final version I added some additional asserts.

I need to use this because it seems that a `lv_indev_drv_del` is missing in the API. So that would be a good addition.

```C
void lv_indev_drv_update(lv_indev_t * indev, lv_indev_drv_t * new_drv)
{
    LV_ASSERT_NULL( indev );
    LV_ASSERT_NULL( indev->driver );
    LV_ASSERT_NULL( indev->driver->read_timer );
    lv_timer_del( indev->driver->read_timer );

    LV_ASSERT_NULL( new_drv );
    if ( new_drv->disp == NULL )
    {
        new_drv->disp = lv_disp_get_default();
    }
    if(new_drv->disp == NULL) {
        LV_LOG_WARN("lv_indev_drv_register: no display registered hence can't attach the indev to "
                     "a display");
        indev->proc.disabled = true;
        return;
    }

    indev->driver = new_drv;
    indev->driver->read_timer = lv_timer_create(lv_indev_read_timer_cb, LV_INDEV_DEF_READ_PERIOD, indev);
    indev->proc.reset_query   = 1;
}
```

geert-KLA-BE, 2021-11-29:
Created pull request
#2850

geert-KLA-BE, 2021-12-01:
Pull request merged. So this one can be closed.

closed 2021-12-01 by geert-KLA-BE