lvgl/lvgl #2842: Simple helloworld not working on M5STACK = ESP32 + ILI9341

spara7C5, 2021-11-27:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [o] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

### To Reproduce

### Expected behavior

### Screenshots or video
Something of the "Hello world" writing is present: it is not centered and is "sliced"

kisvegabor, 2021-11-29:
It seems `#define LV_COLOR_DEPTH 32` should be `16`.

---

We use GitHub Issues for development-related discussions.

Please ask your question in the [Forum](https://forum.lvgl.io) to make it visible for a larger community. You can simply log in with your GitHub account.

closed 2021-11-29 by kisvegabor

spara7C5, 2021-11-29:
[image]

This is the result I get changing the value from 32 to 16

kisvegabor, 2021-11-30:
Sorry, but I can help with it further as I don't really know neither MD5STACK nor the TFT_eSPI. I hope someone have more experience with these in the Forum.