lvgl/lvgl #2847: Crash during benchmarking, and memory access is out of bounds.

Virus-V, 2021-11-29:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

This problem does not appear every time. 

When I was porting lvgl and running benchmark on FreeBSD 13 virtual machine, I encountered several segmentation faults and generated coredump. 

Use the debugger to trace its call stack and local variables, and find that the access to the draw buffer in the `fill_blended` function is out of bounds.

Before the segmentation fault occurs, there will be a phenomenon of blurring.

[image]

### To Reproduce

**Commit ID I use: 5433e0bc2abb6d0db931a1721bf723f4035a73a2**

Here is my lv_conf.h header file:
```c
/**
 * @file lv_conf.h
 * Configuration file for v8.0.0
 */

/*
 * COPY THIS FILE AS `lv_conf.h` NEXT TO the `lvgl` FOLDER
 */

#if 1 /*Set it to "1" to enable content*/

#ifndef LV_CONF_H
#define LV_CONF_H
/*clang-format off*/

#include <stdint.h>

/*====================
   COLOR SETTINGS
 *====================*/

/*Color depth: 1 (1 byte per pixel), 8 (RGB332), 16 (RGB565), 32 (ARGB8888)*/
#define LV_COLOR_DEPTH     32

/*Swap the 2 bytes of RGB565 color. Useful if the display has a 8 bit interface (e.g. SPI)*/
#define LV_COLOR_16_SWAP   0

/*Enable more complex drawing routines to manage screens transparency.
 *Can be used if the UI is above an other layer, e.g. an OSD menu or video player.
 *Requires `LV_COLOR_DEPTH = 32` colors and the screen's `bg_opa` should be set to non LV_OPA_COVER value*/
#define LV_COLOR_SCREEN_TRANSP    0

/*Images pixels with this color will not be drawn if they are  chroma keyed)*/
#define LV_COLOR_CHROMA_KEY    lv_color_hex(0x00ff00)         /*pure green*/

/*=========================
   MEMORY SETTINGS
 *=========================*/

/*1: use custom malloc/free, 0: use the built-in `lv_mem_alloc()` and `lv_mem_free()`*/
#define LV_MEM_CUSTOM      0
#if LV_MEM_CUSTOM == 0
/*Size of the memory available for `lv_mem_alloc()` in bytes (>= 2kB)*/
#  define LV_MEM_SIZE    (32U * 1024U)          /*[bytes]*/

/*Set an address for the memory pool instead of allocating it as a normal array. Can be in external SRAM too.*/
#  define LV_MEM_ADR          0     /*0: unused*/
[... 568 lines cut ...]
$2 = {
  buf1 = 0x801300100,
  buf2 = 0x0,
  buf_act = 0x801300100,
  size = 786432,
  area = {
    x1 = 0,
    y1 = 16,
    x2 = 1023,
    y2 = 767
  },
  flushing = 0,
  flushing_last = 0,
  last_area = 0,
  last_part = 1
}
```

### Expected behavior

The screen display is garbled, resulting in a segmentation fault.

### Screenshots or video

https://user-images.githubusercontent.com/32447708/143835392-00f78eca-db5f-4d47-868f-1d9203bcb815.mov

kisvegabor, 2021-11-30:
Hi,

Does it help if you increase `LV_MEM_SIZE`? E.g. to 48 * 1024.

If not, please enable `LV_USE_LOG` and all `LV_USE_ASSERT_...`. Hopefully, LVGL sees what's the problem. :slightly_smiling_face:

Virus-V, 2021-12-01:
> Hi,
> 
> Does it help if you increase `LV_MEM_SIZE`? E.g. to 48 * 1024.
> 
> If not, please enable `LV_USE_LOG` and all `LV_USE_ASSERT_...`. Hopefully, LVGL sees what's the problem. 🙂

I tried to increase the LV_MEM_SIZE to 1MByte, and open all ASSERT checks, the segmentation error still appears.
```diff
@@ -42,7 +42,7 @@
 #define LV_MEM_CUSTOM      0
 #if LV_MEM_CUSTOM == 0
 /*Size of the memory available for `lv_mem_alloc()` in bytes (>= 2kB)*/
-#  define LV_MEM_SIZE    (32U * 1024U)          /*[bytes]*/
+#  define LV_MEM_SIZE    (1024U * 1024U)          /*[bytes]*/

 /*Set an address for the memory pool instead of allocating it as a normal array. Can be in external SRAM too.*/
 #  define LV_MEM_ADR          0     /*0: unused*/
@@ -137,7 +137,7 @@ e.g. "stm32f769xx.h" or "stm32f429xx.h"*/
  *-----------*/

 /*Enable the log module*/
-#define LV_USE_LOG      0
+#define LV_USE_LOG      1
 #if LV_USE_LOG

 /*How important log should be added:
@@ -173,20 +173,20 @@ e.g. "stm32f769xx.h" or "stm32f429xx.h"*/
  *If LV_USE_LOG is enabled an error message will be printed on failure*/
 #define LV_USE_ASSERT_NULL          1   /*Check if the parameter is NULL. (Very fast, recommended)*/
 #define LV_USE_ASSERT_MALLOC        1   /*Checks is the memory is successfully allocated or no. (Very fast, recommended)*/
-#define LV_USE_ASSERT_STYLE         0   /*Check if the styles are properly initialized. (Very fast, recommended)*/
-#define LV_USE_ASSERT_MEM_INTEGRITY 0   /*Check the integrity of `lv_mem` after critical operations. (Slow)*/
-#define LV_USE_ASSERT_OBJ           0   /*Check the object's type and existence (e.g. not deleted). (Slow)*/
+#define LV_USE_ASSERT_STYLE         1   /*Check if the styles are properly initialized. (Very fast, recommended)*/
+#define LV_USE_ASSERT_MEM_INTEGRITY 1   /*Check the integrity of `lv_mem` after critical operations. (Slow)*/
+#define LV_USE_ASSERT_OBJ           1   /*Check the object's type and existence (e.g. not deleted). (Slow)*/

 /*Add a custom handler when assert happens e.g. to restart the MCU*/
-#define LV_ASSERT_HANDLER_INCLUDE   <stdint.h>
-#define LV_ASSERT_HANDLER   while(1);   /*Halt by default*/
+#define LV_ASSERT_HANDLER_INCLUDE   <stdlib.h>
+#define LV_ASSERT_HANDLER   abort();   /*Halt by default*/

 /*-------------
  * Others
  *-----------*/

 /*1: Show CPU usage and FPS count in the right bottom corner*/
-#define LV_USE_PERF_MONITOR     0
+#define LV_USE_PERF_MONITOR     1

 /*1: Show the used memory and the memory fragmentation  in the left bottom corner
  * Requires LV_MEM_CUSTOM = 0*/
```

Cut the benchmark item to keep only Line, Arc think, Arc thick and all Substr. XXX, segmentation errors will appear very quickly.
```diff
--- a/lv_demo_benchmark.c
+++ b/lv_demo_benchmark.c
@@ -532,6 +532,7 @@ static void sub_text_cb(void)
  *  STATIC VARIABLES
  **********************/
 static scene_dsc_t scenes[] = {
+#if 0
         {.name = "Rectangle",                    .weight = 30, .create_cb = rectangle_cb},
         {.name = "Rectangle rounded",            .weight = 20, .create_cb = rectangle_rounded_cb},
         {.name = "Circle",                       .weight = 10, .create_cb = rectangle_circle_cb},
@@ -576,6 +577,7 @@ static scene_dsc_t scenes[] = {
         {.name = "Text small compressed",       .weight = 3, .create_cb = txt_small_compr_cb},
         {.name = "Text medium compressed",      .weight = 5, .create_cb = txt_medium_compr_cb},
         {.name = "Text large compressed",       .weight = 10, .create_cb = txt_large_compr_cb},
+#endif

         {.name = "Line",                        .weight = 10, .create_cb = line_cb},

         {.name = "Arc think",                   .weight = 10, .create_cb = arc_think_cb},
         {.name = "Arc thick",                   .weight = 10, .create_cb = arc_thick_cb},

         {.name = "Substr. rectangle",          .weight = 10, .create_cb = sub_rectangle_cb},
         {.name = "Substr. border",             .weight = 10, .create_cb = sub_border_cb},
         {.name = "Substr. shadow",             .weight = 10, .create_cb = sub_shadow_cb},
         {.name = "Substr. image",             .weight = 10, .create_cb = sub_img_cb},
         {.name = "Substr. line",               .weight = 10, .create_cb = sub_line_cb},
         {.name = "Substr. arc",                .weight = 10, .create_cb = sub_arc_cb},
         {.name = "Substr. text",               .weight = 10, .create_cb = sub_text_cb},

         {.name = "", .create_cb = NULL}
```

Virus-V, 2021-12-01:
https://github.com/lvgl/lvgl/pull/2860

closed 2021-12-01 by Virus-V