lvgl/lvgl #2855: sanity check in lv_indev.c not working with XPT2046

Linkenelis, 2021-11-30:
### Perform all steps below and tick them with [x]
- [Y] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [Y ] Update lvgl to the latest version
- [Difficult as it is XPT2046 touch] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

### To Reproduce

### Expected behavior

### Screenshots or video

Linkenelis, 2021-11-30:
After commenting out this sanity check it worked as expected. I guess placing this check below line 780 and using proc->types.pointer.act_point.x and proc->types.pointer.act_point.y, will solve the issues.

kisvegabor, 2021-11-30:
You should convert the raw coordinates into "screen coordinates" in your `indev_read` function. So LVGL should get [0..hor_res-1] and [0..ver_res-1] coordinates.

Linkenelis, 2021-11-30:
static void indev_proc_press(_lv_indev_proc_t * proc) ; **x and y are OK and these are handled**
static void indev_pointer_proc(lv_indev_t * i, lv_indev_data_t * data)  ; **x and y are the higer values and gives only Warnings**

So are you sure?

Linkenelis, 2021-11-30:
Am I correct in thinking it needs the original values for the correct rotation?

Linkenelis, 2021-11-30:
When you look into this, maybe check this out too (LV_DISP_ROT_180 || disp->driver->rotated == LV_DISP_ROT__**270**_):
    if(disp->driver->rotated == LV_DISP_ROT_180 || disp->driver->rotated == LV_DISP_ROT_270) {
        data->point.x = disp->driver->hor_res - data->point.x - 1;
        data->point.y = disp->driver->ver_res - data->point.y - 1;
    }
    if(disp->driver->rotated == LV_DISP_ROT_90 || disp->driver->rotated == LV_DISP_ROT_270) {
        lv_coord_t tmp = data->point.y;
        data->point.y = data->point.x;
        data->point.x = disp->driver->ver_res - tmp - 1;
    }

SgtPepperFTW, 2021-12-04:
You know, that XPT2046 is analog converter, he doesn't know, what is the resolution of the screen he is glued to. And one xpt2046 may return different values than another one. Border values of one might different from another. So ideally you should calibrate your XPT2046 before first use, finding out it's border values, then converting them in something useful, for example - [0..hor_res-1] and [0..ver_res-1]

Linkenelis, 2021-12-04:
Thanks SgtPepperFTW.
driver convert from analog to 0-4095 digital and mostly borders are around 300 (but not very precise and might vary).
Then there is rotation...
My point was that it worked just fine, as long as I comment out the sanity check. These warnings-output were clogging everything, even when no "pressing" was done.

Linkenelis, 2021-12-04:
But maybe it is how Tasmota implementation is done...

kisvegabor, 2021-12-06:
> My point was that it worked just fine, as long as I comment out the sanity check. 

These should only add log messages and do not change the behavior.

Anyway, what can I tell for sure, LVGL needs the coordinates in [0..hor_res-1] and [0..ver_res-1] range. If it worked without it then maybe Tasmote did something in the background or it looked correct only by accident.

Linkenelis, 2021-12-07:
OK thanks, I will take it up with Tasmota team.

kisvegabor, 2021-12-22:
As the issue in likely not in LVGL I close this issue.

closed 2021-12-22 by kisvegabor