lvgl/lvgl #2858: lv_btnmatrix_get_selected_btn() == 65535 on LV_EVENT_CLICKED

knifter, 2021-11-30:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
Using a btnmatrix I'm trying to discern between CLICKED and LONG_PRESSED (as these are the only mutual exclusive ones)
But while ```uint32_t id = lv_btnmatrix_get_selected_btn(obj);``` gives me the id in most events, the id == 65535 when the e->code == LV_EVENT_CLICKED.

I can guess why: LV_EVENT_RELEASED is just before that so probably 'the-current-button' is erased by then?
A workaround can be to store the btn_id in a static var in the callback on LV_EVENT_PRESSED, but it is more btnmatrix's job I think.

### To Reproduce
 * Create btnmatrix with event handler for LV_EVENT_ALL
 * In the handler check result of ```lv_btnmatrix_get_selected_btn``` on each event.

```c
void make()
{
	lv_obj_t *butmat1 = lv_btnmatrix_create(_screen);
	lv_obj_add_event_cb(butmat1, butmat1_pressed_cb, LV_EVENT_ALL, NULL);
};
void butmat1_pressed_cb(lv_event_t* e)
{
	debug_event_input(e); // my debug function, see below
	if(e->code != LV_EVENT_SHORT_CLICKED 
	&& e->code != LV_EVENT_LONG_PRESSED 
	&& e->code != LV_EVENT_LONG_PRESSED_REPEAT)
		return;

	lv_obj_t * obj = lv_event_get_target(e);
	uint32_t id = lv_btnmatrix_get_selected_btn(obj);

	DBG("e->code = %d, id = %d", e->code, id);
};
```
-->

### Expected behavior
for ```id``` to be the correct button index

### Screenshots or video
[OK] Long press:
```
lib\lib-soogh\src\soogh-debug.cpp.debug_event_input: : VALUECHANGED
lib\lib-soogh\src\soogh-debug.cpp.debug_event_input: : PRESSED     
lib\lib-soogh\src\soogh-debug.cpp.debug_event_input: : LONG_PRESSED
src\screens\main.cpp.butmat1_pressed_cb: e->code = 5, id = **1**       
lib\lib-soogh\src\soogh-debug.cpp.debug_event_input: : VALUECHANGED       
lib\lib-soogh\src\soogh-debug.cpp.debug_event_input: : LONG_PRESSED_REPEAT
src\screens\main.cpp.butmat1_pressed_cb: e->code = 6, id = **1**
lib\lib-soogh\src\soogh-debug.cpp.debug_event_input: : VALUECHANGED       
lib\lib-soogh\src\soogh-debug.cpp.debug_event_input: : LONG_PRESSED_REPEAT
src\screens\main.cpp.butmat1_pressed_cb: e->code = 6, id = **1**
lib\lib-soogh\src\soogh-debug.cpp.debug_event_input: : RELEASED
lib\lib-soogh\src\soogh-debug.cpp.debug_event_input: : CLICKED 
```
[No ID] Short press:
```
lib\lib-soogh\src\soogh-debug.cpp.debug_event_input: : VALUECHANGED
lib\lib-soogh\src\soogh-debug.cpp.debug_event_input: : PRESSED     
lib\lib-soogh\src\soogh-debug.cpp.debug_event_input: : FOCUSED     
lib\lib-soogh\src\soogh-debug.cpp.debug_event_input: : RELEASED     
lib\lib-soogh\src\soogh-debug.cpp.debug_event_input: : SHORT_CLICKED
src\screens\main.cpp.butmat1_pressed_cb: e->code = 4, id = **65535**    
lib\lib-soogh\src\soogh-debug.cpp.debug_event_input: : CLICKED
```

knifter, 2021-11-30:
Can confirm that the following work-around works:
```c
if(e->code != LV_EVENT_SHORT_CLICKED 
	&& e->code != LV_EVENT_PRESSED 
	&& e->code != LV_EVENT_LONG_PRESSED 
	&& e->code != LV_EVENT_LONG_PRESSED_REPEAT)
		return;

	lv_obj_t * obj = lv_event_get_target(e);

	// workaround for #2858
	static uint32_t id;
	if(e->code == LV_EVENT_PRESSED)
		id = lv_btnmatrix_get_selected_btn(obj);
```

embeddedt, 2021-12-01:
I agree that the current behavior is unintuitive; I ran into it myself while making an example for someone. You're correct, it does clear the selected button when `LV_EVENT_RELEASED` is sent. To solve it in LVGL, we'd most likely have to add a second global for the "last pressed" button.

@kisvegabor What do you think?

knifter, 2021-12-06:
>it does clear the selected button when LV_EVENT_RELEASED is sent

Allright, thats what I suspected. But why? The SHORT_CLICKED does still belong to that 'id', I would argue.

Other ideas: 
 * emit SHORT_CLICKED before RELEASE? 
 * Or maybe CLICKED can clear the variable. 
 * Or maybe not clear it at all until a new one is pressed (what I did in my workaround).

kisvegabor, 2021-12-07:
Hi,

I've just pushed a fix as you suggested in the third point. (I.e. keep the selected button)

closed 2021-12-07 by kisvegabor

Johennes, 2022-02-20:
It looks like the fix broke the key coloring in [unl0kr](https://gitlab.com/cherrypicker/unl0kr/-/blob/master/theme.c#L309). I'm hooking into `LV_EVENT_DRAW_PART_BEGIN` to be able to change the button color on key down and revert it on key up. Since the selected button is not reset anymore on `LV_EVENT_RELEASED`, the button continues to be rendered as pressed now. Any idea how I could distinguish between pressed and unpressed buttons on the drawing level?

kisvegabor, 2022-02-21:
> Any idea how I could distinguish between pressed and unpressed buttons on the drawing level?

You can use `if(lv_obj_has_state(btnm, LV_STATE_PRESSED)) { ... }`

Johennes, 2022-02-21:
It took me a while to get this right. It works when I use both:

```c
bool pressed = lv_btnmatrix_get_selected_btn(obj) == dsc->id && lv_obj_has_state(btnm, LV_STATE_PRESSED);
```

Just `lv_obj_has_state(btnm, LV_STATE_PRESSED)` was not enough because that has the same value for both the currently pressed and the previously released button so I couldn't distinguish between the two.

Thanks a lot for the tip! :slightly_smiling_face:

commit d47cd1d7 (2021-12-07): fix(btnmatrix): keep the selected button even on release

fixes: #2858