lvgl/lvgl #2863: Subpix text rendering does not work properly when using compact draw buffers

andresovela, 2021-12-01:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
Labels render with glitches when subpixel rendering is enabled and your draw buffers do not map 1-to-1 to the display.

#### Context
[Initial report in the LVGL forum](https://forum.lvgl.io/t/how-to-properly-render-fonts/7342)

I am using LVGL to drive a display with the [SH1122 driver](https://www.displayfuture.com/Display/datasheet/controller/SH1122.pdf). The display has 256x64 4-bit grayscale pixels. In order to use memory efficiently, I use the [`set_px_cb()`](https://docs.lvgl.io/master/porting/display.html#_CPPv4N14_lv_disp_drv_t9set_px_cbE) function in order to set pixels in the buffer.

I am currently using two buffers and I have the full refresh option enabled.

I have tracked the issue down and the problem is in these lines:

https://github.com/lvgl/lvgl/blob/ee65410c3725070ed1779c95fb8742107cdd9267/src/draw/sw/lv_draw_sw_letter.c#L442-L447

The problem is that LVGL is assuming that it can just read colors as they are, directly from the buffers. In m particular setup, this is not the case. The coordinates are not even right: coordinate (2, 0) is at `buf[1]` in my case, since I'm storing two 4-bit pixels per byte.

I presume the same issue exists anywhere else where LVGL needs to read data from the draw buffer. I have fixed this issue by adding a `get_px_cb()` function to the display driver and using that function in the rendering code instead of having direct reads from the draw buffer.

I don't know if that's the best way to fix the issue, but it worked for me.

Basically I made this change:

```diff
+ lv_color_t bg_color = disp->driver->get_px_cb(draw_buf->buf_act, col, row);
+ uint8_t bg_rgb[3] = {bg_color.ch.red, bg_color.ch.green, bg_color.ch.blue};
- uint8_t bg_rgb[3] = {disp_buf_buf_tmp->ch.red, disp_buf_buf_tmp->ch.green, disp_buf_buf_tmp->ch.blue};
```

The variable `disp_buf_buf_tmp` was not used then, and I removed it.

I hope this helps, and I hope it can be fixed upstream soon.

### To Reproduce

### Expected behavior
Labels with subpixel rendering should look smooth.

### Screenshots or video
See [initial report in the LVGL forum](https://forum.lvgl.io/t/how-to-properly-render-fonts/7342)

kisvegabor, 2021-12-06:
Hi,

> I am using LVGL to drive a display with the SH1122 driver. The display has 256x64 4-bit grayscale pixels. In order to use memory efficiently, I use the set_px_cb() function in order to set pixels in the buffer.

Subpixel rendering requires an RGB buffer as it provides better quality by utilizing that the RGB channels have a known order in each pixel. This way the R, G, B channels are considered as "sub-pixels". See more here: https://en.wikipedia.org/wiki/Subpixel_rendering

So it short, it doesn't work with grayscale displays.

andresovela, 2021-12-06:
It does work, I have it working just fine. Maybe it doesn't look as smooth as it would look on a color display, but it does look smooth.

kisvegabor, 2021-12-06:
Can you send 2 screenshots with normal bpp=4 and sub pixel version?

andresovela, 2021-12-06:
With:
[image]

Without:
[image]

kisvegabor, 2021-12-06:
It's strange, can you attach the 2 fonts and your set_px_cb?

andresovela, 2021-12-06:
file:///home/andres.oliva/dev/projects/ultima25A/src/display/fonts/inter_medium_40.zip

```
static void on_lvgl_pixel_draw(lv_disp_drv_t * disp_drv,
                               uint8_t * buf,
                               lv_coord_t buf_w,
                               lv_coord_t x,
                               lv_coord_t y,
                               lv_color_t color,
                               lv_opa_t opa) {
    // This is a lookup table to convert a color to a 4-bit brightness number
    // adequate for the monochrome display (faster than converting each pixel at runtime)
    // This table was created as follows:
    // - for each RGB332 pixel from 0 to 255:
    //    double red = color.red * 255 / 7;
    //    double green = color.green * 255 / 7;
    //    double blue = color.blue * 255 / 3;
    //    double luma = red * 0.299 + green * 0.587 + blue * 0.114;
    //    uint8_t eight_bit_luma = luma;
    //    uint8_t four_bit_luma = luma >> 4;
    static const uint8_t rgb332_to_4_bit_luma[256] = {
         0,  0,  1,  1,  1,  1,  2,  3,  2,  3,  3,  4,  3,  4,  5,  5,
         5,  5,  6,  7,  6,  7,  7,  8,  7,  8,  9,  9,  9,  9, 10, 11,
         0,  1,  1,  2,  1,  2,  3,  3,  3,  3,  4,  5,  4,  5,  5,  6,
         5,  6,  7,  7,  7,  7,  8,  9,  8,  9,  9, 10, 10, 10, 11, 11,
         1,  1,  2,  3,  2,  3,  3,  4,  3,  4,  5,  5,  5,  5,  6,  7,
         6,  7,  7,  8,  8,  8,  9,  9,  9,  9, 10, 11, 10, 11, 11, 12,
         2,  2,  3,  3,  3,  3,  4,  5,  4,  5,  5,  6,  6,  6,  7,  7,
         7,  7,  8,  9,  8,  9,  9, 10, 10, 10, 11, 11, 11, 11, 12, 13,
         2,  3,  3,  4,  4,  4,  5,  5,  5,  5,  6,  7,  6,  7,  7,  8,
         8,  8,  9,  9,  9,  9, 10, 11, 10, 11, 11, 12, 12, 12, 13, 13,
         3,  4,  4,  5,  4,  5,  5,  6,  6,  6,  7,  7,  7,  8,  8,  9,
         8,  9,  9, 10, 10, 10, 11, 11, 11, 12, 12, 13, 12, 13, 13, 14,
         4,  4,  5,  5,  5,  6,  6,  7,  6,  7,  7,  8,  8,  8,  9,  9,
         9,  9, 10, 11, 10, 11, 11, 12, 12, 12, 13, 13, 13, 14, 14, 15,
         4,  5,  5,  6,  6,  6,  7,  7,  7,  8,  8,  9,  8,  9,  9, 10,
        10, 10, 11, 11, 11, 12, 12, 13, 12, 13, 13, 14, 14, 14, 15, 15,
    };

    // Unused variables
    (void)disp_drv;
    (void)buf_w;
    (void)opa;

    // Each column has two pixels, therefore we divide by 2
    // Pixels 0 and 1 are in column 0, pixels 2 and 3 are in column 1, etc.
    uint32_t index_in_buffer = (x / 2) + (y * BOARD_DISPLAY_HS1122_COLUMN_COUNT);

    // Transform the color to a 4-bit brightness for the pixel
    uint8_t pixel = rgb332_to_4_bit_luma[color.full];

    // Even pixels go in the high nibble
    // Odd pixels go in the low nibble
    if (x % 2 == 0) {
        buf[index_in_buffer] &= 0x0F;
        buf[index_in_buffer] |= (pixel << 4);
    } else {
        buf[index_in_buffer] &= 0xF0;
        buf[index_in_buffer] |= pixel & 0x0F;
    }
}
```

andresovela, 2021-12-06:
[inter_medium_40.zip](https://github.com/lvgl/lvgl/files/7659832/inter_medium_40.zip)

andresovela, 2021-12-06:
Are you looking for something in particular? As I mentioned in the issue description, I have already isolated the issue and fixed it.

kisvegabor, 2021-12-06:
> Are you looking for something in particular? As I mentioned in the issue description, I have already isolated the issue and fixed it.

Yes, because on the non-subpixel image I don't see the anti-aliasing which should come from bpp=4.

I think I see the problem: You don't use the `opa` parameter in the `set_px_cb`. You should mix the color of the buffer with the color to set with this ratio:
```
result = (color_in_buffer * (255 - opa) + color * opa) / 255
```

andresovela, 2021-12-06:
Thanks for the tip! It does look better but still worse than with subpixel rendering:

[image]

kisvegabor, 2021-12-06:
How does the `set_px_cb` look like now?

andresovela, 2021-12-06:
```
static void on_lvgl_pixel_draw(lv_disp_drv_t * disp_drv,
                               uint8_t * buf,
                               lv_coord_t buf_w,
                               lv_coord_t x,
                               lv_coord_t y,
                               lv_color_t color,
                               lv_opa_t opa) {
    // This is a lookup table to convert a color to a 4-bit brightness number
    // adequate for the monochrome display (faster than converting each pixel at runtime)
    // This table was created as follows:
    // - for each RGB332 pixel from 0 to 255:
    //    double red = color.red * 255 / 7;
    //    double green = color.green * 255 / 7;
    //    double blue = color.blue * 255 / 3;
    //    double luma = red * 0.299 + green * 0.587 + blue * 0.114;
    //    uint8_t eight_bit_luma = luma;
    //    uint8_t four_bit_luma = luma >> 4;
    static const uint8_t rgb332_to_4_bit_luma[256] = {
         0,  0,  1,  1,  1,  1,  2,  3,  2,  3,  3,  4,  3,  4,  5,  5,
         5,  5,  6,  7,  6,  7,  7,  8,  7,  8,  9,  9,  9,  9, 10, 11,
         0,  1,  1,  2,  1,  2,  3,  3,  3,  3,  4,  5,  4,  5,  5,  6,
         5,  6,  7,  7,  7,  7,  8,  9,  8,  9,  9, 10, 10, 10, 11, 11,
         1,  1,  2,  3,  2,  3,  3,  4,  3,  4,  5,  5,  5,  5,  6,  7,
         6,  7,  7,  8,  8,  8,  9,  9,  9,  9, 10, 11, 10, 11, 11, 12,
         2,  2,  3,  3,  3,  3,  4,  5,  4,  5,  5,  6,  6,  6,  7,  7,
         7,  7,  8,  9,  8,  9,  9, 10, 10, 10, 11, 11, 11, 11, 12, 13,
         2,  3,  3,  4,  4,  4,  5,  5,  5,  5,  6,  7,  6,  7,  7,  8,
         8,  8,  9,  9,  9,  9, 10, 11, 10, 11, 11, 12, 12, 12, 13, 13,
         3,  4,  4,  5,  4,  5,  5,  6,  6,  6,  7,  7,  7,  8,  8,  9,
         8,  9,  9, 10, 10, 10, 11, 11, 11, 12, 12, 13, 12, 13, 13, 14,
         4,  4,  5,  5,  5,  6,  6,  7,  6,  7,  7,  8,  8,  8,  9,  9,
         9,  9, 10, 11, 10, 11, 11, 12, 12, 12, 13, 13, 13, 14, 14, 15,
         4,  5,  5,  6,  6,  6,  7,  7,  7,  8,  8,  9,  8,  9,  9, 10,
        10, 10, 11, 11, 11, 12, 12, 13, 12, 13, 13, 14, 14, 14, 15, 15,
    };

    // Unused variables
    (void)disp_drv;
    (void)buf_w;
    (void)opa;

    // Each column has two pixels, therefore we divide by 2
    // Pixels 0 and 1 are in column 0, pixels 2 and 3 are in column 1, etc.
    uint32_t index_in_buffer = (x / 2) + (y * BOARD_DISPLAY_HS1122_COLUMN_COUNT);

    // Transform the color to a 4-bit brightness for the pixel
    uint8_t incoming_pixel = rgb332_to_4_bit_luma[color.full];

    // Even pixels go in the high nibble
    // Odd pixels go in the low nibble
    if (x % 2 == 0) {
        uint8_t pixel_in_buffer = buf[index_in_buffer] >> 4;
        uint8_t blended_pixel = (pixel_in_buffer * (LV_OPA_COVER - opa) + incoming_pixel * opa) / LV_OPA_COVER;
        buf[index_in_buffer] &= 0x0F;
        buf[index_in_buffer] |= (blended_pixel << 4);
    } else {
        uint8_t pixel_in_buffer = buf[index_in_buffer] & 0x0F;
        uint8_t blended_pixel = (pixel_in_buffer * (LV_OPA_COVER - opa) + incoming_pixel * opa) / LV_OPA_COVER;
        buf[index_in_buffer] &= 0xF0;
        buf[index_in_buffer] |= blended_pixel & 0x0F;
    }
}
```

kisvegabor, 2021-12-06:
I also some improvements but it should be much smoother. Please try this:
- display only an `S` letter (with the same font as the "System" label now)
- print the `x`, `y`, `opa`, `color`, `blended_pixel` variables from the set_px_cb

andresovela, 2021-12-06:
Ah, I had antialiasing disabled. Now it basically looks the same as with subpix rendering. 

[image]

kisvegabor, 2021-12-06:
Great! It looks as expected! :slightly_smiling_face: 

A sub-pixel rendering is really for RGB I close this issue.

closed 2021-12-06 by kisvegabor

andresovela, 2021-12-06:
I think it would be good to have a note in the font generator to explain that sub pixel rendering is meant for RGB displays.

kisvegabor, 2021-12-07:
I've added links to the docs here: https://lvgl.io/tools/fontconverter
[image]