lvgl/lvgl #2868: Updating arc causes memory leak in specific situation

DCSBL, 2021-12-02:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

Something is leaking (a lot) of memory in a very specific situation. I want do draw 3 arcs and animate or update them. Found this out while using LVGL on the ESP32-S2 but it is reproducible on the simulator. I can't figure out if it something related to the arc or something generic (or just my code of course).

Leaking only happens when I have 3 arcs with a different size **and** they all 3 are updating its value. It makes no difference if the update was a manual call or an animation.
Leaking won't happen when I, for example;
- use 3 arcs with the same size;
- animate only two arcs
- keep the arc in default shape (remove `lv_arc_set_bg_angles(arc, 0, 360)`)

**Maybe important**: Nothing leaks with LVGL 8.0.x

### To Reproduce

- Using lv_sim_vscode_sdl (https://github.com/lvgl/lv_sim_vscode_sdl/releases/tag/8.0.0)
- Updated to LVGL 8.1.0 (https://github.com/lvgl/lvgl/tree/v8.1.0)
- Replaced `main.c` with code below
- No specific changes in `lv_conf.h`
<details>
<summary>main.c (click to expand)</summary>

```c
/**
 * @file main
 *
 */

/*********************
 *      INCLUDES
 *********************/
#define _DEFAULT_SOURCE /* needed for usleep() */
#include <stdlib.h>
#include <unistd.h>
#define SDL_MAIN_HANDLED /*To fix SDL's "undefined reference to WinMain" issue*/
#include <SDL2/SDL.h>
#include "lvgl/lvgl.h"
#include "lv_drivers/display/monitor.h"

/*********************
 *      DEFINES
 *********************/

/**********************
 *      TYPEDEFS
 **********************/

/**********************
 *  STATIC PROTOTYPES
 **********************/
static void hal_init(void);
static int tick_thread(void *data);

/**********************
 *  STATIC VARIABLES
 **********************/

/**********************
 *      MACROS
 **********************/

/*********************
 *      DEFINES
 *********************/

/**********************
 *      TYPEDEFS
 **********************/
typedef enum
{
    WHEEL_SIZE_BIG,
    WHEEL_SIZE_MEDIUM,
    WHEEL_SIZE_SMALL,
} wheel_size_t;

typedef struct
{
    // lv_obj_t *root; <- Not used for this example
    lv_obj_t *gauge;
} wheel_t;

/**********************
 *   GLOBAL FUNCTIONS
 **********************/
void draw_something();
void draw_wheel(lv_obj_t * parent_container, wheel_size_t size, wheel_t *output);
void animate_wheel(lv_obj_t *obj);

/**********************
 *      VARIABLES
 **********************/

wheel_t obj_wheel_l;
wheel_t obj_wheel_ur;
wheel_t obj_wheel_br;

/**********************
 *  STATIC PROTOTYPES
 **********************/

/**********************
 *   GLOBAL FUNCTIONS
 **********************/

int main(int argc, char **argv)
{
  (void)argc; /*Unused*/
  (void)argv; /*Unused*/

  /*Initialize LVGL*/
  lv_init();

  /*Initialize the HAL (display, input devices, tick) for LVGL*/
  hal_init();
  
  draw_something();

  while(1) {
    /* Periodically call the lv_task handler.
     * It could be done in a timer interrupt or an OS task too.*/
    lv_timer_handler();
    usleep(5 * 1000);
  }

  return 0;
}

void draw_something()
{
    printf("%s:%d (%s)\n", __FILE__, __LINE__, __FUNCTION__);
    
    lv_obj_t * main_container = lv_scr_act();
    draw_wheel(main_container, WHEEL_SIZE_BIG, &obj_wheel_l);
    lv_obj_align_to(obj_wheel_l.gauge, main_container, LV_ALIGN_TOP_LEFT, 0, 0);
    
    draw_wheel(main_container, WHEEL_SIZE_MEDIUM, &obj_wheel_ur);
    lv_obj_align_to(obj_wheel_ur.gauge, main_container, LV_ALIGN_TOP_RIGHT, 0, 0);
    
    draw_wheel(main_container, WHEEL_SIZE_SMALL, &obj_wheel_br);
    lv_obj_align_to(obj_wheel_br.gauge, main_container, LV_ALIGN_BOTTOM_LEFT, 0, 0);
    
    animate_wheel(obj_wheel_l.gauge);
    animate_wheel(obj_wheel_ur.gauge);
    animate_wheel(obj_wheel_br.gauge);
}

void draw_wheel(lv_obj_t * parent_container, wheel_size_t size, wheel_t *output)
{
    printf("%s:%d (%s)\n", __FILE__, __LINE__, __FUNCTION__);
        
    lv_coord_t circle_size = 283;
    
    switch (size)
    {
        case WHEEL_SIZE_BIG:
            circle_size = 283;
        break;
        case WHEEL_SIZE_MEDIUM:
            circle_size = 189;
        break;
        case WHEEL_SIZE_SMALL:
            circle_size = 162;
        break;
    }
    
    // ARC   
    lv_obj_t * arc = lv_arc_create(parent_container);
    output->gauge = arc;
    lv_arc_set_bg_angles(arc, 0, 360);
    lv_arc_set_angles(arc, 270, 270);
    lv_obj_set_size(arc, circle_size, circle_size);
}

void animate_wheel(lv_obj_t *obj)
{
    lv_anim_t a;
    lv_anim_init(&a);
    lv_anim_set_values(&a, 10, 50);
    lv_anim_set_time(&a, 5000);
    lv_anim_set_playback_delay(&a, 1000);
    lv_anim_set_playback_time(&a, 800);
    lv_anim_set_repeat_delay(&a, 100);
    lv_anim_set_repeat_count(&a, LV_ANIM_REPEAT_INFINITE);
    lv_anim_set_exec_cb(&a, (void*) lv_arc_set_value);
    
    lv_anim_set_var(&a, obj);
    lv_arc_set_range(obj, 0, 1000);
    lv_anim_set_values(&a, 0, 1000);
    
    lv_anim_start(&a);
}

/**********************
 *   STATIC FUNCTIONS
 **********************/

/**
 * Initialize the Hardware Abstraction Layer (HAL) for the LVGL graphics
 * library
 */
static void hal_init(void)
{
  /* Use the 'monitor' driver which creates window on PC's monitor to simulate a display*/
  monitor_init();
  /* Tick init.
   * You have to call 'lv_tick_inc()' in periodically to inform LittelvGL about
   * how much time were elapsed Create an SDL thread to do this*/
  SDL_CreateThread(tick_thread, "tick", NULL);

  /*Create a display buffer*/
  static lv_disp_draw_buf_t disp_buf1;
  static lv_color_t buf1_1[MONITOR_HOR_RES * 100];
  static lv_color_t buf1_2[MONITOR_HOR_RES * 100];
  lv_disp_draw_buf_init(&disp_buf1, buf1_1, buf1_2, MONITOR_HOR_RES * 100);

  /*Create a display*/
  static lv_disp_drv_t disp_drv;
  lv_disp_drv_init(&disp_drv); /*Basic initialization*/
  disp_drv.draw_buf = &disp_buf1;
  disp_drv.flush_cb = monitor_flush;
  disp_drv.hor_res = MONITOR_HOR_RES;
  disp_drv.ver_res = MONITOR_VER_RES;
  disp_drv.antialiasing = 1;

  lv_disp_t * disp = lv_disp_drv_register(&disp_drv);

  lv_theme_t * th = lv_theme_default_init(disp, lv_palette_main(LV_PALETTE_BLUE), lv_palette_main(LV_PALETTE_RED), LV_THEME_DEFAULT_DARK, LV_FONT_DEFAULT);
  lv_disp_set_theme(disp, th);

  lv_group_t * g = lv_group_create();
  lv_group_set_default(g);
}

/**
 * A task to measure the elapsed time for LVGL
 * @param data unused
 * @return never return
 */
static int tick_thread(void *data) {
  (void)data;

  while(1) {
    SDL_Delay(5);
    lv_tick_inc(5); /*Tell LittelvGL that 5 milliseconds were elapsed*/
  }

  return 0;
}
```
</details>

### Expected behavior

Animation should happen without leaking.

### Screenshots or video

[image]

DCSBL, 2021-12-02:
Narrowed it down to this commit: https://github.com/lvgl/lvgl/commit/637b706ddc7adc237e8d63292068d2b859843843

embeddedt, 2021-12-02:
Very good catch! Thank you for the PR.

closed 2021-12-02 by DCSBL

ivooesch, 2022-06-03:
Now there is another problem:

When drawing a filled circle, the inner area has a size of 0,
an therefore mask_in_param will not get initialized,
resulting in a hardfault as lv_draw_mask_free_param()
is done always, even if mask_in_param was not initialized.
lv_draw_mask_free_param should only be done if
mask_in_param was initialized, e.g. when 
(lv_area_get_width(&area_in) > 0 && lv_area_get_height(&area_in)) > 0 was true

Code snippet (look for '<<<<<<<'):
 /*Create inner the mask*/
    int16_t mask_in_id = LV_MASK_ID_INV;
    lv_draw_mask_radius_param_t mask_in_param;
    if(lv_area_get_width(&area_in) > 0 && lv_area_get_height(&area_in) > 0) {
 <<<<< Init only done if area is not 0 >>>>>>>>>
        lv_draw_mask_radius_init(&mask_in_param, &area_in, LV_RADIUS_CIRCLE, true);
        mask_in_id = lv_draw_mask_add(&mask_in_param, NULL);
    }

    lv_draw_mask_radius_param_t mask_out_param;
    lv_draw_mask_radius_init(&mask_out_param, &area_out, LV_RADIUS_CIRCLE, false);
    int16_t mask_out_id = lv_draw_mask_add(&mask_out_param, NULL);

    /*Draw a full ring*/
    if(start_angle + 360 == end_angle || start_angle == end_angle + 360) {
        cir_dsc.radius = LV_RADIUS_CIRCLE;
        lv_draw_rect(draw_ctx, &cir_dsc, &area_out);

        lv_draw_mask_remove_id(mask_out_id);
        if(mask_in_id != LV_MASK_ID_INV)  lv_draw_mask_remove_id(mask_in_id);
          

        lv_draw_mask_free_param(&mask_out_param);

      <<<<<< Free always done, results in hardfault when area was 0 >>>>>>>>>>>>
        lv_draw_mask_free_param(&mask_in_param);

        return;
    }