lvgl/lvgl #2871: Menu in a semicircle

alex-ant-debug, 2021-12-02:
### Tell us about the problem
<! -
When building the menu in a semicircle, there was a problem with the arrangement of items. They should be located at the same distance from each other, that is, the angle between them should be the same.
->

### Examples and Cases
<! -
A layout was used to implement the semi-circular menu. Elements scroll normally, but are located at different distances.
Tried to align with:
- lv_obj_set_flex_align;
- lv_obj_set_style_pad_column;
- lv_style_set_pad_column.
I also tried to align by changing the coordinates of objects (my objects are scrolled horizontally, I tried to change the vertical coordinate, but it does not change):
- lv_obj_set_x;
- lv_obj_align;
- lv_obj_set_pos.
->

### Suggested solution
<! -
If you change the vertical coordinate of an object, then, according to calculations, the objects can be positioned at an equal distance.
[image]
In the first photo there is a menu without adjustments, in the second I change the vertical coordinate, but I cannot change the horizontal one. This makes the round menu look more like a triangle.
[image]
->

kisvegabor, 2021-12-03:
Can you send the code of the first semi-circular menu?

alex-ant-debug, 2021-12-03:
static const char * btnm_map[34] = {"_", " ", LV_SYMBOL_OK, LV_SYMBOL_CLOSE, LV_SYMBOL_BACKSPACE, LV_SYMBOL_KEYBOARD, LV_SYMBOL_EJECT, "1?&", "a", "b", "c", "d", "e", "f", "g", "h", "i", "j", "k", "l", "m", "n", "o", "p", "q", "r", "s", "t", "u", "v", "w", "x", "y", "z"};

static void scroll_event_menu_password_cb(lv_event_t * e)
{
    lv_obj_t * menu_object = lv_event_get_target(e);
    lv_event_code_t code_event = lv_event_get_code(e);

    if(code_event == LV_EVENT_SCROLL ) {

        lv_area_t screen_area;
        lv_obj_get_coords(menu_object, &screen_area);
        lv_coord_t screen_center = (lv_coord_t)(screen_area.x1 + lv_area_get_width(&screen_area) / 2);

        lv_coord_t password_menu_radius = (lv_coord_t)((lv_obj_get_width(menu_object) * 5 / 10) - 25);

        uint32_t child_index;
        uint32_t number_screen_children = lv_obj_get_child_cnt(menu_object);

        for(child_index = 0; child_index < number_screen_children; child_index++) {

            lv_obj_t * kids_buttons = lv_obj_get_child(menu_object, (int32_t)child_index);

            lv_area_t child_a;
            lv_obj_get_coords(kids_buttons, &child_a);

            lv_coord_t child_x_center = (lv_coord_t)(child_a.x1 + lv_area_get_width(&child_a) / 2);
            lv_coord_t difference_x = (lv_coord_t)(child_x_center - screen_center);

            /*Get the y of diff_y on a circle.*/
            lv_coord_t y;

            if((difference_x <= password_menu_radius*(-1))||(difference_x >= password_menu_radius)) {
                y += password_menu_radius;
            }else{
                /*Use Pythagoras theorem to get x from radius and y*/
                uint32_t y_sqr = (uint32_t)(password_menu_radius*password_menu_radius - difference_x*difference_x);

                lv_sqrt_res_t res;
                lv_sqrt(y_sqr, &res, 0x8000);
                y = (lv_coord_t)(password_menu_radius - res.i);
            }

            /*Translate the item by the calculated Y coordinate*/
            lv_obj_set_style_translate_y(kids_buttons, y, LV_STATE_DEFAULT);
        }
    }
}

void password(lv_obj_t * object)
{
    const int32_t center_letter = 8;      //center letter 'a'

    lv_obj_t * password_screen = lv_obj_create(object);
    lv_obj_set_size(password_screen, 320, 150);
    lv_obj_align(password_screen, LV_ALIGN_TOP_MID, 0, 0);
    lv_obj_add_style(password_screen, &style, 0);
	
	lv_obj_t * pwd_ta = lv_textarea_create(object);
 
    lv_textarea_set_password_mode(pwd_ta, true);
    lv_textarea_set_align(pwd_ta, LV_TEXT_ALIGN_CENTER);
    lv_textarea_set_one_line(pwd_ta, true);
    lv_obj_set_size(pwd_ta, 110, 40);
    lv_obj_set_pos(pwd_ta, 100, 160);

	lv_obj_t * password_text = lv_label_create(object);
    lv_label_set_recolor(password_text, true);
    lv_obj_align(password_text, LV_ALIGN_TOP_MID, 0, 120);
    lv_label_set_text(password_text, "Password:");
	
	lv_obj_set_flex_flow(password_screen, LV_FLEX_FLOW_ROW);

    lv_obj_add_event_cb(password_screen, scroll_event_menu_password_cb, LV_EVENT_SCROLL, NULL);

    lv_obj_set_scroll_dir(password_screen, LV_DIR_HOR);
    lv_obj_set_scroll_snap_x(password_screen, LV_SCROLL_SNAP_CENTER);
    lv_obj_set_scrollbar_mode(password_screen, LV_SCROLLBAR_MODE_OFF);
    lv_obj_add_flag(password_screen, LV_OBJ_FLAG_SCROLL_ONE);
	
	for(uint32_t i = 0; i < 34; i++) {

		lv_obj_t *btns_keyboard = lv_btn_create(password_screen);
		lv_obj_t *label_keyboard = lv_label_create(btns_keyboard);
		lv_label_set_text_static(label_keyboard, btnm_map[i]);

		lv_obj_center(label_keyboard);
		lv_obj_set_size(btns_keyboard, 25, 25);
	}
	
	lv_event_send(password_screen, LV_EVENT_SCROLL, NULL);

    /*Make sure the center button is "a"*/
    lv_obj_t * obj = lv_obj_get_child(password_screen, center_letter);
    lv_obj_scroll_to_view(obj, LV_ANIM_OFF);
}

kisvegabor, 2021-12-08:
Thanks. I'd do it in the same way, but it really results in an uneven angle between the elements. 

I think the best solution would be to create a new `layout` for this purpose. The problem is that snapping and scrolling won't work normally here as it expects a larger content than the container.  Or we can do some tricks like
- keep the objects in long row
- apply translate x/y based on the scroll position
- make snapping ignore the translates

So LVGL will think it scrolls and snaps to a normal list and we just visually offset the items to form a circle.

What do you think?

alex-ant-debug, 2021-12-09:
I also think that we need to create a separate layout for the semicircular menu. At the moment I am doing this.

https://user-images.githubusercontent.com/60813219/145388885-b34df99f-82a6-4f2d-b42a-26d22a8e987b.mp4

This menu is made with animation only.

kisvegabor, 2021-12-09:
Looks great! But in the end, it'd be nice to support snapping, "scroll-one", elastic scroll, and all the scrolling features too.

Are you interested in contributing to this new layout?

alex-ant-debug, 2021-12-20:
I am interested, but I have a project which there is little time. Because of this, I try to make everything as simple and fast as possible.

kisvegabor, 2021-12-20:
No problem :slightly_smiling_face:

kisvegabor, 2022-01-04:
I think it would be a really great feature so I've added this idea to the ROADMAP as an idea. https://github.com/lvgl/lvgl/blob/master/docs/ROADMAP.md

I close this issue until it gets relevant again.

closed 2022-01-04 by kisvegabor

hyansuper, 2023-05-28:
Can we make it more general and make a full circle or make the angle configurable?

kisvegabor, 2023-05-29:
We could make the angle configurable and let LVGL divide the elements eventy on that angle.

I hope I'll have time to deal with it when v9 is released.