lvgl/lvgl #2886: Click the Calendar widget in lv_demo_widgets() to cause a crash

FASTSHIFT, 2021-12-09:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

Asan output:

```
=================================================================
==27896==ERROR: AddressSanitizer: heap-use-after-free on address 0x60800000ba20 at pc 0x5555559d1e6d bp 0x7fffffffd5b0 sp 0x7fffffffd5a0
READ of size 8 at 0x60800000ba20 thread T0
    #0 0x5555559d1e6c in lv_obj_has_class /home/vifextech/workpath/misc/lv_sim_eclipse_sdl/lvgl/src/core/lv_obj.c:370
    #1 0x555555a04908 in lv_obj_get_event_dsc /home/vifextech/workpath/misc/lv_sim_eclipse_sdl/lvgl/src/core/lv_event.c:395
    #2 0x555555a04908 in event_send_core /home/vifextech/workpath/misc/lv_sim_eclipse_sdl/lvgl/src/core/lv_event.c:416
    #3 0x555555a0585b in lv_event_send /home/vifextech/workpath/misc/lv_sim_eclipse_sdl/lvgl/src/core/lv_event.c:74
    #4 0x5555559ba695 in indev_proc_release /home/vifextech/workpath/misc/lv_sim_eclipse_sdl/lvgl/src/core/lv_indev.c:933
    #5 0x5555559c1378 in indev_pointer_proc /home/vifextech/workpath/misc/lv_sim_eclipse_sdl/lvgl/src/core/lv_indev.c:368
    #6 0x5555559c1378 in lv_indev_read_timer_cb /home/vifextech/workpath/misc/lv_sim_eclipse_sdl/lvgl/src/core/lv_indev.c:100
    #7 0x555555ad9fa9 in lv_timer_exec /home/vifextech/workpath/misc/lv_sim_eclipse_sdl/lvgl/src/misc/lv_timer.c:313
    #8 0x555555ad9fa9 in lv_timer_handler /home/vifextech/workpath/misc/lv_sim_eclipse_sdl/lvgl/src/misc/lv_timer.c:109
    #9 0x5555558f879c in main /home/vifextech/workpath/misc/lv_sim_eclipse_sdl/main.c:100
    #10 0x7ffff50cab96 in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x21b96)
    #11 0x5555558fb3d9 in _start (/home/vifextech/workpath/misc/lv_sim_eclipse_sdl/demo+0x3a73d9)

0x60800000ba20 is located 0 bytes inside of 88-byte region [0x60800000ba20,0x60800000ba78)
freed by thread T0 here:
    #0 0x7ffff72b3c3f in free (/usr/lib/x86_64-linux-gnu/libasan.so.5+0x10bc3f)
    #1 0x555555a00dad in obj_del_core /home/vifextech/workpath/misc/lv_sim_eclipse_sdl/lvgl/src/core/lv_obj_tree.c:354
    #2 0x555555a00dad in obj_del_core /home/vifextech/workpath/misc/lv_sim_eclipse_sdl/lvgl/src/core/lv_obj_tree.c:359
    #3 0x555555a020c5 in obj_del_core /home/vifextech/workpath/misc/lv_sim_eclipse_sdl/lvgl/src/core/lv_obj_tree.c:354
    #4 0x555555a020c5 in lv_obj_del /home/vifextech/workpath/misc/lv_sim_eclipse_sdl/lvgl/src/core/lv_obj_tree.c:64
    #5 0x555555b6f004 in calendar_event_cb /home/vifextech/workpath/misc/lv_sim_eclipse_sdl/lv_demos/src/lv_demo_widgets/lv_demo_widgets.c:1243
    #6 0x555555a04d37 in event_send_core /home/vifextech/workpath/misc/lv_sim_eclipse_sdl/lvgl/src/core/lv_event.c:422
    #7 0x555555a04f7b in event_send_core /home/vifextech/workpath/misc/lv_sim_eclipse_sdl/lvgl/src/core/lv_event.c:434
    #8 0x555555a0585b in lv_event_send /home/vifextech/workpath/misc/lv_sim_eclipse_sdl/lvgl/src/core/lv_event.c:74
    #9 0x555555afcf02 in lv_btnmatrix_event /home/vifextech/workpath/misc/lv_sim_eclipse_sdl/lvgl/src/widgets/lv_btnmatrix.c:501
    #10 0x555555a04894 in lv_obj_event_base /home/vifextech/workpath/misc/lv_sim_eclipse_sdl/lvgl/src/core/lv_event.c:97
    #11 0x555555a04894 in event_send_core /home/vifextech/workpath/misc/lv_sim_eclipse_sdl/lvgl/src/core/lv_event.c:414
    #12 0x555555a0585b in lv_event_send /home/vifextech/workpath/misc/lv_sim_eclipse_sdl/lvgl/src/core/lv_event.c:74
    #13 0x5555559ba695 in indev_proc_release /home/vifextech/workpath/misc/lv_sim_eclipse_sdl/lvgl/src/core/lv_indev.c:933
    #14 0x5555559c1378 in indev_pointer_proc /home/vifextech/workpath/misc/lv_sim_eclipse_sdl/lvgl/src/core/lv_indev.c:368
    #15 0x5555559c1378 in lv_indev_read_timer_cb /home/vifextech/workpath/misc/lv_sim_eclipse_sdl/lvgl/src/core/lv_indev.c:100
    #16 0x555555ad9fa9 in lv_timer_exec /home/vifextech/workpath/misc/lv_sim_eclipse_sdl/lvgl/src/misc/lv_timer.c:313
    #17 0x555555ad9fa9 in lv_timer_handler /home/vifextech/workpath/misc/lv_sim_eclipse_sdl/lvgl/src/misc/lv_timer.c:109
    #18 0x5555558f879c in main /home/vifextech/workpath/misc/lv_sim_eclipse_sdl/main.c:100
    #19 0x7ffff50cab96 in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x21b96)

previously allocated by thread T0 here:
    #0 0x7ffff72b4038 in __interceptor_malloc (/usr/lib/x86_64-linux-gnu/libasan.so.5+0x10c038)
    #1 0x555555ad0c97 in lv_mem_alloc /home/vifextech/workpath/misc/lv_sim_eclipse_sdl/lvgl/src/misc/lv_mem.c:134
    #2 0x555555ad0c97 in lv_mem_alloc /home/vifextech/workpath/misc/lv_sim_eclipse_sdl/lvgl/src/misc/lv_mem.c:123
    #3 0x5555559d70f5 in lv_obj_class_create_obj /home/vifextech/workpath/misc/lv_sim_eclipse_sdl/lvgl/src/core/lv_obj_class.c:47
    #4 0x555555af99e8 in lv_btnmatrix_create /home/vifextech/workpath/misc/lv_sim_eclipse_sdl/lvgl/src/widgets/lv_btnmatrix.c:84
    #5 0x555555964ce9 in lv_calendar_constructor /home/vifextech/workpath/misc/lv_sim_eclipse_sdl/lvgl/src/extra/widgets/calendar/lv_calendar.c:276
    #6 0x5555559d885a in lv_obj_construct /home/vifextech/workpath/misc/lv_sim_eclipse_sdl/lvgl/src/core/lv_obj_class.c:189
    #7 0x5555559d885a in lv_obj_class_init_obj /home/vifextech/workpath/misc/lv_sim_eclipse_sdl/lvgl/src/core/lv_obj_class.c:109
    #8 0x555555963363 in lv_calendar_create /home/vifextech/workpath/misc/lv_sim_eclipse_sdl/lvgl/src/extra/widgets/calendar/lv_calendar.c:64
    #9 0x555555b6ed4d in birthday_event_cb /home/vifextech/workpath/misc/lv_sim_eclipse_sdl/lv_demos/src/lv_demo_widgets/lv_demo_widgets.c:1215
    #10 0x555555a04d37 in event_send_core /home/vifextech/workpath/misc/lv_sim_eclipse_sdl/lvgl/src/core/lv_event.c:422
    #11 0x555555a0585b in lv_event_send /home/vifextech/workpath/misc/lv_sim_eclipse_sdl/lvgl/src/core/lv_event.c:74
    #12 0x5555559b702c in lv_group_focus_obj /home/vifextech/workpath/misc/lv_sim_eclipse_sdl/lvgl/src/core/lv_group.c:244
    #13 0x5555559bd5f9 in indev_click_focus /home/vifextech/workpath/misc/lv_sim_eclipse_sdl/lvgl/src/core/lv_indev.c:1047
    #14 0x5555559bd5f9 in indev_proc_press /home/vifextech/workpath/misc/lv_sim_eclipse_sdl/lvgl/src/core/lv_indev.c:856
    #15 0x5555559c28f9 in indev_pointer_proc /home/vifextech/workpath/misc/lv_sim_eclipse_sdl/lvgl/src/core/lv_indev.c:365
    #16 0x5555559c28f9 in lv_indev_read_timer_cb /home/vifextech/workpath/misc/lv_sim_eclipse_sdl/lvgl/src/core/lv_indev.c:100
    #17 0x555555ad9fa9 in lv_timer_exec /home/vifextech/workpath/misc/lv_sim_eclipse_sdl/lvgl/src/misc/lv_timer.c:313
    #18 0x555555ad9fa9 in lv_timer_handler /home/vifextech/workpath/misc/lv_sim_eclipse_sdl/lvgl/src/misc/lv_timer.c:109
    #19 0x5555558f879c in main /home/vifextech/workpath/misc/lv_sim_eclipse_sdl/main.c:100
    #20 0x7ffff50cab96 in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x21b96)

SUMMARY: AddressSanitizer: heap-use-after-free /home/vifextech/workpath/misc/lv_sim_eclipse_sdl/lvgl/src/core/lv_obj.c:370 in lv_obj_has_class
Shadow bytes around the buggy address:
  0x0c107fff96f0: fa fa fa fa fd fd fd fd fd fd fd fd fd fd fd fa
  0x0c107fff9700: fa fa fa fa fd fd fd fd fd fd fd fd fd fd fd fa
  0x0c107fff9710: fa fa fa fa fd fd fd fd fd fd fd fd fd fd fd fa
  0x0c107fff9720: fa fa fa fa fd fd fd fd fd fd fd fd fd fd fd fa
  0x0c107fff9730: fa fa fa fa fd fd fd fd fd fd fd fd fd fd fd fa
=>0x0c107fff9740: fa fa fa fa[fd]fd fd fd fd fd fd fd fd fd fd fa
  0x0c107fff9750: fa fa fa fa fd fd fd fd fd fd fd fd fd fd fd fd
  0x0c107fff9760: fa fa fa fa fd fd fd fd fd fd fd fd fd fd fd fd
  0x0c107fff9770: fa fa fa fa fd fd fd fd fd fd fd fd fd fd fd fd
  0x0c107fff9780: fa fa fa fa fd fd fd fd fd fd fd fd fd fd fd fd
  0x0c107fff9790: fa fa fa fa fd fd fd fd fd fd fd fd fd fd fd fd
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
  Shadow gap:              cc
==27896==ABORTING
[1] + Done                       "/usr/bin/gdb" --interpreter=mi --tty=${DbgTerm} 0<"/tmp/Microsoft-MIEngine-In-oey4meim.co4" 1>"/tmp/Microsoft-MIEngine-Out-nc1inp1l.i4b"
```

### To Reproduce
Click here:
[image]

### Expected behavior

### Screenshots or video

kisvegabor, 2021-12-10:
Thanks, I pushed a fix.

closed 2021-12-10 by kisvegabor

embeddedt, 2021-12-13:
I looked into enabling AddressSanitizer on the CI to try and catch this type of issue in the future. Unfortunately, it's not very useful without LVGL being configured to use the system heap, and with the system heap `lv_mem_monitor` no longer measures heap usage, so some tests break.

FASTSHIFT, 2021-12-14:
@embeddedt 
Can `LV_MEM_CUSTOM` be enabled when running ci.

embeddedt, 2021-12-14:
That's what I tried, but the tests depend on the ability to measure heap usage, which `LV_MEM_CUSTOM` doesn't support.

kisvegabor, 2021-12-14:
Maybe we can run a special stress test (or the stress test demo) with a different config.

embeddedt, 2022-01-23:
I've just added support for using the system heap to run the tests as well. Note that any memory leak-related asserts need to be wrapped in `LV_HEAP_CHECK` ([example](https://github.com/lvgl/lvgl/commit/446b1ebf2bc1ba38b5349c660534f113a9a066a9#diff-8eb42324d18cea78c11fe69fce433b9ad3d354ba1f090ac812d953d457c90d18R93)) so that they are properly ignored when `LV_MEM_CUSTOM` is enabled.

commit 9810920f (2021-12-10): fix(event): crash if an object was deleted in an event

fixes: #2886