lvgl/lvgl #2887: it will failed display when define lv_disp_draw_buf_t in extern sdram

zsjlovestudy, 2021-12-10:
it will failed display when define lv_disp_draw_buf_t in extern sdram. 
[image]
[image]

The following is not defined in sdram, it's normal.
[image]
[image]

Other related codes:
[image]
[image]

I wanted to know if the external definition would speed things up, because my MCU doesn't have hardware acceleration. meanwhile, If it can speed up the display, how can you do it.

embeddedt, 2021-12-10:
I believe using SDRAM for LVGL's internal buffer is actually more likely to slow down rendering, as SDRAM is slower to access. 

However, I would check that caching is configured correctly (if that applies), and also try enabling [memory swapping](https://github.com/lvgl/lv_port_stm32f746_disco/commit/4f7dda9792996caf965625de02b3f72155f3d461) if you are using STM32.

zsjlovestudy, 2021-12-11:
the sdram's config should be corrected., because  These two buf are normal if lv_disp_draw_buf is be declare in internal flash.

[image]

kisvegabor, 2021-12-14:
> the sdram's config should be corrected., because These two buf are normal if lv_disp_draw_buf is be declare in internal flash.

How do you know if the SDRAM is correct from that it works if the buffers are placed into the internal RAM? (You wrote internal flash but I assume it's a typo)

zsjlovestudy, 2021-12-15:
> How do you know if the SDRAM is correct from that it works if the buffers are placed into the internal RAM? (

If when I put lv_disp_draw_buf into internal RAM, the LCD's two refresh buffers in SDRAM work fine, so I assume SDRAM is correct.  But when I put lv_disp_draw_buf into SDRAM, the LCD will be blurred.

kisvegabor, 2021-12-16:
Maybe I misunderstood it the whole time. By `lv_disp_draw_buf` do you mean only the `lv_disp_draw_buf_t` variable?

zsjlovestudy, 2021-12-17:
Yes, it was my inarticulation.   if  lv_disp_draw_buf_t variable is defined in sdram, the LCD will be blurred. But it will ok, if it in internal ram.

zsjlovestudy, 2021-12-17:
> Maybe I misunderstood it the whole time. By `lv_disp_draw_buf` do you mean only the `lv_disp_draw_buf_t` variable?

Yes, it was my inarticulation. if lv_disp_draw_buf_t variable is defined in sdram, the LCD will be blurred. But it will ok, if it in internal ram.

kisvegabor, 2021-12-20:
I see, it's event weirder. Maybe some tricky SDRAM config issue...
However this variable is only a few bytes. Why is it important to keep it in SDRAM?

zsjlovestudy, 2021-12-20:
> I see, it's event weirder. Maybe some tricky SDRAM config issue... However this variable is only a few bytes. Why is it important to keep it in SDRAM?
If I put this variable in SDRAM, it seems to refresh much faster, and I define this variable as the screen size, my MCU is not big enough to hold that much RAM.

kisvegabor, 2021-12-21:
To be honest I have no idea. There is nothing special in `lv_draw_buf_t`. 

I suggest asking about in the Forum. Hopefully someone already seen something like this.

closed 2021-12-21 by kisvegabor