lvgl/lvgl #2891: PNG decoder not work

SinglWolf, 2021-12-12:
Good day!
I beg your pardon for not writing according to the template again.

[image]
The file system is connected and configured.
The decoder library is also included.
The problem is this code doesn't work
```c
		static lv_obj_t *main_screen = NULL;
		main_screen = lv_obj_create(lv_scr_act());
		lv_obj_set_size(main_screen, LV_HOR_RES, LV_VER_RES);
		lv_obj_align(main_screen, LV_ALIGN_CENTER, 0, 0);
		LOGO = lv_img_create(main_screen);
		lv_img_set_src(LOGO, "S:/sdcard/wink.png");

		lv_fs_dir_t dir;
		lv_fs_res_t res;
		res = lv_fs_dir_open(&dir, "S:/sdcard/");
		if (res != LV_FS_RES_OK)
			ESP_LOGE(TAG, "LV_FS ERROR: %d", res);
		else
		{
			char fn[256];
			while (1)
			{
				res = lv_fs_dir_read(&dir, fn);
				if (res != LV_FS_RES_OK)
				{
					ESP_LOGE(TAG, "LV_FS ERROR: %d", res);
					break;
				}

				/*fn is empty, if not more files to read*/
				if (strlen(fn) == 0)
				{
					break;
				}

				printf("%s\n", fn);
			}

			lv_fs_dir_close(&dir);
		}
```
A white square appears on the display with the inscription "No data" inside. Interestingly, the square has a size of 50x50 pixels. The same as for the downloading file. The file with the smiley face is taken from the examples directory.
To check the operation of the file system, a code has been added that shows the contents of the root directory of the SD card.
```
/LOST.DIR
/Серия - Русская фантастика
org.fdroid.fdroid_1.9(1009000).apk
/Злотников Роман
/DCIM
FOO.TXT
Radiola-S2.png
wink.png
```
[By the way, why is there only one smiley face here?](https://docs.lvgl.io/master/libs/png.html?highlight=lv_png_init)
[image]
There are two of them in the micropython simulator.
[image]
How do you get an image from a file?

kisvegabor, 2021-12-16:
All looks good. What is the value of `LV_MEM_SIZE`? Note that this image needs 50x50x4=10kB RAM, plus the some more for temporary buffer during decoding. 

I suggest adding `printf`s to `lv_png.c` to see where it fails.

SinglWolf, 2021-12-16:
> What is the value of `LV_MEM_SIZE`? Note that this image needs 50x50x4=10kB RAM, plus the some more for temporary buffer during decoding.

`LV_MEM_SIZE` = 16384

```c
LV_IMG_DECLARE(RadiolaS2);
	lv_img_set_src(LOGO, &RadiolaS2);
```
I have a picture of 320x240 pixels, this method is loaded normally.

> I suggest adding `printf`s to `lv_png.c` to see where it fails.

Where exactly to insert?

[By the way, why is there only one smiley face here?](https://docs.lvgl.io/master/libs/png.html?highlight=lv_png_init)

kisvegabor, 2021-12-20:
> LV_MEM_SIZE = 16384

What if you increase it?

> I have a picture of 320x240 pixels, this method is loaded normally.

But it's not PNG, right?

> Where exactly to insert?

In [this](https://github.com/lvgl/lvgl/blob/master/src/extra/libs/png/lv_png.c#L65) and [this](https://github.com/lvgl/lvgl/blob/master/src/extra/libs/png/lv_png.c#L118) functions.

> By the way, why is there only one smiley face here?

Probably it's because the file is not present in online demo. @embeddedt know more about how it works.

SinglWolf, 2021-12-20:
> What if you increase it?

When it was like this, even the emoticon did not load.
[image]
When I changed to this, not only a smiley was loaded, but also a `320x240` pixel picture. I thought there was not enough memory, but everything was loaded. Understood nothing. `320 * 240 * 4` = `307200`.
[image]

kisvegabor, 2021-12-20:
So now you are using the built-in malloc/free whicj probably operates on a larger heap. 
I meant increasing this
[image]

If the 480x320 image is not  a PNG but an array in the flash it doesn't require extra RAM to be displayed as the image is already decoded.

SinglWolf, 2021-12-20:
I'm talking specifically about downloading pictures from the SD card.
E.g.
```c
lv_img_set_src(LOGO, "S:/sdcard/Radiola-S2.png");
```
This picture has a resolution of 480x320 pixels. Why is it displayed normally? In theory, there won't be enough memory to load this picture.

SinglWolf, 2021-12-20:
Yes, I am using extended memory. Now I understand.
Why isn't it in the documentation? When and what does `LVGL` use as a buffer. Or was I looking the wrong way again?
How can I reduce the uploaded image of `480x320` pixels to a display screen size of `320x240` pixels? Sorry off topic.
This is the last question here. Then issue can be closed.

SinglWolf, 2021-12-20:
> How can I reduce the uploaded image of `480x320` pixels to a display screen size of `320x240` pixels?

Will answer myself.
```c
lv_img_set_zoom(LOGO, 192);
```
The issue can be closed.

kisvegabor, 2021-12-21:
> lv_img_set_zoom(LOGO, 192);

Zooming is quite slow. But you can use any image editor software (online or offline) to scale the image.

closed 2021-12-21 by kisvegabor

SinglWolf, 2021-12-21:
> Zooming is quite slow.

Conversion time in this case does not play any role. The picture is loaded at startup and deleted after starting all tasks. I needed to reduce the resolution of the picture to load it on a display with a screen resolution of 320x240 pixels. Software image reduction allowed me to use one picture for two firmwares with different displays.