lvgl/lvgl #2894: Draw architecture improvements

kisvegabor, 2021-12-13:
As discussed in #2733 I further improved the new draw architecture in https://github.com/lvgl/lvgl/tree/arch/draw-restructure.

Some notes:
- All widgets are working
- Almost all drawing features are working (subpx rendering and blend modes are not working yet)
- STM and NXP GPUs are no reimplemented yet.
- [lv_driver_t](https://github.com/lvgl/lvgl/blob/arch/draw-restructure/src/hal/lv_hal_disp.h#L137-L140) contains function to init and deint an [lv_draw_ctx_t](https://github.com/lvgl/lvgl/blob/arch/draw-restructure/src/draw/lv_draw.h#L44) which contains callback to the high level drawing routines
- `lv_draw_ctx_t` is extended to [lv_draw_sw_ctx_t](https://github.com/lvgl/lvgl/blob/arch/draw-restructure/src/draw/sw/lv_draw_sw.h#L31). The init/deinit are [here](https://github.com/lvgl/lvgl/blob/arch/draw-restructure/src/draw/sw/lv_draw_sw.c#L40-L54)
- `lv_disp_drv_init` [sets the ctx init/deinit functions](https://github.com/lvgl/lvgl/blob/arch/draw-restructure/src/hal/lv_hal_disp.c#L90-L110) based on the settings in lv_conf.h but these are only the defaults, you overwrite them.
- `lv_disp_drv_register` creates an instance and initializes it with the init function [here](https://github.com/lvgl/lvgl/blob/arch/draw-restructure/src/hal/lv_hal_disp.c#L154-L160)
- Calling e.g.  [lv_draw_rect](https://github.com/lvgl/lvgl/blob/arch/draw-restructure/src/draw/lv_draw_rect.c#L60) simply calls the `draw_rect` callback from the `draw_ctx` (which can be e.g. [lv_draw_sw_rect](https://github.com/lvgl/lvgl/blob/arch/draw-restructure/src/draw/sw/lv_draw_sw_rect.c#L69))
- `lv_draw_ctx_t` know the current draw buffer, draw area and clip area. The buffer is set by LVGL to a buffer from `lv_draw_buf_t`.
- I haven't tested, but e.g. [DMA2D](https://github.com/lvgl/lvgl/blob/arch/draw-restructure/src/draw/stm32_dma2d/lv_gpu_stm32_dma2d.c#L100-L111) should extend the `draw_ctx` like this.

Changes in the draw API:
- All draw functions get `draw_ctx` as [first argument](https://github.com/lvgl/lvgl/blob/arch/draw-restructure/src/draw/lv_draw.h#L61-L76).
- Blend fill and map is [joined into a "blend" function](https://github.com/lvgl/lvgl/blob/arch/draw-restructure/src/draw/sw/lv_draw_sw_blend.c#L96) which gets a descriptor instead of many parameters. Finally they use absolute coordinates. (Sorry for changing it back and forth, I hope this "absolute" is the absolute final :slightly_smiling_face:  ) 
- @mariotaku It'd be great if you could set the SDL texture as `lv_draw_buf_t` and let LVGL assign it to `draw_ctx->buf`. Now `draw_buf`s are `lv_color_t *` but they should be `void *` in a new release. So now you can cast the texture to `lv_color_t *`. Could it work?
- [lv_snapshot](https://github.com/lvgl/lvgl/blob/arch/draw-restructure/src/extra/others/snapshot/lv_snapshot.c#L108-L144) was improved (now it can handle clipped objects too) (cc @XuNeo)

Questions:

Ultimately the goal was to make draw functions work seeing only the `draw_ctx` and knowing nothing about display, driver, etc. So one can just init a context and call the draw functions.  
It'd be possible only with some API break so it's just partially true now. I'm also not sure you how to use this "new power" of exchangeable draw engines. Maybe we can have many SW renders for monochrome, gray scale, e-Ink, etc displays?

`blend` is not part of `lv_draw_ctx_t` but only `lv_draw_sw_ctx_t` as it's not true the all draw engine will have this blending-like stage in the pipeline or it's accessible this way. E.g. with Cairo we just want to see the draw feature and let Cairo will draw as it wishes. Following this, a generic masks interface should be also parts of `lv_draw_ctx_t` as masks are handled differently in different systems. Now, we use `lv_draw_mask_...` functions we so can't really integrate something like Cairo which works with different masking concept. I postponed it for a next iteration.

---

After trying many approaches I think finally it became an interface which is flexible and easy to use, understand and debug. 
Please let me know what you think.

mariotaku, 2021-12-13:
I have a draft pull request #2890 that improves SDL drawing backend. And I'm happy to refactor based on this PR.

For using texture pointer in draw_buf: Yes that makes sense. The texture is a virtual buffer indeed.
For lv_snapshot I have some interests about that. This may help me implement butter smooth scrolling experience.

This is a good one!

mariotaku, 2021-12-14:
@kisvegabor 

For rebasing: I'll do later this week.

For lv_snapshot: Yes, SDL backend is already an texture blender. It renders shapes with software into texture, caches them and draw with higher performance. If we cache a whole widget into texture, then dozens of draw calls can be reduced to 1. 

For glfw: Only when we use OpenGL, glfw will become useful. For now we don't use it directly.

You may remember that I started to write NanoVG backend. It turns out the performance is quite not ideal as it draws whole shape repeatedly. I'll get it done once I finish SDL one.

kisvegabor, 2021-12-14:
All good and clear, thanks! :slightly_smiling_face:

mariotaku, 2021-12-14:
@kisvegabor Hi, I spend some time rebased the PR. Could you please take a look?

kisvegabor, 2021-12-14:
Merged, thank you!

mariotaku, 2021-12-16:
I have another request, regarding draw arch design.

When I initialize SDL backend, two instances are needed. One is `SDL_Renderer *`, and another is screen sized `SDL_Texture *`. While I put texture into `lv_draw_buf_t.buf1`, renderer instance can only be passed by `lv_disp_drv_t.user_data`.
This also applies to NanoVG: `NVGcontext` and `NVGframebuffer`.

I'm wondering if there are better ways to pass arguments when register a display driver.

kisvegabor, 2021-12-16:
I imagined it exactly like this. I think it's fine to use the `user_data` of the driver. That's why this field exists. :slightly_smiling_face:

XuNeo, 2021-12-17:
Hi guys,

The snapshot has a performance limitation tested on device. The set_px_cb will bypass all GPU functions and render pixel by pixel. 

@FASTSHIFT mentioned to use fake display driver similar to the the real one, and use flush_cb instead, but this cannot handle snapshots with alpha value. 
Any ideas?

Thanks.

mariotaku, 2021-12-17:
How about to provide a callback for snapshot as well?

puzrin, 2021-12-17:
My 5c

In long term, lvgl lacks of abstraction layer "shapes pipeline". Usually, this abstraction isolates rendering engines (software-based & GPU-based)  explicitly. But this layer requires additional buffer for commands. I don't know is it possible to do buffer tune-able, up to zero size.

Currently, this layer is expressed indirectly, via set of `draw_XXX` functions (line, circle, texture and so on). But in theory, is lot of ram available, it's better to stack drawing commands and process those in batches. For example, GPU renderer can do batches very fast, without CPU load.

kisvegabor, 2021-12-20:
> The snapshot has a performance limitation tested on device. The set_px_cb will bypass all GPU functions and render pixel by pixel.

We can optimize the case of ARGB set_pixel_cb. If `LV_COLOR_SCREEN_TRANSP` is enabled the belnd function can handle it without set_px_cb too, but now it work only for 32 bit color format. So we extend the blend function to handle these case with the usual performance.

kisvegabor, 2021-12-20:
> In long term, lvgl lacks of abstraction layer "shapes pipeline". Usually, this abstraction isolates rendering engines (software-based & GPU-based) explicitly. But this layer requires additional buffer for commands. I don't know is it possible to do buffer tune-able, up to zero size.

It's doable now too. In a custom `draw_rect` function we can buffer the commands and send in one batch (or multiple smaller ones if the buffer size is limited).

kisvegabor, 2021-12-21:
Now DMA2D is working too. NXP PXP and VGLITE still needs to be updated.

I'd merge this branch to master to not block further development.

Do you agree?

mariotaku, 2021-12-21:
I have no problem if the PR gets merged.

kisvegabor, 2021-12-21:
Merged! :tada: 

Thank you very much for your contribution! I think it was huge step forward.

I close this issue now. Let's open more specific ones if something comes up.

closed 2021-12-21 by kisvegabor