lvgl/lvgl #2895: Updating from 7.5 to 8.1 issue(s)

elgerg, 2021-12-13:
### Perform all steps below and tick them with [x]
- [x ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ x] Update lvgl to the latest version
- [ x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
Trying to update from 7.5 to 8.1 and having issues like:

App/App.c:240:5: error: unknown type name ‘lv_tabview_ext_t’; did you mean ‘lv_tabview_t’?
  240 |     lv_tabview_ext_t * ext = (lv_tabview_ext_t *)lv_obj_get_ext_attr(tabView); 

### To Reproduce
```c
    lv_tabview_ext_t * ext = (lv_tabview_ext_t *)lv_obj_get_ext_attr(tabView);
    const char * tab_name  = ext->tab_name_ptr[lv_tabview_get_tab_act(tabView)];

    if(strcmp(tab_name, LV_SYMBOL_HOME) == 0) {
        showMenuTime = false;
    } else {
        showMenuTime = true;
    }
```

### Expected behavior
I would have hoped that it would work but there have been chances that are not backwards compatible. 

Is there an upgrade guide that I can follow that has the old types and names to the new ones?

### Screenshots or video

elgerg, 2021-12-17:
hi,

The next one is tables.
In 7.5 the corners were rounded, in 8.1 they are not. How do I enable rounding so the table looks the same as the "object" next to it?
[image]
Thanks!

elgerg, 2021-12-17:
Next,

In 7.5 I managed to get rid of the scroll bar on the right:
[image]
How do i do that in 8.1?
Thanks

elgerg, 2021-12-17:
Same with the scrollbar at the bottom here:
[image]

elgerg, 2021-12-17:
Next...

In 7.5 you could create a label aligned to another item (lets say another label using align out bottom left) and then center the content using lv_label_set_align, how do you do this in 8.1?

Thanks

elgerg, 2021-12-17:
Next one..

How do I change the buttons to look like they used to?

7.5:
[image]

8.1:

elgerg, 2021-12-17:
What is the 8.1 equivalent of lv_switch_on ?

elgerg, 2021-12-18:
Next...

What is the 8.1 for lv_cont_set_fit2 ?

elgerg, 2021-12-18:
@kisvegabor any pointers would be appreciated.
Thanks!
Alex

kisvegabor, 2021-12-20:
Hi,

These questions are really relevant and there is no such a details migration guide at this moment. We have only [this short summary](https://docs.lvgl.io/master/CHANGELOG.html#v8-0-0-01-06-2021) now. I'd be happy to update it based on your questions. 

However, we use GitHub for development related discussion. Please ask these question in the Forum and add the topic's link here. I'll answer all the questions in the Forum.

closed 2021-12-20 by kisvegabor

elgerg, 2021-12-20:
Table corners:
https://forum.lvgl.io/t/migrating-from-7-5-to-8-1-table-corners/7645

elgerg, 2021-12-21:
Buttons:
https://forum.lvgl.io/t/migrating-from-7-5-to-8-1-buttons/7656

elgerg, 2021-12-21:
Table cell colouring:
https://forum.lvgl.io/t/migrating-from-7-5-to-8-1-table-cell-colours/7658

elgerg, 2021-12-21:
Tabview padding:
https://forum.lvgl.io/t/migrating-from-7-5-to-8-1-tabview-padding/7659

elgerg, 2021-12-21:
Remove tab outline:
https://forum.lvgl.io/t/migrating-from-7-5-to-8-1-tabview-outline/7660

elgerg, 2021-12-22:
Table cell alignment:
https://forum.lvgl.io/t/migrating-from-7-5-to-8-1-table-cell-align-lv-table-set-cell-align/7674

elgerg, 2021-12-22:
Table behaving oddly:
https://forum.lvgl.io/t/migrating-from-7-5-to-8-1-table-behaving-odd/7675