lvgl/lvgl #2896: The lv_dropdown widget does not inherit the included font.

SinglWolf, 2021-12-14:
The `lv_dropdown` widget does not inherit the included font.
```c
	lv_obj_t *main_screen = lv_obj_create(lv_scr_act());
	lv_obj_set_size(main_screen, LV_HOR_RES, LV_VER_RES);
	lv_obj_align(main_screen, LV_ALIGN_CENTER, 0, 0);
	lv_obj_set_style_text_font(main_screen, &lv_font_radiola_16, LV_PART_MAIN);
	lv_obj_t *ListStation = lv_dropdown_create(main_screen);
	lv_dropdown_clear_options(ListStation);
	lv_obj_set_width(ListStation, 316);

	//	lv_dropdown_set_show_selected(ListStation, false);
	const char *list_st[] = {" СПИСОК СТАНЦИЙ ", " LIST OF STATIONS "};
	lv_dropdown_set_text(ListStation, list_st[MainConfig->translate]);
	lv_group_add_obj(main_g, ListStation);
	liststation = true;

	for (uint8_t i = 0; i < GetTotalStations(); i++)
	{
		si = getStation(i);
		printf("Name station: %s\n", si->name);
		lv_dropdown_add_option(ListStation, si->name, i);
	}
	lv_dropdown_set_selected(ListStation, GetCurrentStation());
	lv_obj_align(ListStation, LV_ALIGN_TOP_LEFT, 0, 0);
	lv_obj_add_event_cb(ListStation, event_list, LV_EVENT_ALL, NULL);
	lv_dropdown_open(ListStation);
	if (si != NULL)
	{
		free(si);
	}
```
Result on screen:
[image]
Log:
```
Name station: Радио Спутник - Волгоград
Name station: Атмосфера
Name station: Релакс ФМ
Name station: Дорожное
Name station: Карнавал
```
By the way, everything is displayed normally in the title.
The `LVGL` version is fresh (`8.1.1`).

SinglWolf, 2021-12-15:
```c
	lv_obj_t *ListStation = lv_dropdown_create(main_screen);
#if defined CONFIG_LV_TFT_DISPLAY_CONTROLLER_ILI9488
	lv_obj_set_style_text_font(ListStation, &lv_font_radiola_18, LV_PART_MAIN);
#elif CONFIG_LV_TFT_DISPLAY_CONTROLLER_ILI9341
	lv_obj_set_style_text_font(ListStation, &lv_font_radiola_16, LV_PART_MAIN);
#endif
```
Forcing the font to be declared to the widget doesn't fix the problem!
UTF-8 symbols are still not displayed in the list.

kisvegabor, 2021-12-15:
Hi,

It's because you have created the drop-down list on  `main_screen` object (which is not really a screen just a simple object) but the list of the drop-down is internally moved to the screen to not limit it's visibility to the parent of the drop down.

To solve the issue you can:
```
lv_obj_set_style_text_font(lv_dropdown_get_list(ListStation), &lv_font_montserrat_30, LV_PART_MAIN);
```

SinglWolf, 2021-12-15:
> To solve the issue you can:

Thanks! It worked.

closed 2021-12-15 by SinglWolf